corna/me_cleaner #1: Skylake Should Work

benwis, 2016-11-28:
Hi,
I'm thrilled that the community has made progress into disabling the ME for modern intel chipsets. I have an Intel Skylake processor, and I was wondering what should work meant. How does that differ from untested? I'm considering attempting the fix on my CPU/motherboard, and would love some guidance on that. Is there a way to restore it if the 30 min limit is not bypassed?

Apologies if this is the wrong place, happy holidays!

corna, 2016-11-28:
Hi,

please note that this tool does not remove ME completely (yet), it only removes most (but not all) of its code. We're trying to understand if the code can be reduced even more (and it appears that [it can be done](https://www.coreboot.org/pipermail/coreboot/2016-November/082335.html)).

Trammel Hudson has manually created a reduced ME image for Skylake and [it appears to work](https://www.coreboot.org/pipermail/coreboot/2016-November/082408.html). Moreover there is the problem of the Boot Guard, so I don't know if it works universally.
If it works for you using this tool, please report it here #3.

Be careful! The 30 min window is given only if the ME image is missing; if this tool does not work on your device, your PC won't turn on at all, and you will need an external flasher to unbrick your device.

Nimayer, 2016-12-01:
Hi,
I just tested __me_cleaner__ on a Skylake desktop board that should have Intel Boot Guard enabled (MSI Bazooka B150M).
After modifying the blob, the BIOS complains at startup saying:
> The ME FW of system was found abnormal
> It is recommended to re-flash system BIOS by M-Flash to ensure normal system operation.
> Press F1 to Run Setup
> Press F2 to continue

But if you press F2 the computer boots normally and everything works, and it doesn't shut down after 30 minutes.
So I think Boot Guard does not protect from modifying the ME blob.

Regarding what you were asking about the 30 minutes:
if you keep an unmodified dump of your BIOS, you can always restore your PC by flashing it, even if your PC does not turn on, in fact to modify ME blob you always need to flash externally by SPI with your PC shut down.

closed 2016-12-03 by corna