corna/me_cleaner #2: Reduce size of ME by removing modules

zaolin, 2016-11-28:
Hey,

is it possible to reduce the size of the ME instead of 0xFF the space ?
So that the people can have more space in their spi rom.

corna, 2016-11-28:
Hi,

currently if you cut the ME region in ifdtool just after the end of the FTPR partition, it doesn't boot. Trammel Hudson [reduced the image size to 1 MB](https://www.coreboot.org/pipermail/coreboot/2016-November/082335.html) and it worked, but it is currently unclear how to reduce it further or how to relocate the FTPR partition (if possible).

If you feel audacious you can try the experiments_effs_flag branch and try to cut the image just after the end of the FTPR partition (read the me_cleaner.py output to find out its position).

zamaudio, 2016-11-29:
This is pwesome

closed 2017-01-18 by zaolin