corna/me_cleaner #3: me_cleaner status

corna, 2016-11-28:
Please comment here if me_cleaner works on your device.
Specify:
 * CPU architecture
 * CPU model
 * Laptop/motherboard
 * OEM BIOS or Coreboot?
 * Latest working commit

Thanks

corna, 2016-11-28:
Working on:
 * Intel Core i5-2520M
 * Sandy Bridge
 * Lenovo Thinkpad X220
 * Coreboot
 * d1abbca

Working for more than a month now. Everything works perfectly, and the MEI device has disappeared from the PCI bus.

afics, 2016-11-29:
I can confirm it works on the Lenovo Thinkpad X220, but coreboot then recognizes only one of my two 8GB RAM modules. I'm currently investigating.

persmule, 2016-11-30:
Seems working on:

    Intel Core i3-2377M
    Sandy Bridge
    Acer Aspire S3-371
    OEM BIOS
    d1abbca

MEI device has NOT disappeared on this platform, but seems to have been neutralized according to intelmetool's output.
[superiotool.txt](https://github.com/corna/me_cleaner/files/620872/superiotool.txt)
[lspci.txt](https://github.com/corna/me_cleaner/files/620870/lspci.txt)
[dmidecode.txt](https://github.com/corna/me_cleaner/files/620869/dmidecode.txt)
[intelmetool_deme.txt](https://github.com/corna/me_cleaner/files/620871/intelmetool_deme.txt)

sinetek, 2016-11-30:
Hey there, I build an image for my chromebook (C720p) running a Haswell 2955U.
The machine is running right now, will report back in ~22 minutes if no shutdown happens.

sinetek, 2016-11-30:
There is no MEI entry in the lspci list.
Not sure what other tests I can run to see the ME's state, open to running other tests, just tell me.

sinetek, 2016-11-30:
Passed the 30 minutes mark, seems to work. Thanks

corna, 2016-11-30:
@persmule open an issue, so that we can discuss about it

corna, 2016-11-30:
@sinetek That's enough, thanks.

flacks, 2016-12-01:
@sinetek That's interesting... On C720P + 2955U, I built coreboot and added the neutered ME, but only got an unresponsive power button, and the system shuts off after 20 seconds. There is no MEI device in lspci here, either.

sinetek, 2016-12-01:
@flacks  Try using my repo, follow the instructions I wrote and everything should work.
https://github.com/sinetek/coreboot-peppy

Failing that I could send you the ROM directly.

Nimayer, 2016-12-01:
Working on:

- Intel i5-6500
- Skylake
- MSI Bazooka B150M
- Stock AMI Bios
- https://github.com/corna/me_cleaner/commit/61fd606155c54d0ed8a54fe3c4a499f0645434e3

Everything works, the HECI (formerly MEI) device disapperars, a screen at boot notifies that the ME firmware is corrupted, but pressing F2 lets the boot continue.
[image]

zamaudio, 2016-12-02:
- Intel
- Core i5-2540M
- Lenovo Thinkpad X220
- Coreboot
- 48deb6c

^^ Yes that's right, the experimental branch works on this board folks! No lzma modules!

simonepsp, 2016-12-12:
Working on 
- Intel
- Core i5 3320M (Ivy Bridge)
- Lenovo Thinkpad X230
- Coreboot
- d2e2308

tlaurion, 2016-12-12:
Working on
-     Intel Core i5 3320M
-     Ivy Bridge
-     Lenovo Thinkpad X230
-     Stock BIOS.
-     https://github.com/corna/me_cleaner/commit/d2e2308588b906f386b5e7bfaffa9ef66b371945

ilikenwf, 2016-12-19:
* Intel Core i7 4790k
* Haswell
* Desktop, ASRock Z97 Pro4
* Modded Bios (see below)

ASRock's bios packages are all in a proprietary format, but Windows based tools, specifically the UBU pack, allow them to be extracted and the firmware inside upgraded or downgraded. One may flash this modified file directly from the UEFI settings themselves, as it doesn't validate them.

I updated my firmwares, and then used me_cleaner against the resulting bin file. Output is as below.

The only issue I noted was that intel_iommu enabled on my kernel boot line caused a lockup (kernel 4.9). Downgrading the firmware to an earlier version and stripping me again fixed this.

**All that said, the mei and mei_me modules still load on the system, but intelmetool and others segfault when I attempt to determine the status. Is it enabled and working? Is it not? I really can't tell.**

```
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Reading FTPR modules list...
Wiping LZMA section (0xa7680 - 0xcf000)
 UPDATE: removed (0xa7680 - 0xa78aa)
 ROMP: removal of Huffman modules is not supported yet, skipping
 BUP: removal of Huffman modules is not supported yet, skipping
 KERNEL: removal of Huffman modules is not supported yet, skipping
 POLICY: removal of Huffman modules is not supported yet, skipping
 HOSTCOMM: removed (0xa78aa - 0xafbb5)
 TDT: removed (0xafbb5 - 0xb4f71)
 FPF: removed (0xb4f71 - 0xb6a77)
Correcting checksum (0xea)...
Done! Good luck!
```

GenericHero, 2016-12-19:
- Intel Core i7 4790k
- Haswell
- Desktop, ASRock Z97 Extreme6
- Modded bios --  see https://github.com/corna/me_cleaner/issues/3#issuecomment-267880212
- https://github.com/corna/me_cleaner/commit/ffe60d8f7248afd52fa1bb21465415497d54f74f

BIOS file name must be same as Instant Flash bios name, or else instant flash in bios does not detect it. In this case, Z97Ex62.70

/dev/mei0 does not exist, intelmetool reports it doesnt support my system (maybe it doesn't?)

Of note, Intel's own "Intel® Management Engine Verification Utility" in windows is perpetually spinning, which I should have tested beforehand. But everything seems to be working properly so far.

```
Full image detected

The ME region goes from 0x3000 to 0x1fffff

Found FPT header at 0x3010

Found 20 partition(s)
ME firmware version 9.1.10.1000
Found FTPR header: FTPR partition spans from 0x4a000 to 0xd2000
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Reading FTPR modules list...
Wiping LZMA section (0xaa680 - 0xd2000)
 UPDATE          : removed (0xaa680 - 0xaa8aa)
 ROMP            : removal of Huffman modules is not supported yet, skipping
 BUP             : removal of Huffman modules is not supported yet, skipping
 KERNEL          : removal of Huffman modules is not supported yet, skipping
 POLICY          : removal of Huffman modules is not supported yet, skipping
 HOSTCOMM        : removed (0xaa8aa - 0xb2bb5)
 TDT             : removed (0xb2bb5 - 0xb7f71)
 FPF             : removed (0xb7f71 - 0xb9a77)
Correcting checksum (0xea)...
Done! Good luck!
```

nilesr, 2016-12-19:
>I can try to cat /dev/mei0 and I get "no such device" as root...so I guess that's good?

On my system that I've never flashed before (I haven't used me_cleaner yet) I get the same message when trying to read from /dev/mei0.

ilikenwf, 2016-12-19:
That is why I am interested to determine if this is significant or not...

I think my processor, because it is a "K" processor, doesn't have vPro,
which may prevent the ME from working anyway?

ilikenwf, 2016-12-19:
Reading up it appears the unlocked K processors don't include vPro, which I
believe is required for ME to work properly?

I'd say I've stripped it successfully, but had no real reason to do so?

I'll have to try this on an old laptop I have, will report back once I have
the chance.

n1zzo, 2016-12-19:
Working on

- Intel Core i5 2520M
- Sandy Bridge
- Dell Latitude e6220
- OEM BIOS
- ffe60d8

MEI is no more present in lspci output.
The system boots fine without even complaining.

Keep up the good work!

citypw, 2016-12-20:
Working on:
* Intel Core i7-3770
* Sandy Bridge
* Motherboard: GA-B75M-D3V
* OEM BIOS

citypw, 2016-12-24:
Working on:
* Intel Core i3-2370m
* SandyBridge
* Motherboard: Thinkpad x220i
* OEM BIOS and Coreboot
* [ffe60d8f](https://github.com/corna/me_cleaner/commit/ffe60d8f7248afd52fa1bb21465415497d54f74f)

MEI device has disappeared from the PCI bus.

persmule, 2016-12-24:
Working on:

    Intel Core i7-3770T
    IvyBridge
    Motherboard: GA-B75M-D3H
    Coreboot
    ffe60d8f
Sadly I failed to extract a valid OEM BIOS image this time. MEI device has disappeared from the PCI bus.

persmule, 2016-12-24:
Working on:

- Intel Core i7-3770T
- IvyBridge
- Motherboard: GA-B75M-D3H
- Coreboot
- ffe60d8
- Neutralized ME from SnB/IvB laptop (e.g. samsung lumpy's ME image located at 3rdparty/blobs/mainboard/samsung/lumpy/me.bin, with ifd adjusted via modified layout file)
MEI device won't go away and confirmed that ME is broken, and integrated graphic card conpletely cease to work.

ehmry, 2016-12-24:
Works on:
 * Core i5-2520M
 * SandyBridge
 * Lenovo Thinkpad T420
 * OEM BIOS
 * ffe60d8f7248afd52fa1bb21465415497d54f74f

jantatje, 2016-12-25:
* Intel Core i5-2520M
* Lenovo Thinkpad X220
* Coreboot-4.5
* Seabios 1.9.3
* `ffe60d8`

Everything works, but laptop hangs for ~15 seconds after suspend. I also updated coreboot, so not 100% sure this is me_cleaners fault.

persmule, 2016-12-27:
Works on:

- Core i5-2520M
- SandyBridge
- Lenovo Thinkpad T420
- Coreboot-4.5
- Seagrub scheme from libreboot
- 1.5MiB Management Engine
- ffe60d8
MEI device has disappeared from the PCI bus.

persmule, 2016-12-28:
Working on

- Intel Core i5 2520M
- Sandy Bridge
- Dell Latitude e6220
- OEM BIOS
- ffe60d8
- 

MEI is no more present in lspci output. However, the ethernet card needs a warm reboot to be functional.

Kokokokoka, 2016-12-28:
Working on:
T420: Intel core i7 3632qm, Ivy Bridge, seabios+windows. screen goes blank after a while, running a background game app seems to help this somehow.
X220: Intel core i5 2540M, Sandy Bridge, linux payload+qubes, worked fine, don't remember the blank screen issue.

persmule, 2016-12-28:
Working on

- Intel Core i3 2330M
- Sandy Bridge
- HP EliteBook 8460P
- OEM BIOS
- ffe60d8
- 

MEI is no more present in lspci output. However, the ethernet card needs a warm reboot to be functional.

JohnnyLeone, 2016-12-31:
Working on

- Intel Core i5 3320M
- Ivy Bridge
- Lenovo Thinkpad X230
- Coreboot
- ffe60d8

theracermaster, 2017-01-02:
Presumably working on:
 - Intel Core i5-3570K (Ivy Bridge)
 - Gigabyte GA-Z77X-UD5H
 - coreboot with TianoCore UEFI payload
 - ffe60d8

The MEI Controller device still appears in `lspci`, but the integrated graphics card disappeared. I am unsure of the status of the Intel 82579V Ethernet controller, as I haven't gotten it to work yet (`e1000e: probe of 0000:00:19.0 failed with error -3`; this remains the same with normal ME or cleaned ME). The ME appears to have been disabled:

```
[   19.881125] mei_me 0000:00:16.0: wait hw ready failed
[   19.881131] mei_me 0000:00:16.0: hw_start failed ret = -62
[   19.881144] mei_me 0000:00:16.0: H_RST is set = 0x80000015
[   21.929169] mei_me 0000:00:16.0: wait hw ready failed
[   21.929175] mei_me 0000:00:16.0: hw_start failed ret = -62
[   21.929188] mei_me 0000:00:16.0: H_RST is set = 0x80000015
[   23.977227] mei_me 0000:00:16.0: wait hw ready failed
[   23.977233] mei_me 0000:00:16.0: hw_start failed ret = -62
[   23.977236] mei_me 0000:00:16.0: reset: reached maximal consecutive resets: disabling the device
[   23.977238] mei_me 0000:00:16.0: reset failed ret = -19
[   23.977239] mei_me 0000:00:16.0: link layer initialization failed.
[   23.977241] mei_me 0000:00:16.0: init hw failure.
[   23.977366] mei_me 0000:00:16.0: initialization failed.
```

persmule, 2017-01-06:
Working on:
-Intel Core i5-2520M
-Sandy Bridge
-Lenovo Thinkpad X220
-Coreboot
-4e9fc2e

After the KERNEL module of ME is removed, the integrated NIC works after a COLD reboot now.

Kokokokoka, 2017-01-06:
- Intel Core i5-2520M
- Sandy Bridge
- Lenovo Thinkpad X220 tablet
- Coreboot
- 4e9fc2e

cryptofuture, 2017-01-07:
* i7-3517U (Ivy)
* Intel HM76 / Foxconn NanoPC AT-7700
* OEM
* 4e9fc2e

persmule, 2017-01-08:
Working on:

- Intel Core i5-2520M
- Sandy Bridge
- HP EliteBook 8460P
- OEM BIOS
- 4e9fc2e

After the KERNEL module of ME is removed, the integrated NIC works after a COLD reboot now.

drkhsh, 2017-01-11:
- Intel Core i5-2520M
- Sandy Bridge
- Lenovo Thinkpad T420
- Coreboot
- 4e9fc2e

**Everything** works perfectly for one day now!

citypw, 2017-01-13:
Working on:

* Intel Xeon E3-1230 v5
* Skylake
* Motherboard: P10S-M WS
* OEM BIOS
* [4e9fc2e](https://github.com/corna/me_cleaner/commit/4e9fc2ef59e097c5a6dc7d10d72b76e193236bc7)

```
#lspci -vnnt
-[0000:00]-+-00.0  Intel Corporation Sky Lake Host Bridge/DRAM Registers [8086:1918]
           +-01.0-[01]--+-00.0  Advanced Micro Devices, Inc. [AMD/ATI] Tahiti XT [Radeon HD 7970/8970 OEM / R9 280X] [1002:6798]
           |            \-00.1  Advanced Micro Devices, Inc. [AMD/ATI] Tahiti XT HDMI Audio [Radeon HD 7970 Series] [1002:aaa0]
           +-13.0  Intel Corporation Sunrise Point-H Integrated Sensor Hub [8086:a135]
           +-14.0  Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller [8086:a12f]
           +-14.2  Intel Corporation Sunrise Point-H Thermal subsystem [8086:a131]
           +-16.0  Intel Corporation Sunrise Point-H CSME HECI #1 [8086:a13a]
           +-16.1  Intel Corporation Sunrise Point-H CSME HECI #2 [8086:a13b]
           +-17.0  Intel Corporation Device [8086:a102]
           +-1c.0-[02]----00.0  Intel Corporation I210 Gigabit Network Connection [8086:1533]
           +-1c.5-[03]----00.0  Intel Corporation I210 Gigabit Network Connection [8086:1533]
           +-1f.0  Intel Corporation Sunrise Point-H LPC Controller [8086:a149]
           +-1f.2  Intel Corporation Sunrise Point-H PMC [8086:a121]
           +-1f.3  Intel Corporation Sunrise Point-H HD Audio [8086:a170]
           \-1f.4  Intel Corporation Sunrise Point-H SMBus [8086:a123]

#me_cleaner.py factory_p10s-m_ws.rom 
Full image detected
This image does not contains an ME firmware (NR = 0)

#ifdtool -x factory_p10s-m_ws.rom 
File factory_p10s-m_ws.rom is 16777216 bytes
  Flash Region 0 (Flash Descriptor): 00000000 - 00000fff 
  Flash Region 1 (BIOS): 00800000 - 00ffffff 
  Flash Region 2 (Intel ME): 00001000 - 007fffff 
  Flash Region 3 (GbE): 07fff000 - 00000fff (unused)
  Flash Region 4 (Platform Data): 07fff000 - 00000fff (unused)
  Flash Region 5 (Reserved): 07fff000 - 00000fff (unused)
  Flash Region 6 (Reserved): 07fff000 - 00000fff (unused)
  Flash Region 7 (Reserved): 07fff000 - 00000fff (unused)
  Flash Region 8 (EC): 07fff000 - 00000fff (unused)

#me_cleaner.py flashregion_2_intel_me.bin 
ME image detected
Found FPT header at 0x10
Found 15 partition(s)
ME firmware version 4.0.3.75
Found FTPR header: FTPR partition spans from 0xa000 to 0x6a000
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x3e)...
Modules removal in ME v11 or greater is not yet supported
Done! Good luck!

#intelmetool -s
Bad news, you have a `Sunrise Point-H CSME HECI #2` so you have ME hardware on board and it is very difficult to remove, continuing...
RCBA at 0x00000000
MEI not hidden on PCI, checking if visible
MEI found: [8086:a13b] Sunrise Point-H CSME HECI #2

ME Status   : 0x0
ME Status 2 : 0x0

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Reset
ME: Current Operation State : Preboot
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : ROM Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : BEGIN

PCI READ [bc] : 0x000000bc
ME: Extend Feature not present

ME seems okay on this board
ME: failed to become ready
WRITE    [00] : CB: 0x80040007
WRITE    [00] : CB: 0x000002ff
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
WRITE    [00] : CB: 0x80080007
WRITE    [00] : CB: 0x00000203
WRITE    [00] : CB: 0x00000000
ME: failed to become ready
ME: GET FWCAPS message failed
exiting
```

mytbk, 2017-01-13:
Working on:
- Intel Core i5 3320M (Ivy Bridge)
- Lenovo Thinkpad X230
- coreboot
- 4e9fc2e

Using the ME image from google/link, no issues found.

persmule, 2017-01-16:
Working on:

- Intel Celeron 867
- Sandy Bridge
- Samsung Chromebook lumpy
- Coreboot
- 4e9fc2e

MEI device has disappeared from the PCI bus initially, but after programming back from the scheme below the MEI reappears and keeps present. ME is confirmed broken with dmesg:

> [    5.578881] mei_me 0000:00:16.0: wait hw ready failed
> [    5.578944] mei_me 0000:00:16.0: hw_start failed ret = -62
> [    5.579013] mei_me 0000:00:16.0: H_RST is set = 0x80000015
> [    7.594911] mei_me 0000:00:16.0: wait hw ready failed
> [    7.594969] mei_me 0000:00:16.0: hw_start failed ret = -62
> [    7.595058] mei_me 0000:00:16.0: H_RST is set = 0x80000015
> [    9.610945] mei_me 0000:00:16.0: wait hw ready failed
> [    9.611009] mei_me 0000:00:16.0: hw_start failed ret = -62
> [    9.611066] mei_me 0000:00:16.0: reset: reached maximal consecutive resets: disabling the device
> [    9.611128] mei_me 0000:00:16.0: reset failed ret = -19
> [    9.611183] mei_me 0000:00:16.0: link layer initialization failed.
> [    9.611240] mei_me 0000:00:16.0: init hw failure.
> [    9.611449] mei_me 0000:00:16.0: initialization failed.

and 
`#intelmetool -sd`

> Bad news, you have a `HM65 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...
> 
> MEI not hidden on PCI, checking if visible
> MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI ~�3
> 
> ME Status   : 0x1e003052
> ME Status 2 : 0x16320002
> 
> ME: FW Partition Table      : OK
> ME: Bringup Loader Failure  : NO
> ME: Firmware Init Complete  : NO
> ME: Manufacturing Mode      : YES
> ME: Boot Options Present    : NO
> ME: Update In Progress      : NO
> ME: Current Working State   : Recovery
> ME: Current Operation State : M0 with UMA
> ME: Current Operation Mode  : Normal
> ME: Error Code              : Image Failure
> ME: Progress Phase          : BUP Phase
> ME: Power Management Event  : Pseudo-global reset
> ME: Progress Phase State    : M0 kernel load
> 
> PCI READ [bc] : 0x000000bc
> ME: Extend SHA-256: da64ab79be833b0909c8a5d4214ba413744331d5e50f41bf3927fc53c0bf3b9b
> 
> ME: has a broken implementation on your board with this BIOS
> ME: failed to become ready
> WRITE    [00] : CB: 0x80040007
> WRITE    [00] : CB: 0x000002ff
> ME: failed to become ready
> ME: GET FW VERSION message failed
> ME: failed to become ready
> WRITE    [00] : CB: 0x80080007
> WRITE    [00] : CB: 0x00000203
> WRITE    [00] : CB: 0x00000000
> ME: failed to become ready
> ME: GET FWCAPS message failed

persmule, 2017-01-16:
Working on:

- Intel Core i5 3320M (Ivy Bridge)
- Lenovo Thinkpad X230
- coreboot
- 4e9fc2e
- 1.5MiB Management Engine from Samsung lumpy

MEI device has disappeared from the PCI bus initially, but after programming back from the scheme below the MEI reappears and keeps present. ME is confirmed broken with dmesg:

> [    8.258125] mei_me 0000:00:16.0: wait hw ready failed
> [    8.258178] mei_me 0000:00:16.0: hw_start failed ret = -62
> [    8.258232] mei_me 0000:00:16.0: H_RST is set = 0x80000015
> [   10.274168] mei_me 0000:00:16.0: wait hw ready failed
> [   10.274217] mei_me 0000:00:16.0: hw_start failed ret = -62
> [   10.274268] mei_me 0000:00:16.0: H_RST is set = 0x80000015
> [   12.290189] mei_me 0000:00:16.0: wait hw ready failed
> [   12.290238] mei_me 0000:00:16.0: hw_start failed ret = -62
> [   12.290271] mei_me 0000:00:16.0: reset: reached maximal consecutive resets: disabling the device
> [   12.290318] mei_me 0000:00:16.0: reset failed ret = -19
> [   12.290347] mei_me 0000:00:16.0: link layer initialization failed.
> [   12.290382] mei_me 0000:00:16.0: init hw failure.
> [   12.290544] mei_me 0000:00:16.0: initialization failed.

Streetwalrus, 2017-01-16:
Success
- Ivy Bridge
- 3770k
- Asus P8Z77-V Pro (Panther Point, desktop)
- OEM BIOS
- 4e9fc2e

I tried patching the OEM firmware update and flashing it with the [flashback feature](http://www.win-raid.com/t455f16-Guide-How-to-flash-a-modded-ASUS-ASRock-Gigabyte-AMI-UEFI-BIOS.html), but that apparently didn't work. Ended up figuring out why flashrom wouldn't work (had to boot with `iomem=relaxed`), dumped, patched and flashed from userspace successfully. Not sure if flashback could have unprotected the ME section or if it's just not protected by default on my board.

No more `/dev/mei0`, no more ME in `lspci`, system appears to work properly.

nroach44, 2017-01-24:
* Sandy Bridge
* i5-2520M
* Thinkpad X220
* Coreboot
* 89bbaf27c2a4f47178355b3cee0e7c16a8616874 (42ba8e84abff5806c79e71f997a307fbeaa2273d)

All good (but it was run over an image that had previously been me_cleaned) (Not been up for 30 minutes yet though)
````
ME: FWS2: 0x164e0002
ME:  Bist in progress: 0x0
ME:  ICC Status      : 0x1
ME:  Invoke MEBx     : 0x0
ME:  CPU replaced    : 0x0
ME:  MBP ready       : 0x0
ME:  MFS failure     : 0x0
ME:  Warm reset req  : 0x0
ME:  CPU repl valid  : 0x0
ME:  (Reserved)      : 0x0
ME:  FW update req   : 0x0
ME:  (Reserved)      : 0x0
ME:  Current state   : 0x4e
ME:  Current PM event: 0x6
ME:  Progress code   : 0x1
Waited long enough, or CPU was not replaced, continue...
PASSED! Tell ME that DRAM is ready
ME: FWS2: 0x162c0002
ME:  Bist in progress: 0x0
ME:  ICC Status      : 0x1
ME:  Invoke MEBx     : 0x0
ME:  CPU replaced    : 0x0
ME:  MBP ready       : 0x0
ME:  MFS failure     : 0x0
ME:  Warm reset req  : 0x0
ME:  CPU repl valid  : 0x0
ME:  (Reserved)      : 0x0
ME:  FW update req   : 0x0
ME:  (Reserved)      : 0x0
ME:  Current state   : 0x2c
ME:  Current PM event: 0x6
ME:  Progress code   : 0x1
ME: Requested BIOS Action: Continue to boot
````

persmule, 2017-01-25:
Working on:

- Intel(R) Celeron(R) CPU        U3400 (Nehalem)
- Lenovo Thinkpad X201i
- OEM BIOS (Coreboot's native ram init for x201 failed to apply)
- 89bbaf2

MEI device won't go away and confirmed that ME is broken, and the machine survives beyond 30 minute.

ilikenwf, 2017-01-30:
* x86_64
* Core i7 Skylake 6500U
* Laptop, HP Spectre x360 (2015)
* OEM Bios, stripped
* Master

Stripped and flashed, no issues.

lpereira, 2017-02-07:
Working on:

- Intel Core i7-2640M
- Sandy Bridge
- Lenovo Thinkpad X220
- Coreboot
- 7f21d97

mfc, 2017-03-02:
working on:

* Intel Core i5-3320M
* Ivy Bridge
* Lenovo Thinkpad x230
* Coreboot / Heads
* [`250b2ec`](https://github.com/corna/me_cleaner/commit/250b2ec123d70bb34b7653feacc33c977a25daa1)

H2KChicken, 2017-03-04:
-Intel Core i7-6700k
-Skylake
-Asus Z170 Pro Gaming
-OEM BIOS
[-250b2ec](https://github.com/corna/me_cleaner/commit/250b2ec123d70bb34b7653feacc33c977a25daa1)

Flashed with no issues, survives past 30 minutes without shutting down. "Intel Corporation Sunrise Point-H CSME HECI #1" disappeared from the PCI bus with the modified image. 

The BIOS throws a "Secure Boot Violation" error once flashed, but after pushing "ok" the system boots.

#python me_cleaner.py 'flashregion_2_intel_me.bin' 

ME/TXE image detected
Found FPT header at 0x10
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x1000 to 0xa8000
ME/TXE firmware version 11.0.10.1002
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x0a)...
Modules removal in ME v11 or greater is not yet supported
Checking FTPR RSA signature... VALID
Done! Good luck!

[image]

htruong, 2017-03-05:
Working on:

- Intel(R) Core(TM) i3-5005U
- Broadwell
- Laptop/Dell Chromebook 13
- Coreboot
- me_cleaner on Coreboot head (0da186c3ffb1d9aa7433a5d0d5263aba7a25ad60). This should be equivalent to 250b2ec.

corna, 2017-03-06:
Working on:
 * Intel Core i7-3770
 * Ivy Bridge
 * Sapphire Pure Platinum H61
 * Coreboot
 * 250b2ec

gnustomp, 2017-03-13:
Working on:
- Intel Core i5-2500
- Sandy Bridge
- ASUS P8Z77-V LX
- OEM BIOS
- 250b2ec

No Secure Boot or overclocking issues.

protozone, 2017-03-14:
How would I use  "me_cleaner.py"  in a external flasher since I can remove my cmos bios chip from my motherboard.?
Sorry for the newby question.

corna, 2017-03-14:
@protozone 
I suggest you to follow the guide [here](https://hardenedlinux.github.io/firmware/2016/11/17/neutralize_ME_firmware_on_sandybridge_and_ivybridge.html) to understand better the procedure
If you can extract the flash chip from your motherboard the procedure is mostly the same, you just need a couple of pullup resistors.

**Be careful, these procedures have a high risk of bricking.**

Please don't ask questions in this thread, it is here for success reporting only. I'll delete your and my message shortly.

protozone, 2017-03-14:
Where do I ask questions? I have flashed many chips with my burners.

corna, 2017-03-14:
Open a new issue, we can talk there

protozone, 2017-03-14:
"*Be careful, these procedures have a high risk of bricking."*
* I do mostly server boards.*

*Removing the chip from the motherboard and installing it in a burner gives
you the option of backing up the original and burning the mod. No bricking
yet.*

*I gather that your script can't be used in a burner. Unfortunate.*

protozone, 2017-03-14:
P.S. Some tools
MMTool.EXE
AMITool.EXE
IASL.EXE
PhoenixTool.EXE

flacks, 2017-03-16:
Working on:

- Intel Core i5-3470
- Ivy Bridge
- Dell Inspiron 660s
- OEM BIOS
- 250b2ec

No issues. MEI not showing in lspci. This machine's particular motherboard has a jumper setting labelled `ME disable`. When set, all flash regions become unlocked, so I was able to write to the IC without an external programmer.

protozone, 2017-03-17:
It would be nice to know the trace that the jumper turns on/off. -:)

crat0z, 2017-03-18:
@Kokokokoka, sorry to ask but have you got TPM/VT-d/SLAT on your X220 with Qubes installed and the new ME firmware? I have a T420s which has full Qubes hardware support that I want to try this on, but I still want full hardware support (which on Qubes requires TXT I believe..).

[458 more replies]