corna/me_cleaner #4: Effect on Acer Aspire S3-371

persmule, 2016-11-30:
Seems working on:

    Intel Core i3-2377M
    Sandy Bridge
    Acer Aspire S3-371
    OEM BIOS
    d1abbca

MEI device has NOT disappeared on this platform, but seems to have been neutralized according to intelmetool's output.
[superiotool.txt](https://github.com/corna/me_cleaner/files/620872/superiotool.txt)
[lspci.txt](https://github.com/corna/me_cleaner/files/620870/lspci.txt)
[dmidecode.txt](https://github.com/corna/me_cleaner/files/620869/dmidecode.txt)
[intelmetool_deme.txt](https://github.com/corna/me_cleaner/files/620871/intelmetool_deme.txt)

corna, 2016-11-30:
Your reports of ME on for OEM BIOS are very strange.

**1.**
From hardenedlinux: what do you mean with

> The NIC doesn’t work after a cold boot (it cannot even be recognized as an NIC), but does after a warm boot. (It may be possible to add some code to Coreboot to work around this, but it has not yet been done.)

I had problems with me_cleaner only once after applying the modification, after a complete power cycle (power off and power on, not just reboot) everything worked perfectly (including the NIC).

**2.**
I have no idea about this Acer: maybe the i3 line behaves differently, but it indeed appears that ME is broken on this device

> Intel ME reset due to exception

I am going to upgrade this tool with an additional cleaning step, we can try again and see if the MEI disappears

corna, 2016-11-30:
(to keep #3 clean I removed your post there)

persmule, 2016-11-30:
body p { margin-bottom: 0cm; margin-top: 0pt; } 
  
  
    In reality, such NIC behavior remains
      even after I successfully flashed coreboot on by x220, of course,
      with me neutralized.
      
      在 2016年12月01日 00:09, Nicola Corna 写道:
    
    
      Your reports of ME on for OEM BIOS are very strange.
      1.
        From hardenedlinux: what do you mean with
      
        The NIC doesn’t work after a cold boot (it cannot even be
          recognized as an NIC), but does after a warm boot. (It may be
          possible to add some code to Coreboot to work around this, but
          it has not yet been done.)
      
      I had problems with me_cleaner only once after applying the
        modification, after a complete power cycle (power off and power
        on, not just reboot) everything worked perfectly (including the
        NIC).
      
    
    
    The Acer's PCH is HM77. Unlike QM67, it is designed for IvB
    originally, which I believe is the main difference.
    
    在 2016年12月01日 00:09, Nicola Corna 写道:
    
      2.
        I have no idea about this Acer: maybe the i3 line behaves
        differently, but it indeed appears that ME is broken on this
        device
      
        Intel ME reset due to exception
      
      I am going to upgrade this tool with an additional cleaning
        step, we can try again and see if the MEI disappears

corna, 2017-01-08:
I'm glad to see that the NIC problem has disappeared with 4e9fc2e; is the MEI still visible on the PCI bus also with 4e9fc2e?

closed 2017-11-25 by corna