corna/me_cleaner #5: What about having zero ME partitions?

zamaudio, 2016-12-02:
Has anyone tried using a ME image with zero partitions and fixing the entries/checksum?
ie, something like this:
```
$ hd flashregion_2_intel_me.bin
00000000  20 20 80 0f 40 00 00 10  00 00 00 00 00 00 00 00  |  ..@...........|
00000010  24 46 50 54 00 00 00 00  20 10 30 ed 07 00 64 00  |$FPT.... .0...d.|
00000020  20 00 00 00 01 fc ff ff  00 00 00 00 00 00 00 00  | ...............|
00000030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
004fd000
```

zamaudio, 2016-12-02:
I would try it myself on my X220 but I don't have a flash tool available right now to recover if it doesn't work.

corna, 2016-12-02:
Unfortunately it doesn't work, as ME seems to require the FTPR (Factory partition and recovery)
See [here](https://www.coreboot.org/pipermail/coreboot/2016-November/082331.html)

closed 2016-12-02 by corna