corna/me_cleaner #7: Removing huffman modules

zamaudio, 2016-12-07:
Hi, I have written a patch that removes all huffman modules in the LLUT table by setting the flag for all huffman chunks to 0, meaning that the chunks get interpreted as already uncompressed data.  Then I wiped the actual data.  This might work if the hash of each binary huffman module is not verified against its corresponding hash in the module header.  It seems the LZMA module hashes are not verified so perhaps the huffman ones are not verified too.  Anyway, this is worth a shot... 

Please test https://github.com/zamaudio/me_cleaner/tree/experiments_remove_huffman

platomav, 2016-12-10:
It does not work at Panther Point ME8. Also, there is a bug in which a lot of zeroes/padding are appended to the end of the firmware (~2MB --> ~32MB). I first removed the excess and then tested your script's resulting image.

corna, 2016-12-12:
I think that the problem with your solution is that, now that they're uncompressed, each chunk overlaps with the following one. We should set all of them as uncompressed, re-align the offsets (each one should be the previous one plus the chunk size, usually 1024 bytes in my image), and set to EMPTY the extra ones (the ones that falls into the following module).
I'll try to implement it

skochinsky, 2016-12-28:
Both the LZMA and Huffman modules *are* verified after decompression (their hashes are stored in the signed manifest header), however if the decompression or hash check fails, they're just skipped (unless it's a critical module like BUP or KERNEL in  which case the boot fails).

corna, 2017-01-03:
I'm experimenting with the removal of Huffman modules and I'm now able to remove most of them.

The only module really needed seems to be BUP (and maybe ROMP, if present), KERNEL is not necessary.
If I try to remove BUP the system still loads, but goes into the 30-min window, like when no ME image is provided.

platomav, 2017-01-06:
Just tested the latest 4e9fc2e commit at ME8 1.5MB, works just fine past the 30 minute window. For testing purposes, I removed either ROMP or BUP from the blacklist but they both resulted in an unbootable system. So these two are indeed essential. The ME seems to get stuck at KERNEL loading.

[image]
[image]

corna, 2017-01-07:
Done in 4e9fc2e, I don't think BUP or ROMP are removable, so this is the best we can do

closed 2017-01-07 by corna

zamaudio, 2017-01-08:
This does not remove the KERNEL etc huffman modules completely, if you use unhuffme on the resulting images there is still tons of data left in the modules... please fix.

corna, 2017-01-09:
Which kind of data is your unhuffme showing? On my images the uncompressed data is just a bunch of 1k blocks filled with 0x08 or 0xd0

zamaudio, 2017-01-10:
I get some of those too, but I get valid data blocks as well, almost like some of the blocks are still intact.

corna, 2017-01-10:
On which modules does this happen? On which platform?

zamaudio, 2017-01-10:
Eg, sandy/ivy with ME 7, the KERNEL module as well as the FTCS and POLICY.

corna, 2017-01-17:
Got it
The Huffman chunks are not always in a linear order, but sometimes they are fragmented through the Huffman stream and sometimes they are used by more than one module. I've pushed 42ba8e8 to wipe the chunks one by one and save only the essential ones.

Can you try again with the latest version? If some data remains in the removable modules it means that the same chunk is used by BUP/ROMP and can't be removed, but it doesn't matter as long as the hashes are invalidated: run `unhuffme` and check that every removable Huffman module is `incomplete` (but ROMP and BUP are `[MATCH]`)

platomav, 2017-01-18:
Commit 42ba8e8 works just fine after the 30 minute mark at ME8 1.5MB, same as shown [here](https://github.com/corna/me_cleaner/issues/7#issuecomment-270801399). You can set it as working.

For ME6 Ignition, you should always set is as "WORKS" at the wiki since it can work without a ME region. It's the only PCH system which allows that. Any further commits regarding FTPR module removal, such as Huffman ones, do not change that fact.

Also, why is ME6 1.5/5MB always set as "Not Working"?

corna, 2017-01-18:
Updated, thank you

> Also, why is ME6 1.5/5MB always set as "Not Working"?

Because we tried an early version of me_cleaner (which just removed some partitions) on a Thinkpad X201 and it didn't work. Unfortunately unbricking that laptop is not easy and we don't have yet the right tools.
Do you have any test with ME6?

platomav, 2017-01-18:
Not with ME6 1.5/5MB, only Ignition. I doubt that me_cleaner works at everything else and not ME6. Maybe that system had TPM activated or the early me_cleaner removed modules without fixing the checksum or it had trouble removing the last region because it had no size back then (ME2-ME6) etc. Lots of variables. I believe it should work but maybe that system was a special case. Maybe people are not testing their (recoverable) ME6 systems because they assume it will definitely not work based on the wiki. Just an assumption.

corna, 2017-01-18:
You're right, I'll add a comment

zamaudio, 2017-01-19:
Works with this full huffman wiper (except BUP of course).

zamaudio, 2017-01-19:
Ethernet card works now!