corna/me_cleaner #9: FPT not found

simonepsp, 2016-12-09:
Hi,
I'm trying to deblob the intel ME image of my Thinkpad X230 but the script reports "FPT not found".
I checked with an hex editor and the issue seems related to the $FPT position. The script is looking for "$FPT"on 0x30 but on my Intel ME image it is located on 0x3010...

Fixing the seek to 0x3010 isn't enough 'cause even the FPTR partition is "wrongly" located and the script can't find it.
Thanks for your help

ME Backup rom
[image]
__________________________
[image]

P.S. I'm sure that the dump is correct. I have already double-checked it.

corna, 2016-12-09:
Hi,

I think you're using the full dump, me_cleaner doesn't support yet the full image
Extract the image before with ifdtool -x, then use me_cleaner on the resulting me region

simonepsp, 2016-12-09:
Thanks for pointing me out that :)
Unfortunately I still have a problem with the checksum correction:

> Correcting checksum...
Traceback (most recent call last):
  File "me_cleaner.py", line 84, in <module>
    (sum(checksum_bytes) - checksum_bytes[0x1b]) &
TypeError: unsupported operand type(s) for +: 'int' and 'str'

How can I fix it? Thanks

corna, 2016-12-10:
Try to replace checksum_bytes[0x1b] with unpack("<B", checksum_bytes[0x1b])[0]

simonepsp, 2016-12-10:
same error. I think the issue regards "sum(checksum_bytes)". Probably it can't sum() a string.

[image]

> Correcting checksum...
Traceback (most recent call last):
  File "me_cleaner.py", line 208, in <module>
    (sum(checksum_bytes) - unpack("<B", checksum_bytes[0x1b])[0]) &
TypeError: unsupported operand type(s) for +: 'int' and 'str'

simonepsp, 2016-12-11:
solved. The script must be run with python 3 to work.

corna, 2016-12-12:
Yes, as stated in the [shebang](https://github.com/corna/me_cleaner/blob/master/me_cleaner.py#L1)
I'll add a notice in the README
Please add a comment in #3 if it works on your thinkpad

closed 2016-12-12 by corna