corna/me_cleaner #12: What are the Risks?

jadenPete, 2016-12-12:
What's the worst that could happen? I have a Sandy Bridge Intel i5 2410m and I don't really want to blow off firmware for the ME without knowing the possible consequences.

corna, 2016-12-12:
If everything works correctly (and Sandy Bridge has been widely tested) the MEI device should disappear from the PCI bus and the system should work exactly as before. Rarely the NIC stops working (see #4).
If me_cleaner does not work on your device, the PC bricks (without the 30 mins window, it doesn't turn on at all) and an external programmer is required to flash back the original image.

corna, 2016-12-12:
Since an external flasher is required anyways to apply the modifications in the first place (as the BIOS locks the ME region when the PC is turned on), the worst thing that can happen is that you have to flash the ROM again to restore the original firmware.

zamaudio, 2016-12-12:
Make sure you BACK UP your factory rom before you do anything!  (aka Don't try this at home, kids!)

corna, 2016-12-12:
Right, thanks for pointing it out

platomav, 2016-12-12:
Crippling it in such a way does not come without consequences. Even if the 30-minute power off does not kick in (it shouldn't as the code's current state as proven by various people), the ME deals with Platform Clocks Control (ICC), Thermal Monitoring, Fan Control, Power Management, Overclocking, Boot Guard, Silicon Workaround (resolves silicon bugs which would have otherwise required a new cpu stepping), Anti-Theft Technology (EOL 01/2015), Identity Protection Technology, Rapid Start Technology, Smart Connect Technology, Sensor Hub Controller (ISHC), Active Management Technology (AMT), Small Business Advantage (SBA), Wireless Display, Protected Video/Audio Path etc. Especially the first 7 things are important for some people.

Also, the 30-minute shut down can be activated due to Intel Anti-Theft being enabled by the manufacturer which may be triggered after modifying the ME in such a way or even by itself nowadays because it is defunct and the OEMs have not bothered to remove it (usually the can't because the FD is locked with no software-field way to unlock and reflash a ME region with AT disabled completely). So for some cases, the 30-minute shutdown can be caused by AT as well as a side-effect of me_cleaner.

corna, 2016-12-12:
True, but if someone is concerned with ME it has probably an open source BIOS and/or OS, therefore most of those features are not used (or reimplemented in software). In my opinion the only real useful feature is the silicon workaround but currently NIC problems (#4) are the only effects (not confirmed yet).

I'm not sure about Intel AT: according to Igor Skochinsky in his [2014 presentation](https://recon.cx/2014/slides/Recon%202014%20Skochinsky.pdf) (slide 17), the code that manages Intel AT is inside the TDT module (LZMA compressed, inside the FTPR partition), removed by this script, therefore I suppose that AT shouldn't trigger under any condition (and it can probably be bypassed if previously activated, this should be investigated).

But I started working on Intel ME just a month ago, so feel free to correct me if I'm wrong.

platomav, 2016-12-12:
Yes, certainly. My point was that such a modification should not be done by anyone with any system thinking that he/she will a) be safer and b) have a fully working system. In the future it may be possible to find workarounds for most of the above features with open source BIOS (apart from silicon workaround as you pointed out as well) but for now there are certainly consequences, some more important than others. Some less knowledgeable (on the subject) people with OEM BIOS may think that there are no issues and that is not true, that's all. :)

Yes, AT is removed by me_cleaner completely now. It's configuration is kept at the DATA/EFFS section so if that's not removed as well, it's very likely that AT will kick in and assume the system is compromised. So maybe it is ok if the AT module + EFFS region are removed. But I don't know if something is kept at the ME itself or if it had some other way to detect that the firmware got reflashed. Based on cases in which I've helped people stop the 30 minute timer due to AT kicking in (corrupted EFFS settings) and then reflashed the ME region with AT disabled (FD+EFFS), the current script should not cause any AT issue. But I agree, this needs more investigating. It would be a shame to clean something more from the ME only to rollback on that good/working commit because the 30 minute problem was caused by AT without the researcher's knowledge.

jadenPete, 2016-12-15:
Thank you for the information. I will back off for now, as stated I can't afford bricking my computer.

closed 2016-12-15 by jadenPete