corna/me_cleaner #13: Not working on ThinkPad x131e (Ivy Bridge)

Nephiel, 2016-12-18:
**Laptop/motherboard:** ThinkPad x131e (Intel Core i3-3227U, Ivy Bridge)
**BIOS:** OEM BIOS 2.92 (latest version from manufacturer's website)
**Issue:** After flashing me_cleaned BIOS, laptop doesn't boot. Fan starts spinning and power LED lights up as normal, but the screen stays off. No network activity. Power button has to be held down to turn off the laptop. After restoring the BIOS backup, the laptop boots normally again.

I tried the following with the same results:
- ffe60d8 (latest as of this writing)
- 61fd606
- Removing all power (battery, AC, RTC button cell) after flashing for a few minutes.
- Booting with an external monitor.

**Notes:**
- OEM BIOS has an option to set Intel AT and Computrace to _Permanently Disabled_, and this was already set before the attempt to neutralize IME.

**Hardware notes:**
- This laptop has 2 EEPROM ICs under the bottom cover, right next to each other: 
  - EON EN25Q32B (4MB), sits under the RTC button cell, which is stuck to it with double-sided foam tape. The contents are not recognized by ifdtool.
  - EON EN25Q64 (8MB), sits under a plastic strut from the bottom chassis. The strut is in the way for a SOIC clip. I decided to snip it off.  
  ifdtool only recognizes the contents of the 8MB EEPROM so I ignored the 4MB one.
- The flasher was a Raspberry Pi 2 Model B with a SOIC clip. Wake-on-LAN method did not power the EEPROM. I had to supply 3.3V from GPIO pin 17 on the Raspberry, but also leave the battery in - otherwise flashrom did not see the IC.

Nephiel, 2016-12-18:
Attaching logs:
[ifdtool-dump.txt](https://github.com/corna/me_cleaner/files/659744/ifdtool-dump.txt)
[me_cleaner-output.txt](https://github.com/corna/me_cleaner/files/659745/me_cleaner-output.txt)

corna, 2016-12-31:
The two chips are concatenated by software (if you look at the ifdtool output you can see that the image is 12MB), like in the [X230](https://www.coreboot.org/Board:lenovo/x230#Hardware_Flashring). Try to dump bot of them, concatenate them (8MB first, then 4MB) and re-run ifdtool and me_cleaner

Nephiel, 2017-01-01:
I ran **ifdtool -d** and latest **me_cleaner** with the concatenated 12MB file and compared the results:
- The only difference ifdtool reports is the file size.
- The first 8MB of the 12MB file is identical to the result of running me_cleaner with the 8MB file.
- The last 4MB of the file stays unchanged (identical to the original dump).

Just in case, I also tried concatenating 4MB first then 8MB: the resulting file is not readable by ifdtool (_Unknown descriptor version: 7_) and me_cleaner reports _Unknown image_.

Nephiel, 2017-02-19:
Tried again with 7f21d974. Unhuffme output seems correct. Same result: not booting, black screen.  
&nbsp;  
If I unplug the CMOS battery before powering on, the screen works, BIOS shows an expected wrong date/time error message, and from there I can press F1 to enter BIOS settings. However, after that it won't boot again until I flash back the unmodified dump.

gnustomp, 2019-05-22:
Hi,

Sorry the hijack this issue. I'm working on Coreboot for X131e, and it would really help if I could get BIOS dumps of different X131e laptops.

@Nephiel do you still have the dumps (both 8M and 4M) you made? Thanks.

Nephiel, 2019-05-25:
@gnustomp they should be on one of my old drives, hopefully not among the ones I lost back when I moved... I'll let you know if I find them.