corna/me_cleaner #14: With 0xff wiping, screen goes blank after a while

zamaudio, 2016-12-27:
I noticed that after a while of use on my sandybridge board, the screen goes a random colour while in X11 and can no longer reset the display.  Needs to hard shutdown.
I seemed to fix it by wiping the firmware modules with 0x00 instead of 0xff.  This seems to do the trick.
I suggest changing 0xff -> 0x00.

Kokokokoka, 2016-12-28:
I've got the same issue on T420 with seabios and windows. I thought that the reason for that was the native graphics init, but didn't managed to test this. The interesting thing is that when there is a background
game this happens less often.

corna, 2016-12-31:
I also thought that it was related to my native initialization, and I don't remember if it ever happened before me_cleaner, however it seems unlikely that the cause is Intel ME. I'll test it.

@Kokokokoka you can just replace `b"\xff"` with `b"\x00"` on [line 183](https://github.com/corna/me_cleaner/blob/master/me_cleaner.py#L183) if you want to try

corna, 2017-02-07:
If you discover this problem is caused by me_cleaner (and not by the native graphics init), feel free to reopen this issue.

closed 2017-02-07 by corna

Kokokokoka, 2017-02-07:
I would like to say that this issue still exists, even without NGI. But I've found out by adding hybrid to t420 KCONFIG that it is likely (hopefully) intel/vbios issue as nv gpu is still working now (1 day uptime) and intel gpu
crashes rather fast. I've been using 2170 from the coreboot wiki, maybe it's the cause. But I don't know 
which vbios should be used on the  i7 3632-qm (as it has ivy graphics), is sandy vbios enough or should one use x*3* ones? still, I don't have spare ivybridge vbios to test, if anyone could share vbios, I'll gladly test this. As it really looks suspicious.