corna/me_cleaner #15: me_cleaner doesn't work on Alienware 13 R2

citypw, 2017-01-03:
I failed to dump the ROM via flashrom but chipsec works for dump:
`sudo chipsec_util spi dump rom.bin

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.2.5
****** Chipsec Linux Kernel module is licensed under GPL 2.0
[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] Executing command 'spi' with args ['dump', 'rom.bin']

[CHIPSEC] dumping entire SPI flash memory to 'rom.bin'
[CHIPSEC] it may take a few minutes (use DEBUG or VERBOSE logger options to see progress)
[CHIPSEC] BIOS region: base = 0x00800000, limit = 0x00FFFFFF
[CHIPSEC] dumping 0x01000000 bytes (to the end of BIOS region)
[spi] reading 0x1000000 bytes from SPI at FLA = 0x0 (in 262144 0x40-byte chunks + 0x0-byte remainder)
[CHIPSEC] completed SPI flash dump to 'rom.bin'
[CHIPSEC] (spi dump) time elapsed 67.021

shawn@shawn-fortress /tmp/rerun $ ifdtool -x rom.bin 
File rom.bin is 16777216 bytes
  Flash Region 0 (Flash Descriptor): 00000000 - 00000fff 
  Flash Region 1 (BIOS): 00800000 - 00ffffff 
  Flash Region 2 (Intel ME): 00001000 - 007fffff 
  Flash Region 3 (GbE): 07fff000 - 00000fff (unused)
  Flash Region 4 (Platform Data): 07fff000 - 00000fff (unused)
  Flash Region 5 (Reserved): 07fff000 - 00000fff (unused)
  Flash Region 6 (Reserved): 07fff000 - 00000fff (unused)
  Flash Region 7 (Reserved): 07fff000 - 00000fff (unused)
  Flash Region 8 (EC): 07fff000 - 00000fff (unused)
shawn@shawn-fortress /tmp/rerun $ me_cleaner.py flashregion_2_intel_me.bin 
Unknown image
`

I uploaded the ROM into:
https://github.com/hardenedlinux/hardenedlinux_profiles/raw/master/coreboot/shitty_me/rom.bin

citypw, 2017-01-03:
This is not the issue of me_cleaner but wrong data:

hexdump flashregion_2_intel_me.bin 
0000000 ffff ffff ffff ffff ffff ffff ffff ffff
*
07ff000

Maybe something went wrong with dumping flash via chipsec?

corna, 2017-01-03:
The ROM you uploaded has the ME region locked (like every other OEM ROM):

     $ ifdtool rom.bin -d

     [...]

     FLMSTR1:   0x00a00b00 (Host CPU/BIOS)

     [...]

       Intel ME Region Read Access:       disabled

     [...]

You have to read and write the flash with an external programmer.

You can also unlock the ME region from ifdtool, but you have to use an external programmer to flash the unlocked image.

citypw, 2017-01-03:
Thanks for the reply. An external programmer then......

corna, 2017-01-03:
Unfortunately yes, the vendors usually lock the ME region (as suggested by Intel).

closed 2017-01-03 by corna