corna/me_cleaner #16: Lenovo x230, MEI remain on PCI bus after cleaning ME

dodgydogee, 2017-01-04:
Everything went properly, flashed used is a raspberry pi model b. CPU is i5-3320M, Coreboot bios.

[image]

[image]
[image]

corna, 2017-01-04:
On some boards the MEI interface doesn't disappear (like [this one](https://github.com/corna/me_cleaner/issues/3#issuecomment-268167874) or [this one](https://github.com/corna/me_cleaner/issues/3#issuecomment-269930103)), can you run

     # intelmetool -s

You can get it from the [coreboot repo](https://github.com/coreboot/coreboot), `util/intelmetool`

bayo400, 2017-01-04:
Output from running `intelmetool -s` on x230
```
# ./intelmetool -s
Bad news, you have a QM77 Express Chipset LPC Controller so you have ME hardware on board and you can't control or disable it, continuing...

MEI device not found
```

corna, 2017-01-07:
Strange
Can you post the output of `sudo lspci -v`?
Can you also try with the latest version of me_cleaner?

dodgydogee, 2017-01-07:
```
$ sudo lspci -v

00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev ff) (prog-if ff)
	!!! Unknown header type 7f
	Kernel modules: mei_me
```

I will try again with the latest version of me_cleaner when i have some time later today.

dodgydogee, 2017-01-09:
result with using latest version of me_cleaner

```
$ sudo ./intelmetool -s
Bad news, you have a `QM77 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

Error mapping physical memory 0x0000004275159040 [0x4000] ERRNO=1
Segmentation fault
```

```
$ sudo lspci -v
00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev ff) (prog-if ff)
	!!! Unknown header type 7f
	Kernel modules: mei_me

```

dodgydogee, 2017-01-09:
after a power cycle
```

$ sudo ./intelmetool -s
Bad news, you have a `QM77 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI device not found

```

dodgydogee, 2017-01-09:
for no apparent reason, after a few more power cycles, it changes to this.

```
$ sudo lspci -v
00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev 04)
	Subsystem: Lenovo Device 21fa
	Flags: fast devsel
	Memory at f1641000 (64-bit, non-prefetchable) [size=16]
	Capabilities: [50] Power Management version 3
	Capabilities: [8c] MSI: Enable- Count=1/1 Maskable- 64bit+
	Kernel modules: mei_me
```

robcross, 2017-01-13:
@dodgydogee how have things been working since then? I'm thinking about attempting this on my x230.

dodgydogee, 2017-01-13:
@robcross no change, MEI still appears on PCI bus, but the ME parts of the rom does seem to been mostly overwritten by 0xff

step21, 2017-01-14:
So out of curiosity ... you flashed from Virtualbox?

dodgydogee, 2017-01-14:
@step21 no 

>  flasher used is a raspberry pi model b

corna, 2017-01-17:
Can you try with the latest revision and post the output of unhuffme (follow [this](https://github.com/corna/me_cleaner/wiki/How-does-it-work%3F#hashes))?

Kokokokoka, 2017-01-17:
>Can you try with the latest revision and post the output of unhuffme (follow this)?
Sorry to interrupt, but I've done this with the latest image, and have those results:
```
Code partition: FTPR(9 modules, v7.1.40.1161)
                 0           UPDATE     lzma at:   441b1 va:2003d000+2000
                 1              BUP  huffman at:     780 va:20040000+11000
                 2           KERNEL  huffman at:     780 va:20055000+29000
                 3           POLICY  huffman at:     780 va:20089000+1b000
                 4         HOSTCOMM     lzma at:   44243 va:200d3000+b000
                 5              RSA     lzma at:   49804 va:200e8000+c000
                 6              CLS     lzma at:   4e2ab va:200f7000+a000
                 7              TDT     lzma at:   52cb2 va:20108000+e000
                 8             FTCS  huffman at:     780 va:200a6000+7000

        UPDATE  13f1e1e6479e383099dcf7bb2db126b55d1d64c2daa082cbe138d433940ec97b lzma
           BUP  9d775c4a37e4178add05779e37a26333812ae08727c85aa77cd4cd220375916b [MATCH]
        KERNEL  99e8816c59cd7db70024059ec203fcc1624d7225daa6cbc5052ccbd3371571a7 incomplete
        POLICY  1dfe3976be5bff7f6fee481ae316618b4e1ec409f44bac5a575b4c62824588d6 incomplete
      HOSTCOMM  f7d7f07d5531fd87b118ae5734a763bb7b46a20a9a8084adf31cc25c3c569e31 lzma
           RSA  27be8e5381487255ab23be946c023f56655260910d2fc1c10c926d6c7c18cbea lzma
           CLS  20e7855c48e7b625e8f7896d8e98fc8acc02a8d82e2f7a06db6b61a7b5ff1a8c lzma
           TDT  61695c72342b75b62912d8bcbe27e16ac5a17211b62e9afa9d9eef63d0cfc489 lzma
          FTCS  285db9a45a5259b35861280b8b7e93e68679369a8c92d9750772e650ff157566 incomplete

```

and the previous one:
```
Flash partition table (1 entries):
        partition:  FTPR(type:0) at 00037000, size:6a000

Code partition: FTPR(9 modules, v7.1.40.1161)
                 0           UPDATE     lzma at:   441b1 va:2003d000+2000
                 1              BUP  huffman at:     780 va:20040000+11000
                 2           KERNEL  huffman at:     780 va:20055000+29000
                 3           POLICY  huffman at:     780 va:20089000+1b000
                 4         HOSTCOMM     lzma at:   44243 va:200d3000+b000
                 5              RSA     lzma at:   49804 va:200e8000+c000
                 6              CLS     lzma at:   4e2ab va:200f7000+a000
                 7              TDT     lzma at:   52cb2 va:20108000+e000
                 8             FTCS  huffman at:     780 va:200a6000+7000

        UPDATE  13f1e1e6479e383099dcf7bb2db126b55d1d64c2daa082cbe138d433940ec97b lzma
           BUP  9d775c4a37e4178add05779e37a26333812ae08727c85aa77cd4cd220375916b [MATCH]
        KERNEL  99e8816c59cd7db70024059ec203fcc1624d7225daa6cbc5052ccbd3371571a7 incomplete
        POLICY  1dfe3976be5bff7f6fee481ae316618b4e1ec409f44bac5a575b4c62824588d6 incomplete
      HOSTCOMM  f7d7f07d5531fd87b118ae5734a763bb7b46a20a9a8084adf31cc25c3c569e31 lzma
           RSA  27be8e5381487255ab23be946c023f56655260910d2fc1c10c926d6c7c18cbea lzma
           CLS  20e7855c48e7b625e8f7896d8e98fc8acc02a8d82e2f7a06db6b61a7b5ff1a8c lzma
           TDT  61695c72342b75b62912d8bcbe27e16ac5a17211b62e9afa9d9eef63d0cfc489 lzma
          FTCS  285db9a45a5259b35861280b8b7e93e68679369a8c92d9750772e650ff157566 incomplete

```

corna, 2017-01-17:
Can you open the mod/*.lzma files and tell me if they contain some data or just 0xff?

Kokokokoka, 2017-01-17:
old:
`(.synapse)[root@edev-build mod]# od -xc CLS-200f7000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0045000    ffff    ffff    ffff    00ff
        377 377 377 377 377 377 377
0045007
(.synapse)[root@edev-build mod]# od -xc HOSTCOMM-200d3000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0052700    00ff
        377
0052701
(.synapse)[root@edev-build mod]# od -xc FTCS-200a8000.mod
0000000    d0d0    d0d0    d0d0    d0d0    d0d0    d0d0    d0d0    d0d0
        320 320 320 320 320 320 320 320 320 320 320 320 320 320 320 320
*
0002000    0808    0808    0808    0808    0808    0808    0808    0808
         \b  \b  \b  \b  \b  \b  \b  \b  \b  \b  \b  \b  \b  \b  \b  \b
*
0044000    d0d0    d0d0    d0d0    d0d0    d0d0    d0d0    d0d0    d0d0
        320 320 320 320 320 320 320 320 320 320 320 320 320 320 320 320
*
0050000
(.synapse)[root@edev-build mod]# od -xc RSA-200e8000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0045240    ffff    ffff    ffff    00ff
        377 377 377 377 377 377 377
0045247
(.synapse)[root@edev-build mod]# od -xc TDT-20108000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0060640    ffff    ffff    ffff    ffff    ffff    ffff    ffff    00ff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
0060657
(.synapse)[root@edev-build mod]# od -xc UPDATE-2003d000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0000220    ffff
        377 377
0000222
`
new:

`(.synapse)[root@edev-build mod]# od -xc CLS-200f7000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0045000    ffff    ffff    ffff    00ff
        377 377 377 377 377 377 377
0045007
(.synapse)[root@edev-build mod]# od -xc HOSTCOMM-200d3000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0052700    00ff
        377
0052701
(.synapse)[root@edev-build mod]# od -xc RSA-200e8000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0045240    ffff    ffff    ffff    00ff
        377 377 377 377 377 377 377
0045247
(.synapse)[root@edev-build mod]# od -xc TDT-20108000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0060640    ffff    ffff    ffff    ffff    ffff    ffff    ffff    00ff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
0060657
(.synapse)[root@edev-build mod]# od -xc UPDATE-2003d000.mod.lzma
0000000    ffff    ffff    ffff    ffff    ffff    ffff    ffff    ffff
        377 377 377 377 377 377 377 377 377 377 377 377 377 377 377 377
*
0000220    ffff
        377 377
0000222
`

It seems that they're all empty in the old and in the new release, firstly I've tried to open them with mc, but it printed that it's an empty input, and I've found only od in centos, so this is the result and vi finds only one word, so this seems to be an 0xff file.

dodgydogee, 2017-01-17:
New image

```
number of flash regions: 4
	region: 0 in range: 00000fff-00000000
	region: 1 in range: 00bfffff-00500000
	region: 2 in range: 004fffff-00003000
	region: 3 in range: 00002fff-00001000

Flash partition table (1 entries):
	partition:  FTPR(type:0) at 00180000, size:ca000

Code partition: FTPR(12 modules, v8.0.12.1498)
		 0           UPDATE     lzma at:   4b8ef va:2003c000+3000
		 1             ROMP  huffman at:     8c0 va:2015e000+2000
		 2              BUP  huffman at:     8c0 va:20161000+13000
		 3           KERNEL  huffman at:     8c0 va:20178000+2e000
		 4           POLICY  huffman at:     8c0 va:201b1000+1e000
		 5         HOSTCOMM     lzma at:   4baad va:2048b000+d000
		 6              RSA     lzma at:   527c7 va:20447000+d000
		 7              CLS     lzma at:   57a2e va:204a4000+c000
		 8              TDT     lzma at:   5d170 va:204bb000+f000
		 9             FTCS  huffman at:     8c0 va:201d1000+8000
		10          ClsPriv     lzma at:   63854 va:20455000+3000
		11          SESSMGR     lzma at:   63c34 va:20459000+2c000

        UPDATE	5edd1876999f8ca7f9e0068fe00d4948b9368e58f42419ee9beb03d124618639 lzma
          ROMP	287487f975688650e073db490efd5975eb33ca3535b74be7e2901b6728b1dc12 [MATCH]
           BUP	e4366b9179809d22ad5d622e86fda5093dbbcd344b7ef04144419cfb9fcd2929 [MATCH]
        KERNEL	e5d1c96c115527c98080727b724c55d81f8ef58482f675b622b62c093dc3ba9b incomplete
        POLICY	8ca79f0f0d6e73ec1e2f2a79a41a857eafba73d6740043047af430481705db83 incomplete
      HOSTCOMM	f48504c671b8a484c9c5e9be6ab3a0bf9b464bd5d4112bf8fab9273329a051ec lzma
           RSA	adf36e2c8811eb58257b9eedb017e4380151973f7f9b3ef6b547063eb688a366 lzma
           CLS	9d3130cd643c6339237d40b8058e99047a94ee427e082a95ccfd6e27c1d484c6 lzma
           TDT	17ad2f93ad8d340e0f608d3a195c3a4eb46e7511839f9ac46c60c7a86520eefb lzma
          FTCS	5b3bb7ee2d4e3c66f6679cfa3ca7f975fed5e0c478328aca9510d431fb1feee0 incomplete
       ClsPriv	ee2106c9978b6fdea7b47817ce0d9d7c8da8113027595a85836b8003f26786f2 lzma
       SESSMGR	f698b0a784d5471b426dfff0800a72cf574c7045529c15d156d94b36cc9eee0c lzma
```

previous image

```
number of flash regions: 4
	region: 0 in range: 00000fff-00000000
	region: 1 in range: 00bfffff-00500000
	region: 2 in range: 004fffff-00003000
	region: 3 in range: 00002fff-00001000

Flash partition table (1 entries):
	partition:  FTPR(type:0) at 00180000, size:ca000

Code partition: FTPR(12 modules, v8.0.12.1498)
		 0           UPDATE     lzma at:   4b8ef va:2003c000+3000
		 1             ROMP  huffman at:     8c0 va:2015e000+2000
		 2              BUP  huffman at:     8c0 va:20161000+13000
		 3           KERNEL  huffman at:     8c0 va:20178000+2e000
		 4           POLICY  huffman at:     8c0 va:201b1000+1e000
		 5         HOSTCOMM     lzma at:   4baad va:2048b000+d000
		 6              RSA     lzma at:   527c7 va:20447000+d000
		 7              CLS     lzma at:   57a2e va:204a4000+c000
		 8              TDT     lzma at:   5d170 va:204bb000+f000
		 9             FTCS  huffman at:     8c0 va:201d1000+8000
		10          ClsPriv     lzma at:   63854 va:20455000+3000
		11          SESSMGR     lzma at:   63c34 va:20459000+2c000

        UPDATE	5edd1876999f8ca7f9e0068fe00d4948b9368e58f42419ee9beb03d124618639 lzma
          ROMP	287487f975688650e073db490efd5975eb33ca3535b74be7e2901b6728b1dc12 [MATCH]
           BUP	e4366b9179809d22ad5d622e86fda5093dbbcd344b7ef04144419cfb9fcd2929 [MATCH]
        KERNEL	e5d1c96c115527c98080727b724c55d81f8ef58482f675b622b62c093dc3ba9b incomplete
        POLICY	8ca79f0f0d6e73ec1e2f2a79a41a857eafba73d6740043047af430481705db83 incomplete
      HOSTCOMM	f48504c671b8a484c9c5e9be6ab3a0bf9b464bd5d4112bf8fab9273329a051ec lzma
           RSA	adf36e2c8811eb58257b9eedb017e4380151973f7f9b3ef6b547063eb688a366 lzma
           CLS	9d3130cd643c6339237d40b8058e99047a94ee427e082a95ccfd6e27c1d484c6 lzma
           TDT	17ad2f93ad8d340e0f608d3a195c3a4eb46e7511839f9ac46c60c7a86520eefb lzma
          FTCS	5b3bb7ee2d4e3c66f6679cfa3ca7f975fed5e0c478328aca9510d431fb1feee0 incomplete
       ClsPriv	ee2106c9978b6fdea7b47817ce0d9d7c8da8113027595a85836b8003f26786f2 lzma
       SESSMGR	f698b0a784d5471b426dfff0800a72cf574c7045529c15d156d94b36cc9eee0c lzma
```

corna, 2017-01-17:
Nah, it's the same
The output of intelmetool is still a segfault?

dodgydogee, 2017-01-18:
```
$ sudo ./intelmetool -s
Bad news, you have a `QM77 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI device not found

```

dodgydogee, 2017-01-18:
right after boot:
```
$ sudo lspci -nnv

00:16.0 0780: 8086:1e3a (rev 04)
	Subsystem: 17aa:21fa
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Region 0: Memory at f1641000 (64-bit, non-prefetchable) [size=16]
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [8c] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Kernel modules: mei_me

```
after wake from suspend:
```
00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev ff) (prog-if ff)
	!!! Unknown header type 7f
	Kernel modules: mei_me

```

intelmetool with patch applied:

```
sudo ./intelmetool -s
Bad news, you have a `QM77 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

Error mapping physical memory 0x0000004275159040 [0x4000] ERRNO=1
Segmentation fault

```

dodgydogee, 2017-01-23:
Update:
flashed back to stock bios today, and MEI is gone from PCI bus.

piccaruse, 2017-03-22:
Is there any updates to this? Looking forward to use this on my x230 but it seems it doesn't work as intended?

corna, 2017-03-23:
It should work as intended, the only difference with other models is the fact that the MEI interface is still visible, while it often disappear after the use of me_cleaner.

Can you test it and report the status of [intelmetool](https://github.com/coreboot/coreboot/tree/master/util/intelmetool), so that I can close this issue?

nariox, 2017-04-02:
Hi Corna,
not sure if helps, but I was having the same issue:
`Error mapping physical memory 0x0000004275159040 [0x4000] ERRNO=1`
which I fixed by setting `iomem=relaxed` on GRUB. After that intelmetool started working properly

zamaudio, 2017-04-08:
I wrote intelmetool, can you *please* `sudo rmmod mei_me` and `sudo rmmod mei` before using my tool.  It won't work otherwise.

nariox, 2017-04-10:
Hi @zamaudio, seems like the mei and mei_me modules aren't auto-loaded after flashing the new BIOS, however the iomem=relaxed was necessary to avoid the error above (in both my systems). It might be a while before I can reflash the original BIOS and test, but we should probably ask @dodgydogee, so I don't completely misdirect his thread.

closed 2017-05-11 by corna