corna/me_cleaner #18: Issue with me_2.0.5.3105 on Braswell SoC

hermitcrabslab, 2017-01-20:
`c:\Python27>me_cleaner.py me_2.0.5.3105.rgn
ME image detected
Found FPT header at 0x10
Found 20 partition(s)
Found FTPR header: FTPR partition spans from 0x6f000 to 0xe7000
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xed)...
ME firmware version 2.0.5.3105
Reading FTPR modules list...
Wiping LZMA section (0x1fffffffe - 0xe7000)...
Can't find the LLUT region in the FTPR partition
 BUP              (LZMA, 0x070000 - 0x088000): outside the LZMA region (0x1fffffffe - 0xe7000), skipping
                  (LZMA, 0x822523 - 0x19ae09b2): outside the LZMA region (0x1fffffffe - 0xe7000), skipping
Traceback (most recent call last):
  File "C:\Python27\me_cleaner.py", line 299, in <module>
    lzma_removed, huffman_removed)
  File "C:\Python27\me_cleaner.py", line 96, in module_removal_report
    name = mod_header[0x04:0x14].rstrip(b"\x00").decode("ascii")
UnicodeDecodeError: 'ascii' codec can't decode byte 0x90 in position 1: ordinal not in range(128)`

[https://www.dropbox.com/s/dxs1p7kv1en7mo1/me_2.0.5.3105.rar?dl=0](url)

Also size of the .rgn is 8GB after the script stops.

platomav, 2017-01-21:
That is TXE, not ME firmware. The former is not supported by me_cleaner yet. The implementation is very easy though @corna . TXE 1,2 firmware have only LZMA or uncompressed modules (no huffman) and their $MME are 0x80 long. Each module offset is found at 0x38 - 0x3C and it's relative to $MN2, FTPR start. The uncompressed & compressed sizes are found at 0x3C - 0x40 and 0x40 - 0x44 respectively. At uncompressed modules, the two sizes are equal. As far as module necessity is concerned, people should test how TXE reacts. Maybe it can work without TXE firmware at all or (most probably) requires at least FTPR > BUP.

hermitcrabslab, 2017-01-21:
Yes I know is TXE :)
Was more like a feature/request for support then an issue.
Thanks for tips, hope @corna will add support for TXE.

corna, 2017-01-22:
I've finally cleaned the mess in the code for the modules removal and I've added the support for TXE images. Can you test it? It is available in the branch experiments_TXE

hermitcrabslab, 2017-01-22:
Tested today:
Tool log on full ROM: http://pastebin.com/ydKBLgs3
Flashing log: http://pastebin.com/DKn9jLgS
Flashing back full ROM log: http://pastebin.com/PvChLghX
After flash on BIOS: https://www.dropbox.com/s/xb737g69uwfyufu/IMG_20170122_213529.jpg?dl=0
Boot time to Windows increased at least 5x time, TXE still present on PCI Devices.
Hope this report helps, thanks for looking into it!

platomav, 2017-01-23:
@hermitcrabslab 

Can you do 3 more tests? For different reasons/test cases, these might result in non-booting system which will require SPI reflash. Since you went ahead and tested corna's TXE script, I assume you can recover in case of a bad flash, right? If yes, [here](http://www.mediafire.com/file/w12u8crw95uhyiw/txe_tests.rar) are the TXE regions I'd like you to test.

For each test, after the flash is over, shut down the system, remove power (psu cord + psu switch to off + press power button 1-2 times) and wait for 30 seconds. Now check if it boots.

txe_move --> FTPR intact but moved to different offset
txe_bup --> FTPR with all modules removed, BUP included
txe_empty --> Completely empty/padded TXE region

hermitcrabslab, 2017-01-23:
@platomav Sure I can recover from disaster :D 
Just let me get it right, you want me to replace the me/txe region on BIOS with those you uploaded on mediafire then use @corna script on modified BIOS and flash?
Or just replace and flash?

platomav, 2017-01-23:
They are ready. Replace at SPI and flash.

hermitcrabslab, 2017-01-23:
OK will do it but when I get home, expect report in about 12h.

corna, 2017-01-23:
Great, thank you

hermitcrabslab, 2017-01-23:
Sure, my pleasure, this little Braswell Soc will be soon a pfsense box so I want to get rid of any "unnecessary garbage/hole" from it...

platomav, 2017-01-23:
Oups, I did say "using Corna's latest master commit" before, thus your confusion. Fixed it. :)

hermitcrabslab, 2017-01-24:
Sad to report that all 3 tests result in bricked BIOS :(
However there is a BIOS settings "Reset System with ME disable Mode" and clicking that it will reset system and after reset looking on BIOS TXE looks like is disabled(same as with @corna  patched txe) but this time windows boot time is normal.
Looks like Trusted Execution Engine cannot be completely disabled from PCI bus and what does that BIOS setting is to disable the memory space (on quick look)
For reference lspci dump from windows:
with txe enabled: http://pastebin.com/36qwXP2r
with txe disabled:  http://pastebin.com/NbHkQLAz
Now I am looking on a easy way to make that permanent.

platomav, 2017-01-24:
Thanks for the reports. I expected "bup" and "empty" to not work but I was hoping "move" would. However, the first "move" was extremely likely to brick. The [second one](http://www.mediafire.com/file/3oruif1vb703a1f/txe_move_2.rar) might also brick but if we're lucky it won't and you'll be able to save some rom space. Please try it like before, with no special BIOS options enabled or similar.

As for the BIOS option you found, it is most probably not that useful. I assume it sends the ME into Recovery mode via the Host ME Region Flash Protection Override (HMRFPO) command. This enables Flash Descriptor read/write access to the ME region and halts its operation so that it can be repaired. Very useful to have a BIOS option but not really related to what me_cleaner tries to achieve.

hermitcrabslab, 2017-01-24:
@platomav you where right this one posted, but as with the one resulted with @corna script boot time increased very hard.
Device Manager reports:
 This device cannot start. (Code 10)

STATUS_DEVICE_POWER_FAILURE

lspci_txe_move2: http://pastebin.com/sDAhx4jA

platomav, 2017-01-24:
Very interesting. This is what we have learned so far:

1) TXE requires FTPR and its BUP module to be present to give control to the BIOS
2) The FTPR digital signature must be valid to give control to the BIOS
3) The FTPR starting offset at $MCP does not matter *

* If it is also fixed (first "move" test), the FTPR signature breaks > Brick
* If left unfixed, the FTPR signature is valid > OK
* If left unfixed but FTPR moved with $FPT offset adjusted > OK

So, since TXE does not have Huffman tables (offsets at FTPR relative to $FPT so readjustment will break Signature and cause a brick) and the FTPR starting offset at $MCP does not matter, we can indeed move the FTPR up and save flash space for something else. If we move it all the way to the top, the final TXE size will be 480KB instead of the original 1.37MB (your TXE region is 3MB because the OEM left useless at its end).

One question is, how far up can be move FTPR? Maybe $FPT has an undocumented minimum expected size before any partition starts. At your stock TXE region, the first started at 0x3C0. I created a [test file](http://www.mediafire.com/file/48k8c0xmo462idt/txe_cleaned_1.rar) (only FTPR with BUP, moved) which moves it just 0x10 below the end of $FPT, so it starts at offset 0x60. Let's see if it works.

corna, 2017-01-24:
> 3. The FTPR starting offset at $MCP does not matter

Why do you say this?
I tried to move the FTPR with [this branch](https://github.com/corna/me_cleaner/tree/not_working_relocate_ftpr) without success. With this version everything (except for the signed parts) is corrected ($FPT offset, Huffman start offset, Huffman chunks offsets), but still it doesn't work.

> * If left unfixed but FTPR moved with $FPT offset adjusted > OK

How did you move the partition?

corna, 2017-01-24:
> offsets at FTPR relative to $FPT so readjustment will break Signature and cause a brick

What do you mean?

platomav, 2017-01-24:
The $MCP partition start offset does not matter because apparently it's not (double-)checked by the ME after FTPR is found from $FPT. Also, it seems that the $MCP hash is not checked (maybe because it's part of the FTPR signature verification already, why do an extra useless check?). So in the end, not fixing the offset at $MCP does not matter. I am talking about partition/FTPR offset only as I don't know about LUT in ME (this is TXE2 firmware without Huffman code).

For such tests. I move the FTPR partition manually with cut-paste/insert followed by the equivalent $FPT fix.

> offsets at FTPR relative to $FPT so readjustment will break Signature and cause a brick

I think I got confused there with the offsets found at LLUT which are $FPT relative if I remember properly. I am not familiar with Huffman structure/flow yet like you are though.

hermitcrabslab, 2017-01-24:
@platomav txe_cleaned_1 posted, no change on device manager or windows boot time.

platomav, 2017-01-24:
Alright. So for TXE 1 & 2, after me_cleaner is done it can also move the FTPR to 0x50 (exactly after the single $FPT entry) and fix $FPT. No other change is needed. The image you tested had FTPR at 0x60 but it should be ok if we remove these extra 0x10 I added, after $FPT and before FTPR, for visual purposes.

The option "Reset System with ME disable Mode" does indeed what I said. It is used to temporarily disable the TXE and allow read/write access to its region for reflashing/updating. If you search that exact phrase at google you'll find various OEM instructions during BIOS/SPI updates like [this](http://resources.mini-box.com/online/SYS-NUC-HBJC311U93W-2930-B/SYS-NUC-HBJC311U93W-2930-B-manual.pdf), [this](ftp://211.214.88.34/bios/C7/C7P67/1.0b/Read_This_Before_Flash.txt) etc.

Now, what do you mean "not working"? How do you know it does nothing? Regardless, it makes sense. The TXE does not need to do what HMRFPO says since it's already at Recovery/FTPR state after me_cleaner's magic.

hermitcrabslab, 2017-01-24:
"Not Working" means in "normal" way when you select it and press enter it reboots, now it does nothing.

corna, 2017-01-24:
> The 2nd "move" test proved that

I had missed the 2nd move :)
Great!

>> offsets at FTPR relative to $FPT so readjustment will break Signature and cause a brick
>
> I think I got confused there with the offsets found at LLUT which are $FPT relative if I remember properly. 

They are, but they're not signed and we can adjust them as well

> So for TXE 1 & 2, after me_cleaner is done it can also move the FTPR to 0x50 (exactly after the single $FPT entry) and fix $FPT. No other change is needed.

On the ME images we should move the FTPR and:
 * Correct the corresponding $FPT entry
 * Correct the (absolute) Huffman start offset (bytes 0x14:0x18 of the LLUT)
 * Correct each (absolute) offset inside the chunks (bytes 0x00:0x03 of the chunks entries)

I'll check carefully the code inside the [not_working_relocate_ftpr branch](https://github.com/corna/me_cleaner/tree/not_working_relocate_ftpr) to understand what went wrong.

platomav, 2017-01-25:
> Now what I don't get is why with TXE in "recovery mode" on original TXE boot to windows is normal and with hacked TXE is slow?

What part is slow? Before or after the OS starts to load?

> I'll check carefully the code inside the not_working_relocate_ftpr branch to understand what went wrong.

If you find out what's wrong, I can test it at a ME8 system. Also, this branch seems to be based on an older master commit (ME version from $FPT, no signature check etc). Is this a mistake or by design?

hermitcrabslab, 2017-01-25:
@platomav both post and boot time but the worst is the second.
With BIOS setting i get same "This device cannot start. (Code 10) STATUS_DEVICE_POWER_FAILURE" On TXE/Device Manager but boot time is normal.

corna, 2017-01-25:
> Also, this branch seems to be based on an older master commit (ME version from $FPT, no signature check etc). Is this a mistake or by design?

I made this branch some time ago, now I've merged it with master

platomav, 2017-01-25:
@hermitcrabslab Could be various things. Maybe TXE or (probably) BIOS related. You can try removing TXE partitions one by one (can add them to me_cleaner's blacklist to easily test) until you start to see the slowdown. I don't have time to assist on this one unfortunately but the process is somewhat similar to what was done [here](https://github.com/corna/me_cleaner/issues/17).

hermitcrabslab, 2017-01-27:
OK now I am really pissed..
I tried to disable TXE using FUNC_DIS register at EFI level and it did disappear from PCI bus but guess what after few minutes computer reboots( no matter OS, did the same on linux too) :(
Intel WTF are you doing?
Not to mention that Intel docs where especial made wrong, i.e they say B_PCH_PMC_FUNC_DIS_SEC is on BIT20 while is on BIT19, way to go Intel...
Looks like i have to do what @platomav  suggested and test what module removed slow down the boot.

@platomav can you share more about Host ME Region Flash Protection Override (HMRFPO) command?
Thanks!

platomav, 2017-01-28:
Every 30 minutes. Yes, that is intentional and happens when the Engine firmware is found missing/corrupted. That time is given so that someone can manually repair its firmware, otherwise the platform hard-resets. Maybe the same applies if you do what you tried.

It depends on what you want to do. Removing or disabling it entirely is not an option at such a platform. But if you want to go as near as possible to that, you can use me_cleaner. Moreover, if you want the boot time to not be affected (that is probably the BIOS's "fault" trying to communicate with the Engine), find what is doing that. I suggest you start by removing partitions from $FPT + padding the actual partitions in the image. Don't forget to change the Number of Regions (0x14) as well as Checksum (0x1B) found at $FPT. You can use ME Analyzer to tell you what checksum is expected after any changes to the first 0x30 bytes of $FPT.

HMRFPO details are only available at BIOS Writer's Guides which are not public of course. So I know as much as you do give or take. Also, it doesn't help you as its purpose is to enable read/write access to the Engine region of the SPI flash for on-field repair (reflash/update/recovery). You already have that with the programmer and thus you can easily change the Flash Descriptor access settings.

As for the BIOS option, I don't know what it does exactly but it certainly looks like an implementation of HMRFPO to temporarily unlock FD read/write access to the ME for the aforementioned reasons. If you are interested, enable it and when it's set boot into Windows and check what status is reported at MEInfo -verbose tool.

corna, 2017-01-28:
At first I suggest you to remove all the partitions except for FTPR and leave its modules intact, as the communication over MEI should be handled by the HOSTCOMM module.
To do that you can change [line 269](https://github.com/corna/me_cleaner/blob/master/me_cleaner.py#L269) to `if False:` to disable the modules removal.

platomav, 2017-01-28:
@hermitcrabslab What is the model of the system you are using?

> [...] remove all the partitions except for FTPR and leave its modules intact [...]

The "move" tests had FTPR [intact](https://github.com/corna/me_cleaner/issues/18#issuecomment-274595900). I'm thinking it's some sort of BIOS check like the [previous case](https://github.com/corna/me_cleaner/issues/17https://github.com/corna/me_cleaner/issues/17).

hermitcrabslab, 2017-01-28:
@platomav This tiny thing http://www.aliexpress.com/item/Big-Promotion-Fanless-desktop-computer-Intel-Celeron-N3150-Quad-Core-Celeron-N3050-Dual-Core-Dual-HDMI/32595134091.html?spm=2114.13010608.0.0.eXkOpO ;)
And I replaced wifi card with this http://www.jetwayipc.com/content/?ADMPEIDLB_3450.html to get Intel LAN's that are more "friendly" with pfsense/freebsd.
On my case it reboots after 4-5 minutes.
@corna I will try what you and @platomav suggested but next week since I am away from little thing...

platomav, 2017-01-28:
Ah yes, these typical no-brand Chinese little machines. It's good that it even has a working Engine firmware. Some of these no-brand manufactures intentionally disable the Engine because "customers cannot find drivers" which causes power management issues (fans at 100%), 30-minute shutdowns and all the relevant crap. I've seen a lot such cases, but you are not one of them.

So, back on topic, that 4-5 minute reboot is something else. Maybe something gets triggered (security check etc) but I don't think it's related to this case. Other than that, try what corna and I mentioned whenever you have access to the machine.

hermitcrabslab, 2017-01-28:
Yeah I bet is some BIOS SMM driver that cause the restart and I would love to find it and hack it...
Plus, at least on this little thing, it works just fine without TXE, Intel should allow this IMHO...
Back to the thing, I was surprised to find good support on the seller, they even gave me a recent BIOS update and tools to flash it, this is something I did not expected to ever happen from that part of the world based on my experience with "made in Taiwan".
And was branded (PCI subsystem id) by ECS until first BIOS update, now is generic Intel 💃

closed 2017-02-07 by corna