corna/me_cleaner #19: About Nehalem and ME

persmule, 2017-01-22:
I have heard that Nehalem platforms (at least some) are possible to survive completely without ME inside the flash, just as GM45, but fail to retrieve the source of the information.

Have you ever heard such things? Is that true?

platomav, 2017-01-22:
It is true for Ibex Peak PCH that use the Ignition ME SKU (6.0.x.xxxx) as I showed [here](http://www.win-raid.com/t2443f39-Remove-ME-Ignition-firmware-completely-amp-Panther-Point-PCH-observations.html#msg33301) if that's what you mean.

It has not been tested at the rest of the Ibex Peak PCH SKUs (1.5MB, 5MB MB, 5MB DT) and at the Cave/Coleto Creek PCH Ignition SKU (6.0.50.xxx). Do you have such a 1.5/5MB Ibex Peak system to test?

persmule, 2017-01-25:
I have got access to a thinkpad x201i (which does use Ibex Peak PCH according to coreboot's source code) with 5MB ME. How would we proceed?

corna, 2017-01-25:
If you can easily unbrick you can just try the latest version of me_cleaner on your ME image

persmule, 2017-01-25:
Will it completely remove the ME or leave a BUP module?

persmule, 2017-01-25:
Result posted in issue #3 .

Nimayer, 2017-01-25:
I see that your laptop does not complete raminit on Coreboot without Intel ME.
My guess would be that the actual Nehalem raminit waits for a running Intel ME, and do not work/ go into timeout if ME is not working, this can be hinted by the following function from Nehalem raminit:
```
/* FIXME: add timeout.  */
static void wait_heci_cb_avail(int len)
{
...
}
```

persmule, 2017-01-25:
Is it exact? I manage to get TWO of its debug output. The first one is the result of first boot after flash, and the second is the result of reboot via power button.

[cblog_x201i.txt](https://github.com/corna/me_cleaner/files/729622/cblog_x201i.txt)

platomav, 2017-01-25:
So ME6 1.5/5MB does work after me_cleaner, as expected. The [other system](https://github.com/corna/me_cleaner/issues/7#issuecomment-273557154) corna was testing must have had some issue.

@persmule Try filling the entire Engine region with padding and check if it boots. It most probably wont but it is worth a try.

corna, 2017-01-25:
> The other system corna was testing must have had some issue.

We tested with coreboot, and the combination coreboot + me_cleaner probably triggered the bug indicated by @Nimayer 

@persmule can you remove all the entries (BUP and ROMP) from me_cleaner's unremovable_modules list and flash the resulting image?
Then run `sudo intelmetool -s` and check the line
`ME: Current Working State`
It it says `Platform Disable Wait` it means that the 30-min window has been triggered.

persmule, 2017-01-26:
I have done the test mentioned above on my x201i. After BUP's removal, the HECI(MEI) disappeared from the PCI bus, so I am unable to get ME's status via `intelmetool`, but the 30-min [defective by design](https://www.defectivebydesign.org/) finally explodes.

platomav, 2017-01-26:
The fact that it even posts with BUP removed shows that ME6 behaved differently compared to anything newer. A removed BUP certainly causes the Engine to tag the firmware as corrupted but maybe if the region is missing/padded it behaves differently. It might actually post that way as well since removing the BUP is pretty much equivalent to that but I suspect the real issue is the timer. As for status. you can probably use Intel's MEInfo -verbose and/or MEManuf -verbose to get it in such cases. [These](http://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html#msg10191) are Windows tools.

corna, 2017-01-26:
ME7 and ME8 behaves in the same way. Probably the Engine starts the 30-min timer also if the FTPR partition signature is valid but the hash of the BUP is not.

platomav, 2017-01-26:
Not in [my ME8 case](https://github.com/corna/me_cleaner/issues/7#issuecomment-270801399). The only difference is at that old commit the Huffman chucks were not properly removed but that proves my case even more I suppose (BUP and/or ROMP were not completely gone but no post).

closed 2017-02-07 by corna

forksmith, 2018-06-22:
So, x201 works on coreboot master again. Details and config are in the coreboot board status repo.
I have partially stripped the ME of my x201:
`
$ ./me_cleaner.py -c ../../build/coreboot.rom
Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
Found 3 partition(s)
Found FTPR header: FTPR partition spans from 0xf2000 to 0x1b2000
ME/TXE firmware version 6.2.61.3535
Public key match: Intel ME, firmware versions 6.x.x.x
The AltMeDisable bit is NOT SET
Checking the FTPR RSA signature... VALID
`
$ ./ifdtool -d ../../build/coreboot.rom
`
File ../../build/coreboot.rom is 8388608 bytes
FLMAP0:    0x03040002
  NR:      3
  FRBA:    0x40
  NC:      1
  FCBA:    0x20
FLMAP1:    0x10100206
  ISL:     0x10
  FPSBA:   0x100
  NM:      2
  FMBA:    0x60
FLMAP2:    0x00000020
  PSL:     0x0000
  FMSBA:   0x200
FLUMAP1:   0x00000aed
  Intel ME VSCC Table Length (VTL):        10
  Intel ME VSCC Table Base Address (VTBA): 0x000ed0

ME VSCC table:
  JID0:  0x001720c2
    SPI Componend Device ID 1:          0x17
    SPI Componend Device ID 0:          0x20
    SPI Componend Vendor ID:            0xc2
  VSCC0: 0x20052005
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID1:  0x001730ef
    SPI Componend Device ID 1:          0x17
    SPI Componend Device ID 0:          0x30
    SPI Componend Vendor ID:            0xef
  VSCC1: 0x20052005
    Lower Erase Opcode:                 0x20
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    4KB
    Upper Erase Opcode:                 0x20
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    4KB
  JID2:  0x001740ef
    SPI Componend Device ID 1:          0x17
    SPI Componend Device ID 0:          0x40
    SPI Componend Vendor ID:            0xef
  VSCC2: 0x20052005
[... 118 lines cut ...]
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x0000

FLMSTR3:   0x08080118 (GbE)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x0118

Found Processor Strap Section
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff

Mojo4242, 2018-08-14:
Hi! @forksmith , do you mean that we can state that coreboot + me_cleaner on Nehalem works for ME 6.0 and 6.x? If yes, maybe this [this page](https://github.com/corna/me_cleaner/wiki/me_cleaner-status) can be updated?

corna, 2018-08-15:
@Mojo4242 thanks, I forgot to update it