corna/me_cleaner #21: Power consumption and other side-effects?

nariox, 2017-02-28:
Hi,
First of all, thanks for the script! It does something I really wanted to apply to my computers, but is way beyond my technical capabilities!
This is not much of an issue, but more of a suggestion and a question. According to the documentation:
> You are losing something. Intel ME doesn't only provides some services (that you may or may not use), but it also does low-level stuff (like silicon workaround, thermal management, fan control...). Most of these things are often controlled by something else, so they're not really needed, but who can be sure?
Have there been any reports of (noticeable) changes due to using me_cleaner (in the latest version)? Either way, might be worth adding something something to that effect, like: "So far, no one has reported any adverse effects" or "In previous versions, some NICs has been found to respond adversely to me_cleaner, but workarounds are available and fixes implemented"

Now for the question, has anyone noticed any difference in power consumption? I imagine the ME chip is probably very efficient, but I just wanted to check if not running the RTOS could potentially offer other advantages to the system.

corna, 2017-03-06:
Hi,

> Have there been any reports of (noticeable) changes due to using me_cleaner (in the latest version)? Either way, might be worth adding something something to that effect, like: "So far, no one has [...]

You're right, at this point it's clear that me_cleaner does not introduce any major issue. I'll correct the documentation.

> Now for the question, has anyone noticed any difference in power consumption? I imagine the ME chip is probably very efficient, but I just wanted to check if not running the RTOS could potentially offer other advantages to the system.

The power consumption of the ME subsystem is probably so low that it's unnoticeable.

nariox, 2017-03-07:
Great. Thank you for the feedback. I wouldn't call it a "correction", I just thought it'd convey a more accurate message that we "might be losing something", but so far nothing has come up (performance, battery, functionality, etc).
Alright, closing. Thank you again.

closed 2017-03-07 by nariox