corna/me_cleaner #25: Socket P and penryn has earlier ME...

forkoz, 2017-03-24:
I have some socket P laptops with T9600/P9600, etc. The ME is "disabled" because they never paid for it on my particular examples but I know the code and the proc is there in every CPU. Can support be added to knock it out?

Chipsets are PM45 but there are a few around this series. What's worse is that they are exploitable per the initial research on ME. I get the idea that the dormant ME actually isn't.

persmule, 2017-03-24:
You should use libreboot's [ich9deblob](https://libreboot.org/docs/hcl/gm45_remove_me.html) to complete remove it. Besides, ME is not located in CPU, but in (G)MCH or PCH.

forkoz, 2017-03-24:
Thanks. Libreboot doesn't support my Dell M4400. Can I just deblob the factory image? I have a sacrificial laptop that overheats to test on. I'd love to de-blob and then flash with win (bleh) tools. Then I could see if anything like the network card dies, etc.

corna, 2017-03-24:
I think so.
Removing ME before Nehalem is easy, as removing the ME region from the Intel Flash Descriptor doesn't trigger the 30-min window (which is probably what ich9deblob does).

> Can I just deblob the factory image? 

I think so. If it doesn't work you can do it by hand with ifdtool.

closed 2017-03-24 by corna