corna/me_cleaner #26: Intel Xeon E3-1505M v5 ?

chadfurman, 2017-03-26:
Wondering how I can help.  This is my processor.  It pitches itself as using AMT ("Active Management Technology") instead of ME -- not sure if they're the same thing.

corna, 2017-03-27:
Intel AMT is a part (probably the most famous one) of Intel ME.
To apply me_cleaner you can just follow the usual steps, as explained [here](https://hardenedlinux.github.io/firmware/2016/11/17/neutralize_ME_firmware_on_sandybridge_and_ivybridge.html) (this is for a Sandy Bridge laptop, but it should more or less the same).
Since you have a Skylale processor you should check if Intel Boot Guard has been set in Verified Boot mode, as this mode prevents the use of me_cleaner.

chadfurman, 2017-03-27:
What are the chances of this bricking my laptop?

corna, 2017-03-27:
If your laptop has Intel Boot Guard in verified boot mode, the brick is guaranteed, otherwise it will probably work. **You must have a way to flash back the original firmware in case something goes wrong**

closed 2017-04-02 by corna