corna/me_cleaner #27: Remove Just Potentially Malicious Part(s) of Intel ME?

RedSoxFan04, 2017-03-30:
At some future date, will me_cleaner be able to remove **just** the **potentially malicious parts** of the Intel Management Engine? 

I’m curious because currently me_cleaner removes the **entire** Intel Management Engine, which includes **useful** and (in my opinion) **necessary** things like **fan control** and **thermal management**. I would like to be able to remove the malicious part of Intel Management Engine without having to give up **useful stuff** (mentioned above), and also without the high risk of **bricking** my next laptop (currently using Late 2011 13” MacBook Pro).

In general, I have no problem with proprietary firmware - I try to be _practical_ in such matters. I only have a problem with proprietary firmware when it actually does something _malicious_. The Intel Management Engine is the first malicious firmware I’m aware of - though only **part** of it is malicious while _the rest of it is useful_. 

The ability of the CPU to allow an independent or government hacker to _spy on me and possibly plant evidence of a crime_ is a big concern to me. I thought I had gotten away from that sort of thing when I switched over from Windows to MacOS, and eventually from MacOS to Linux - I never thought such functionality would be built into a piece of hardware, making it _far more difficult to remove_. It used to be that you could get away from that sort of stuff just by installing Linux on your PC, but now Intel has put it in firmware. While I would like to be able to get rid of the backdoor which is part of the Intel Management Engine, I would like to be able to do so without removing **useful** and/or **important** functionality, and also without possibly **bricking** my next laptop.

corna, 2017-03-30:
First I would like to point out that the features in Intel ME are mostly unused in the majority of the cases. The two you pointed out (fan control, Intel QST, and thermal management) are generally controlled by something else: for example in my Lenovo X220t they are controlled by the embedded controller, while in my Sapphire Pure Platinum H61 they are controlled by the SuperIO chip (see [this part of devicetree.cb](https://github.com/coreboot/coreboot/blob/1bea5b7df226b7d632edcf9dc735e4a382e4d026/src/mainboard/sapphire/pureplatinumh61/devicetree.cb#L111-L124) of my [coreboot port](https://github.com/coreboot/coreboot/tree/master/src/mainboard/sapphire/pureplatinumh61)). Moreover most of the me_cleaner users doesn't experience any problem, as highlighted by #3.

You won't probably experience any kind of regression, unless you are actively using some AMT features.

> [...] without the high risk of bricking my next laptop

The risk of brick while playing with the firmware is always present, even if you change a single bit. Counter-intuitively it seems that more parts of ME removed lead to less issues, as removing most of the FTPR modules [fixed an issue on a X220](https://github.com/corna/me_cleaner/issues/4#issuecomment-271150673) and [allowed to relocate the code](https://github.com/corna/me_cleaner/issues/20#issuecomment-274790828). So removing less code doesn't seem to mean "safer".

Luckily, if you can externally program your ROM chip and you have a valid backup, you should be always able to unbrick.

> The Intel Management Engine is the first malicious firmware I’m aware of - though only part of it is malicious while the rest of it is useful.

Now a word in defense of Intel: Intel ME isn't intrinsically _malicious_, we have no evidence that there is something _bad_ inside it, we just don't know and its architecture is too dangerous.

> At some future date, will me_cleaner be able to remove just the potentially malicious parts of the Intel Management Engine?

No, but just because we don't know which (if any) parts are malicious and which are not, and the only way we have is to remove as much code as possible.

Currently, me_cleaner can:
 * remove single modules from the FTPR partition (the fundamental one)
 * remove whole partitions

So, if you can figure out which modules/partitions should be kept, you can insert them in a whitelist and leave them. You can start by removing only the NFTP partition (the network) and see if everything works well, but the chance of bricking something is probably higher than the "standard" me_cleaner.

RedSoxFan04, 2017-03-30:
The only reason I’m worried about stuff like thermal management and fan control is because [here](https://github.com/corna/me_cleaner/wiki/How-does-it-work%3F) I found this: 
"_Intel ME doesn't only provide some services (that you may or may not use), but it also does low-level stuff (like silicon workaround, thermal management, fan control...). Luckily, no user has reported any side effect so far.
Often the ME region is not writeable by software: in these cases you need an external programmer to write the modified firmware_.”

So, was that bit there false from the start, or was it true when it was written and something has changed to make it false?

corna, 2017-03-30:
It's still true, but I don't exactly know which features are really used and, since me_cleaner can be applied to a wide range of processor, I added that warning. Probably on your MacBook you won't experience any issue (unless the MacBook firmware uses/requires some ME features).

RedSoxFan04, 2017-03-30:
I wasn’t planning on using on my MacBook. Rather, I was planning on using with whichever laptop I’ll get to replace my MacBook. I’m thinking of getting either a System76 Oryx Pro or an Alienware 15.

corna, 2017-03-31:
Note that on Haswell and later processor you should make sure that Intel Boot Guard is not set in Verified Boot mode, otherwise me_cleaner doesn't work.

closed 2017-03-31 by corna

Claudiomfa, 2017-04-22:
Hi,

I have this toshiba l50-a-165 (vgst/vgstg motherboard - I5-4200u Haswell) and i have done me_cleaner on the bios dump and effectivelly removes all intel management engine interface but dont remove the 30 minutes shutdown.

Anyone can give me some solution? the notebook works very well except the 30 minutes shutdown. i have tried many bios dump and no one solve my problem.

Thanks

RedSoxFan04, 2017-07-10:
I just thought of something. Isn't the X220 an older laptop, one which would have an older version of the Intel Management Engine? In which case, functions like fan control and thermal management might not be handled by a separate chip?

corna, 2017-07-10:
To be honest, I've never found a PC where thermal and fan management are controlled by ME, usually they are controlled by a superio controller in desktops/servers and by an EC in laptops, so the removal of Intel ME shouldn't have any impact on them. Intel QST (the ME software for the fan control) seems deprecated, as the latest release of [its SDK](https://software.intel.com/en-us/articles/Intel_Quiet_System_Technology_Software_Development_Kit) is dated 2011 and it supports only old chipsets.

Btw, my X220 with coreboot and a crippled ME works flawlessly.

RedSoxFan04, 2017-07-10:
I just looked it up, and your X220 was released in 2011. I'm starting to think you're right about it being fine. On the other hand, it might depend on the computer. Either way, I'd want to know it works on whichever laptop I go with before I use it - I don't want to spend a couple thousand dollars on a laptop and end up bricking it.

RedSoxFan04, 2017-07-11:
I Direct Messaged Intel on Twitter about the Management Engine and they confirmed what you said about it only being for remote access.`The Intel® Management Engine (ME) is an embedded microcontroller (integrated on Intel® chipsets) running a lightweight Microkernel Operating System that provides a low-power, OOB (Out Of Band) execution engine for management services. It brings to IT administrators access to and control over a PC: it can power on or shut down the PC, read all open files, examine all running applications, track all keys pressed and mouse movements, and even capture or display images on the screen. Intel® Management Engine doesn’t control the CPU’s temperatures; they are controlled by sensors in the motherboard.
`

corna, 2017-07-12:
> No, our systems don't have the IME

If they build Intel x86 systems, they do have Intel ME. Maybe they are referring to Intel AMT, which [can be avoided by OEMs](https://puri.sm/learn/avoiding-intel-amt/).

ThomasBrierley, 2017-10-11:
Just chiming in... I'm planning on attempting this my rather old MacbookPro 5,1 which has an SST25VF032B, here on the right: https://d3nevzfk7ii3be.cloudfront.net/igi/plv2D1eTcMZFmIQY.huge