corna/me_cleaner #28: What about CAP files?

userdsp, 2017-04-01:
Can you help me for my rig? http://valid.x86.fr/cd8mqc

Genuine Intel® CPU 0000 @ 2.60GHz
Threading	1 CPU - 4 Core - 8 Threads
Frequency	2992.86 MHz (30 * 99.76 MHz) - Uncore: 2992.9 MHz
Multiplier	Current: 30 / Min: 8 / Max: 30
Architecture	Haswell / B0-Step (22 nm)
Cpuid / Ext.	6.C.2 / 6.3C
IA Extensions	MMX, SSE, SSE2, SSE3, SSSE3, SSE4.1, SSE4.2, EM64T, VT-x, AES, AVX, AVX2, FMA3, TSX
Caches	L1D : 32 KB / L2 : 256 KB / L3 : 8192 KB
Caches Assoc.	L1D : 8-way / L2 : 8-way / L3 : 16-way
Microcode	Rev. 0x00000FFFF0006
TDP / Vcore	65 Watts / 0.976 Volts
Temperature	53 °C / 127 °F
Type	Engineering Sample
Motherboard
Model	Asus MAXIMUS VI HERO
Socket	Socket 1150 LGA
North Bridge	Intel Haswell rev 02
South Bridge	Intel Z87 rev C2
BIOS	American Megatrends Inc. 1603 (08/15/2014)

corna, 2017-04-02:
It seems that CAP files are used on ASUS motherboards to flash the BIOS in their own way. It seems to be a "AMI Aptio capsule", which can be extracted with [UEFITool](https://github.com/LongSoft/UEFITool): open the CAP file in UEFITool, right click and "Extract body as" to get the plain dump ("ROM" format). Now you can use me_cleaner on the resulting file and flash it somehow.

Not that:
 * I've not tested this procedure
 * If your MB has Intel Boot Guard enabled this procedure **will brick** your MB, and you'll need a way to flash back the original firmware (maybe the Asus flashback feature works, I don't know)

userdsp, 2017-04-02:
Yes flashback works. Will i be able to use my ES qde8 cpu without 30 min
power off? And can you edit this bios for me pls? I will pay. I can flash
it from uefi, but will everything write? Maybe i muat use a programmer
device?

userdsp, 2017-04-02:
I was try to edin by manual AE part using HxD hex editor. I was searching for first bit's 20 20 80 0F  and make them 00 but nothing change. PC still power off at 30min.

userdsp, 2017-04-02:
I was follow your steps.
root@kali:~/Desktop/me_cleaner# python me_cleaner.py m6h.bin 
Unknown image

userdsp, 2017-04-02:
How to convert it back to .cap file? Just renaming?

userdsp, 2017-04-02:
My easy flash in bios give a message "Thr file format is not correct. Ensure that the filename extension is .CAP .

userdsp, 2017-04-02:
All bios ver. for this board have the same size and the new one is smaller. Will be possible to put 0 in places where parts must be deleted to prevent the size difference?

corna, 2017-04-02:
> Will i be able to use my ES qde8 cpu without 30 min power off?

Maybe, it depends on how Intel Boot Guard is configured on your platform.

> And can you edit this bios for me pls?

Unfortunately no, the firmware you linked is not redistributable, I can only show you the steps to follow.

> I will pay.

If you want you can make a donation to the project [here](https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=B5HCXCLZVCVZ8).

> How to convert it back to .cap file? Just renaming?

You can use the ASUS tool [" Bios Renamer for USB BIOS Flashback and Crashfree BIOS 2"](http://dlcdnet.asus.com/pub/ASUS/mb/LGA1155/P8Z77-M/BRenamer.zip) (for Windows), it converts .ROM files to .CAP.

userdsp, 2017-04-02:
Bios Renamer just rename the file and doesn't help :( Bios must have exact same size like the original one and in the beginning of the bios there is a verification check. Your script just erase the first 2kbytes from this file :/

H2KChicken, 2017-04-06:
Hi, userdsp. 

I believe it may be best to flash your board using an external SPI flasher instead of the built in flasher. Your Asus board uses a Winbond BIOS chip (DIP-8). This means you can remove it, put it on a breadboard, and connect it to a SPI flasher like a beagle bone to bypass the verification check.

(Also, thank you Corna for explaining how to convert CAP files)

corna, 2017-04-06:
> I believe it may be best to flash your board using an external SPI flasher instead of the built in flasher. 

I agree, it's safer. You can also buy a spare BIOS chip (they're quite cheap and you can find them easily on ebay) and keep the original untouched if you want to be extra sure.

closed 2017-04-06 by corna

userdsp, 2017-04-07:
My BIOS chip is bricked and cant flash with SPI programmer :) I will buy a new one! Thanks a lot!!!

corna, 2017-04-07:
Have you extracted the chip from the motherboard? Which software and programmer are you using? Have you pulled up the WP and HOLD pins?

KirbyFan102, 2024-02-18:
> It seems that CAP files are used on ASUS motherboards to flash the BIOS in their own way. It seems to be a "AMI Aptio capsule", which can be extracted with [...]

Do you think this procedure would also work on my Samsung Expert NP350xbe-XD1BR, which also has a .cap file in its bios executable I am trying to flash? Is there a way to disable Intel Boot Guard?