corna/me_cleaner #30: me_cleaner haswell

Claudiomfa, 2017-04-22:
Hi,

I have this toshiba l50-a-165 (vgst/vgstg motherboard - I5-4200u Haswell) and i have done me_cleaner on the bios dump and effectivelly removes all intel management engine interface but dont remove the 30 minutes shutdown.

Anyone can give me some solution? the notebook works very well except the 30 minutes shutdown. i have tried many bios dump and no one solve my problem.

Thanks

corna, 2017-04-24:
Hi @Claudiomfa,

Is there a BIOS image (containing a ME image) available on the manufacturer's website? Can you also provide the output of `intelmetool -s` (available on the [coreboot's repo](https://github.com/coreboot/coreboot/tree/master/util/intelmetool))?

Claudiomfa, 2017-04-24:
@corna 

On this link i attached the manufacturer's website files and a dump.

http://cmfa.duckdns.org/cloud/public.php?service=files&t=b2806d052d97630ab2422ad25178f799

In relation of intelmetool can i use them in windows or i have use that on linux?

corna, 2017-04-24:
Linux
If you don't have it, I suggest you a distro with a kernel < 4.9 (like Ubuntu 16.04), to avoid possible problems on newer kernels

Claudiomfa, 2017-04-24:
@corna 

How can I use that comand in ubuntu 14.04 lts?
where I can download the intelmetool files?

corna, 2017-04-24:
$ sudo apt-get install libpci-dev 
     $ git clone --depth=1 http://review.coreboot.org/p/coreboot
     $ cd coreboot/util/intelmetool
     $ make
     $ sudo ./intelmetool -s

Claudiomfa, 2017-04-25:
@corna this is the result of intelmetool

Bad news, you have a `Lynx Point-LP LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI was hidden on PCI, now unlocked
MEI found: [8086:9c3a] Lynx Point-LP HECI #0

ME Status   : 0x3026
ME Status 2 : 0x80000

ME: FW Partition Table      : BAD
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Platform Disable Wait
ME: Current Operation State : Preboot
ME: Current Operation Mode  : Normal
ME: Error Code              : Image Failure
ME: Progress Phase          : ROM Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : (null)

ME: Extend Register not valid

ME: has a broken implementation on your board with this BIOS
ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
Re-hiding MEI device...done

corna, 2017-05-02:
From the output of intelmetool it appears that:
 * The FPT is damaged (wrong checksum)
 * ME is disabled and is in the 30-min window

However the input files are good, and the cleaning process seems OK. Can you check if the flashed ME image is identical to the input one (flash, dump again and do a diff between the original and the dumped).

DarkCoridor, 2017-05-12:
@corna can you elaborate on the `iomem=relaxed` you mention for kernels >= 4.9 as my output is:

```
user@host:~/coreboot/util/intelmetool$ sudo ./intelmetool -s 
Bad news, you have a `QM67 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

Error mapping physical memory 0x0000004275159040 [0x4000] ERRNO=1 Operation not permitted
Could not map MEI PCI device memory
```

Where do I specify that iomem config value?

corna, 2017-05-12:
It is a kernel parameter, so it should be added in the GRUB configuration:

`/etc/default/grub`

     [...]
     GRUB_CMDLINE_LINUX_DEFAULT="iomem=relaxed ..."
     [...]

Then regenerate the GRUB configuration with `sudo update-grub`

stone212, 2017-06-30:
Oh yah.  Okay, much better.  That kernel parameter made all the different.  My output now matches what I see on your page here: 
https://github.com/corna/me_cleaner/wiki/External-flashing

On a side note: if you ever feel like you need a hand with documenting things, send me a note here.  We can find a way to talk and I'll be happy to do some writing, now that I have _some_ idea how this all works.

Diamondinthedirt, 2019-03-20:
Hey, how to run me-cleaner ? it has no .bat or .exe files

persmule, 2019-03-20:
> it has no .bat or .exe files
It does have .py files, so you should try to leave the bottom of the M$WIN and learn how to run them.

youiopmop, 2019-03-20:
> Hey, how to run me-cleaner ? it has no .bat or .exe files

i wouldn't recommend trying to do something to your computer that can potentially brick it if you don't know how to fix it. if you don't even know how to run a python script, how are you going to be able to unbrick it if something goes wrong?

Diamondinthedirt, 2019-03-20:
> 
> 
> > it has no .bat or .exe files
> 
> It does have .py files, so you should try to leave the bottom of the M$WIN and learn how to run them.

can i ask you to provide a link please? i wanna know how to run this script, there must be guide for newbs

youiopmop, 2019-03-20:
> can i ask you to provide a link please? i wanna know how to run this script, there must be guide for newbs

https://www.python.org/doc/

Diamondinthedirt, 2019-03-20:
> i wouldn't recommend trying to do something to your computer that can potentially brick it if you don't know how to fix it. if you don't even know how to run a python script, how are you going to be [...]

thank you, one day i gonna break up with this ME region once and forever

Diamondinthedirt, 2019-03-20:
> 
> 
> > can i ask you to provide a link please? i wanna know how to run this script, there must be guide for newbs
> 
> https://www.python.org/doc/

I have already replaced ME region in my original bios image with clean ME region using Intel ME&TXE Injector but i suspect it hasnt been overwritten completely

manontheotherside, 2021-07-11:
@Claudiomfa

30 min shutdown is a bootguard profile setting, you have to change bootguard profile to 0 (no FVME) or to 3 (VM). 
No need to use me cleaner, use official Intel CSME system tools

manontheotherside, 2021-07-11:
even setting a HAP bit on Haswell breaks memory profiles and drop RAM to lowest frequency