corna/me_cleaner #31: BIOS region alignment on ivybridge platforms.

persmule, 2017-04-27:


corna, 2017-05-11:
Interesting.
Are you sure that this isn't an isolated case?

dogroll, 2017-05-11:
This might be related to the issue I'm having.

I am running coreboot 4.6-86-g72f730e23c on my T520 (with i5-3320M). With me_cleaner enabled in the coreboot build, I get frequent boot times (Windows 10) of around 60 seconds---incredibly long. With me_cleaner disabled, I get a standard ~8 second boot from my mSATA SSD. With me_cleaner enabled, the unusually long boots tend to occur immediately after a restart/shutdown. _Most_, but not all of the time, when starting the machine from a long term power off state (e.g. I shutdown and leave the laptop for a few minutes before turning it back on again) I would get an incredibly fast ~3 second boot.

K1773R, 2017-05-11:
@dogroll im planing to flash my T520 aswell. Are there any other issues or everything working fine after applying me_cleaner? Excuse the offtopic ;)

dogroll, 2017-05-11:
Hi, I couldn't discern any other issues, no 30 minute shutoff etc, so me_cleaner seems to work fine for the most part. But obviously the boot time issue is a bit of a turn off.

I'm not a linux user so I didn't test that, perhaps this issue only affects windows.

persmule, 2017-05-11:
> Are you sure that this isn't an isolated case?

corna, 2017-07-25:
I finally had the time to check this.
On my Sapphire Pure Platinum H61 with an i7-3770 and coreboot I have this layout

```
00000000:00000fff fd
00018000:003fffff bios
00001000:00017fff me
```

but the PC boots fine without any delay. Do you have a timestamp log of those boots? If not build coreboot with the option `CONFIG_COLLECT_TIMESTAMPS`, flash it and then run `cbmem -t` **on the first boot after the flash**.

@dogroll having a timestamp log for each boot (slow and fast) could be particularly useful. I'm afraid you'll need Linux though.

persmule, 2017-07-28:
I have completed the test on my thinkpad x230.
Below are two time stamp reports, each of which is concatenated after its corresponding ifd layout:

bios region aligned to 0x1000
> 00000000:00000fff fd
> 0001b000:00bfffff bios
> 00003000:0001afff me
> 00001000:00002fff gbe
> 23 entries total:
> 
>    0:1st timestamp                                     2,294
>    1:start of rom stage                                65,436 (63,141)
>    2:before ram initialization                         74,347 (8,910)
>    3:after ram initialization                          89,931,234 (89,856,887)
>    4:end of romstage                                   89,985,753 (54,519)
>    8:starting to load ramstage                         89,988,974 (3,220)
>   15:starting LZMA decompress (ignore for x86)         89,989,162 (187)
>   16:finished LZMA decompress (ignore for x86)         90,014,342 (25,180)
>    9:finished loading ramstage                         90,014,573 (231)
>   10:start of ramstage                                 90,014,708 (134)
>   30:device enumeration                                90,014,719 (10)
>   40:device configuration                              90,020,601 (5,882)
>   50:device enable                                     90,022,900 (2,298)
>   60:device initialization                             90,023,040 (140)
>   70:device setup done                                 90,188,315 (165,274)
>   75:cbmem post                                        90,188,319 (3)
>   80:write tables                                      90,881,194 (692,875)
>   85:finalize chips                                    90,913,790 (32,595)
>   90:load payload                                      90,913,794 (4)
>   15:starting LZMA decompress (ignore for x86)         90,914,222 (427)
>   16:finished LZMA decompress (ignore for x86)         90,928,435 (14,213)
>   15:starting LZMA decompress (ignore for x86)         90,928,443 (8)
>   16:finished LZMA decompress (ignore for x86)         91,152,623 (224,179)
>   99:selfboot jump                                     91,152,651 (27)
> 
> Total Time: 91,150,344

bios region aligned to 0x10000:

> 00000000:00000fff fd
> 00020000:00bfffff bios
> 00003000:0001ffff me
> 00001000:00002fff gbe
> 23 entries total:
> 
>    0:1st timestamp                                     2,312
>    1:start of rom stage                                65,350 (63,037)
>    2:before ram initialization                         67,009 (1,659)
>    3:after ram initialization                          1,526,964 (1,459,955)
>    4:end of romstage                                   1,575,840 (48,875)
>    8:starting to load ramstage                         1,579,432 (3,592)
>   15:starting LZMA decompress (ignore for x86)         1,579,768 (336)
>   16:finished LZMA decompress (ignore for x86)         1,609,570 (29,801)
>    9:finished loading ramstage                         1,609,778 (208)
>   10:start of ramstage                                 1,609,892 (113)
>   30:device enumeration                                1,609,903 (11)
>   40:device configuration                              1,615,792 (5,889)
>   50:device enable                                     1,618,090 (2,297)
>   60:device initialization                             1,618,231 (141)
>   70:device setup done                                 1,783,521 (165,289)
>   75:cbmem post                                        1,783,525 (4)
>   80:write tables                                      2,475,097 (691,571)
>   85:finalize chips                                    2,507,645 (32,548)
>   90:load payload                                      2,507,649 (4)
>   15:starting LZMA decompress (ignore for x86)         2,508,034 (384)
>   16:finished LZMA decompress (ignore for x86)         2,522,401 (14,367)
>   15:starting LZMA decompress (ignore for x86)         2,522,409 (8)
>   16:finished LZMA decompress (ignore for x86)         2,745,607 (223,197)
>   99:selfboot jump                                     2,745,635 (27)
> 
> Total Time: 2,743,313

corna, 2017-08-28:
I've checked again on my Ivy Bridge platform with coreboot with full RAM training (MRC cache cleared) and with different RAM sticks combinations and the raminit is still fast, even if the BIOS region is not 64k-aligned. At this point it seems to me that your problem is specific for your board.

You should try to get a verbose raminit log and ask on the coreboot ML.

I had problems with RAM training on my board (with the ME intact) that required me to dump and restore the MRC cache at every coreboot update, but with recent builds the RAM training [gets performed correctly with good performance](https://review.coreboot.org/cgit/board-status.git/tree/sapphire/pureplatinumh61/4.6-1237-g3f3025d7f1/2017-08-26T16_30_37Z/coreboot_timestamps.txt?): can you try again with the latest coreboot commits?

persmule, 2017-08-28:
Timestamp for the first boot after flash on my x230 with the latest coreboot commits.
[timestamp-x230.txt](https://github.com/corna/me_cleaner/files/1255868/timestamp-x230.txt)

corna, 2017-08-28:
Do you have a verbose raminit log?

persmule, 2017-08-29:
I have just collected one, along with the corresponding timestamp:
[x230-log.txt](https://github.com/corna/me_cleaner/files/1259052/x230-log.txt)
[timestamp-x230.txt](https://github.com/corna/me_cleaner/files/1259062/timestamp-x230.txt)