corna/me_cleaner #33: Need commandline param to override ME detection.

perillamint, 2017-05-02:
Some firmwares, like Dell XPS 9550 (OEM firmware v1.2.21), has ME firmware but its nr value is zero.

I looked around some other project like https://github.com/skochinsky/me-tools, they overrides nr>2 detection. (https://github.com/skochinsky/me-tools/blob/master/me_unpack.py#L1009)

So I tried overriding that detection routine at https://github.com/corna/me_cleaner/blob/master/me_cleaner.py#L335, like if nr > 2 or True and it removed ME blobs from my firmware dump like this:

```
Full image detected
The ME/TXE region goes from 0x3000 to 0x6ff000
Found FPT header at 0x3010
Found 13 partition(s)
Found FTPR header: FTPR partition spans from 0x4000 to 0x133000
ME/TXE firmware version 11.0.18.1002
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x0c)...
Modules removal in ME v11 or greater is not yet supported
Checking FTPR RSA signature... VALID
Done! Good luck!
```

So I think we need commandline param to override nr > 2 check.

corna, 2017-05-02:
The NR >= 2 check can probably be safely ignored in all the cases (without the need of an additional parameter), but I need a sample of that BIOS. Is it available somewhere on the Dell website?

corna, 2017-05-02:
If not the output of `ifdtool -d <dump>` should be sufficient

perillamint, 2017-05-03:
@corna 

I unpacked dell official bios updater (available from [here](http://www.dell.com/support/home/kr/ko/krdhs1/Drivers/DriversDetails?driverId=6RV34&fileId=3657572186&osCode=WT64A&productCode=xps-15-9550-laptop&languageCode=ko&categoryId=BI), packed in EXE, zlib'ed blob at 0x67810) using binwalk -e, but its header mismatches "full image" from EEPROM but it contains $FPT.

If you need full image dump from EEPROM, I can send you my ROM dump. Since that dump may contain some sensitive information (like, Dell service tag or other sensitive one), I need some way to deliver it privately. could you suggest way to share my EEPROM dump?

p.s. I'll post ifdtool result after few hrs when I back to my home. (I forgot to run sshd on my sub-laptop and currently I have no way to access files in my machine remotely)

perillamint, 2017-05-03:
BTW, does anyone know using Bus Pirate v4 as flasher (Using Vcc = 3V3) on Skylake laptops are "safe"?

My laptop didn't boot after I restored backup of original BIOS

Thiblizz, 2017-05-03:
@perillamint Yep, it should work great on the latest release: [https://github.com/BusPirate/Bus_Pirate](https://github.com/BusPirate/Bus_Pirate).

Are you using flashrom?

perillamint, 2017-05-03:
@Thiblizz Yes, I'm using flashrom. 

I wonder is, Using 3V3 Vcc is right voltage for doing that... I failed to boot my laptop even after I restored backup (original) dump.. or it might be fault of my crappy EEPROM clip corrupted dump or flashing process...

corna, 2017-05-03:
@perillamint It shouldn't, but I just need the descriptor (usually the first 4kB), which can be extracted with `ifdtool -x` and shouldn't contain any sensitive information. You can send it to me at nicola@corna.info

perillamint, 2017-05-03:
@corna I sent flash descriptor dump to your mail.

corna, 2017-05-04:
Fixed in ba885fa9fce412436594bbd98c5a644c9c3dbf40, thanks for your help

closed 2017-05-04 by corna