corna/me_cleaner #34: Does not work on XPS 9550

perillamint, 2017-05-02:
**Laptop/Motherboard**: Dell XPS 9550 (i7-6700HQ)
**BIOS**: OEM BIOS 1.2.21
**Issue**: Does not boot at all.

I dumped XPS BIOS using bus pirate. I need to do some hack as described in #33, and it successfully removed tables.

After flashing, it does not boot at all. It shuts itself after few seconds. No screen output.

I tried 0ac4b4.

**Hardware notes**: This motherboard has two flash. One is 1MB one with some device descriptor things and other thing is 16MB which stores BIOS.

corna, 2017-05-02:
Can you run `ifdtool -d` on the dump of the 1MB flash chip?

perillamint, 2017-05-03:
@corna It says

```
File UT2_backup.rom is 1048576 bytes
No Flash Descriptor found in this image
```

corna, 2017-05-04:
And what about `ifdtool -d` on the 16MB one?

corna, 2017-05-04:
Oh right, I already have that output ;)
The 16MB one seems to contain BIOS, ME and GbE, I don't know the purpose of the 1MB one.

5685C4A059D5, 2017-05-04:
Have you checked with MEInfo to see if boot guard is enabled?

perillamint, 2017-05-04:
@5685C4A059D5 Not yet. Can you describe how can I do that?

perillamint, 2017-05-04:
@5685C4A059D5 I searched web but it seems I can't run that.

My system isn't working at all. It does not boot(it seems it cannot load BIOS) so I cannot run ME System Tools on my laptop

5685C4A059D5, 2017-05-04:
Flash the original BIOS back then run the tool.

perillamint, 2017-05-04:
@5685C4A059D5 I can't do that now because my cheap soic8 is broken while I trying In-circuit programming. 

I need to buy clip to continue this. (few weeks shipping time required -- I'll buy promona clip through Digi-key rather then el-cheapo made in china clip)

corna, 2017-05-04:
I personally hate SOIC8 clips, they always break and you don't know if it's actually flashing or not.
If you want something more reliable you can buy SMD clips (like [this one](https://www.aliexpress.com/item/12Pcs-Mini-Grabber-6-Color-Test-Probe-Hook-For-SMD-IC-Test-Cilps-SMD-IC-Hook/32684876176.html?ws_ab_test=searchweb0_0,searchweb201602_5_10152_10065_10151_10068_436_10136_10137_10157_10060_10138_10155_10062_10156_10154_10056_10055_10054_10059_10099_10103_10102_10096_10147_10052_10053_10142_10107_10050_10051_10084_10083_10080_10082_10081_10177_10110_10111_10112_10113_10114_10181_10183_10182_10078_10079_10073_10070_10123,searchweb201603_9,ppcSwitch_5&btsid=e85aa9cd-fbab-4d5c-9b91-993c09410a63&algo_expid=2663a6da-91bd-4e5e-8cf8-da1679a3a1bd-3&algo_pvid=2663a6da-91bd-4e5e-8cf8-da1679a3a1bd)): they take a bit longer to be connected, but they're much more reliable.

Let me know when you can flash the original BIOS back, then we can resume the analysis.

Hikari-chin, 2017-05-05:
It is extremely likely BootGuard is the cause; Dell advertises the 9550 (and now 9560, as well as their enterprise brothers the Precision 5510/5520)  as being Intel BootGuard enabled. This device has a 2.0 version TPM as well (though Dell provides a switch to switch between 1.2 and 2.0 as well as 1.2-only TPM firmware images).

vincent-legoll, 2017-05-06:
@Hikari-chin : How does one get that output ?

Hikari-chin, 2017-05-06:
Run MEinfo from the Intel ME System Tools package available [here](http://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html#msg10191) in CMD, PowerShell, etc. If you're not running Windows, you'll need to use MEAnalyzer or something.

libcg, 2017-05-10:
Does that mean it's not possible to disable ME on these laptop? I'd like to give it a shot on my Precision 5510, which is roughly equivalent to the XPS 9550.

5685C4A059D5, 2017-05-10:
So far nothing can be done for machines with BootGuard enabled.

corna, 2017-05-11:
Like @5685C4A059D5 said, if Boot Guard is set in Verified Boot - Immediate shutdown (FVE or FVME) we can do nothing. It seems that some manufacturers didn't enable Boot Guard properly (as exposed by [Alexander Ermolov - Safeguarding rootkits: Intel BootGuard - Zeronights 2016](https://2016.zeronights.ru/wp-content/uploads/2017/03/Intel-BootGuard.pdf), slide 54), so it might be still possible to use me_cleaner on those platforms.

Hikari-chin, 2017-10-21:
It may be possible to disable (but not remove) the ME using the HAP bit - me_cleaner has this ability via the `-s` option now. It may be that doing this does not trigger the BootGuard immediate shutdown, but it has not been tested. If it does trigger BootGuard, recovery would require opening the device and externally programming the flash chips on the motherboard. 

I do not have the tools for this, but if someone does and would be willing to test, it would be nice to know if the HAP bit allows us a disabled ME even with BootGuard.

corna, 2017-10-23:
I'm currently working on a X1 Carbon (Broadwell) with BG set in Verified Boot - Immediate Shutdown, and it seems to work with the `-s` option, but not with the `-S`. Intel ME, when BG is enabled, seems to require more modules or partitions, I'm trying to determine which ones are really needed.

al3xtjames, 2017-11-18:
Can someone post a picture of the correct SOIC-8 chip on the motherboard? Currently don't have a SOIC clip (broke mine and need to order another), but I can test once I get one again.

perillamint, 2017-11-18:
There is *3 chips*, at first time, I missed one chip among this. (or more, if I missed another one). One chip is buried under plastic film. They're located near audio jack I remember.

Sorry for no photo. I'm not in my home now.

5685C4A059D5, 2017-11-18:
I don't think they use soic on the newer models. My X260 uses a wson
package.

shawnanastasio, 2018-01-01:
@perillamint Do you recall where the location of the third chip is? I'm currently disassembling one and can only find the two under the adhesive strip near the headphone jack. Is it on the reverse side of the motherboard?

al3xtjames, 2018-01-01:
Unfortunately for those with the XPS 15 9560 (and apparently the XPS 13 9360), the main 16 MB SOIC-8 package (containing ME firmware + UEFI) has been replaced with a 32 MB WSON-8 package.

perillamint, 2018-01-01:
@shawnanastasio One near eDP connector. If I remember correctly, you have to remove eDP
connector holding bracket to access that chip.
[image]

perillamint, 2018-01-01:
Note that I suspect powering SOIC chip through SOIC clip damaged my previous board. Please be careful.

shawnanastasio, 2018-01-01:
@perillamint Thanks much. It turns out that I was mistaken and the motherboard was in fact from an XPS 15 9650 and had a WSON-8 chip like @al3xtjames described.

oguzhantopcu, 2018-01-07:
did anyone tried `-s` option with 9550 and succeeded? since I do not have these soic things and the knowledge about how to use them I do not want to take risk for now.

goodwin, 2018-01-19:
Is there somebody who has full SPI dump from 9550, made with hardware programmer? I have my Intel ME corrupted, and need another dump from some working machine.
If somebody can - send me to good.win.alexs at gmail

al3xtjames, 2018-01-19:
If it's just the ME region that's bad, you can extract the ME region from the latest BIOS update (use 7-Zip/binwalk/whatever + [PFSExtractor](https://github.com/LongSoft/PFSExtractor).

goodwin, 2018-01-19:
Yes, official intel's Flash Image Tools rejects to open my ME dump bcz of corrupted ME data. So i can't cleanup my own ME dump to flash it back. And my ME stuck on initializing phase - so it's not functioning properly. And that happened after installing official Dell bios update. So now i just don't know how i can fix my ME. Only hope if i can find somewhere any other dump with ME in working condition to use for cleanup and reflash.

al3xtjames, 2018-01-19:
Well, here is the ME firmware (extracted from BIOS 1.6.1 updater for the 9550).

[section_2_11.8.50.3426.data.zip](https://github.com/corna/me_cleaner/files/1647697/section_2_11.8.50.3426.data.zip)

If FITC doesn't let you flash, you may want to try the BIOS recovery procedure, which is documented [here](http://forum.notebookreview.com/threads/xps-9560-disabling-integrated-graphics.806855/) (for the 9560 at least, I believe it is similar for the 9550). I *think* this should rewrite the ME firmware, but I am not sure.

goodwin, 2018-01-19:
No, it will not rewrite ME. I did try dozen times. Same as "fptw -greset" and resetting RTC (with clear ME on rtc reset option enabled). on win-raid forum there is guide how to clean own dumped ME from existing "data" to leave only clean state of me. but it doesn't work on my image - due to existing issues with me data inside of image. Will try to work with that image from updater - let see what i can do. In worst case i have current dump.

RomSand, 2018-04-06:
I got xps 9550 that stuck on Dell logo, strange thing it turns on right away then I plug the power, when I disconnected the battery it goes to diagnostic mode there it passes everything except battery but after reboot it still freezes on Dell logo, I'm thinking to rewrite SPI bios but can't find bump or backup anywhere.

RomSand, 2018-04-06:
I need backup of the bios , I don't think mine is good any more...

yuukieve, 2018-06-20:
Has anyone attempted this on the 9550 with just the HAP bit set? If so, does it work without issues?

lorantsz, 2018-09-21:
I managed to make it work for a 9550 with -s-O however I did the mistake of trying to update the BIOS trough the .EXE file from the dell website and it obviously bricked my device.

I'm suspecting that the .exe is writing some parts on all 3 of the bios chips (16M, 4MB, 1MB)

Now the laptop fires up with blank screen and gives a CPU Fail code (1 white 2 Amber) error code.

Would anyone have handy a dump of these 3 chips? [MB is : LA-C361P Rev2.0 (A01)]

yuukieve, 2019-02-19:
@lorantsz Can you confirm how you dumped and flashed the 9550 ME firmware, as well as the *exact* me_cleaner command that worked for you?