corna/me_cleaner #35: T430 SPI Dump - "Unknown Image"

n4ru, 2017-05-04:
me_cleaner seems to not like my fresh images pulled from my T430. v2.72 of BIOS, pulled with flashrom.

n4ru, 2017-05-04:
Same issue seems to occur when pulled with software through Windows (12MB filesize)

Thiblizz, 2017-05-04:
@n4ru Can you run `ifdtool -d` on the dump and post the output.

The BIOS should be located in the 4MB binary file and the descriptor as well as other regions in the 8MB binary file.
Note that both the descriptor and ME are locked, so you will need an external programmer.

cutamar, 2018-12-05:
Same problem on a T430, @n4ru could you solve it?

@Thiblizz here is my output:
```
File bios1.bin is 4194304 bytes
ICH Revision: 100 series Sunrise Point
FLMAP0:    0x5ec0940f
  NR:      6
  FRBA:    0xc00
  NC:      1
  FCBA:    0xf0
FLMAP1:    0x5ec032c3
  ISL:     0x5e
  FPSBA:   0xc00
  NM:      2
  FMBA:    0xc30
FLMAP2:    0x026856c3
  PSL:     0x6856
  FMSBA:   0xc30
FLUMAP1:   0xced7002c
  Intel ME VSCC Table Length (VTL):        0
  Intel ME VSCC Table Base Address (VTBA): 0x0002c0

ME VSCC table:

OEM Section:
00: cc d4 32 e6 86 d4 e5 33 97 19 93 25 68 fa 0c 7b
10: c6 2f ba 95 0c 33 61 17 f8 bd 74 04 a0 39 1e 9d
20: a1 7b e9 fe 56 38 52 4c 06 56 5f 7a 0c 44 87 d9
30: 15 c5 e4 c9 e7 0c 63 94 84 56 df 1c ee bc fe 37

Found Region Section
FLREG0:    0x66f1a3c3
  Flash Region 0 (Flash Descriptor): 023c3000 - 066f1fff 
FLREG1:    0xb9767f31
  Flash Region 1 (BIOS): 07f31000 - 03976fff (unused)
FLREG2:    0xb533caf2
  Flash Region 2 (Intel ME): 04af2000 - 03533fff (unused)
FLREG3:    0x2a09d666
  Flash Region 3 (GbE): 05666000 - 02a09fff (unused)
FLREG4:    0xd1d277d6
  Flash Region 4 (Platform Data): 077d6000 - 051d2fff (unused)
FLREG5:    0xdf29c170
  Flash Region 5 (Reserved): 04170000 - 05f29fff 
FLREG6:    0x73cfb98f
  Flash Region 6 (Reserved): 0398f000 - 073cffff 
FLREG7:    0xfa4489f4
  Flash Region 7 (Reserved): 009f4000 - 07a44fff 
FLREG8:    0xc384dce4
  Flash Region 8 (EC): 05ce4000 - 04384fff (unused)

Found Component Section
FLCOMP     0xbc48970a
  Dual Output Fast Read Support:       not supported
  Read ID/Read Status Clock Frequency: unknown<7>MHz
  Write/Erase Clock Frequency:         30MHz
  Fast Read Clock Frequency:           48MHz
  Fast Read Support:                   not supported
  Read Clock Frequency:                30MHz
  Component 2 Density:                 512KB
  Component 1 Density:                 unknown<a>MB
FLILL      0x48be0cba
  Invalid Instruction 3: 0x48
  Invalid Instruction 2: 0xbe
  Invalid Instruction 1: 0x0c
  Invalid Instruction 0: 0xba
FLPB       0x8dce9f91
  Flash Partition Boundary Address: 0xf91000

Found PCH Strap Section
PCHSTRP0:  0x66f1a3c3
PCHSTRP1:  0xb9767f31
PCHSTRP2:  0xb533caf2
PCHSTRP3:  0x2a09d666
PCHSTRP4:  0xd1d277d6
PCHSTRP5:  0xdf29c170
PCHSTRP6:  0x73cfb98f
PCHSTRP7:  0xfa4489f4
PCHSTRP8:  0xc384dce4
PCHSTRP9:  0x463b1300
PCHSTRP10: 0x52158d16
PCHSTRP11: 0x51230c37
PCHSTRP12: 0xf1b75d09
PCHSTRP13: 0x8bf2d9f3
PCHSTRP14: 0x07d7b9b5
PCHSTRP15: 0xcd4db80c
PCHSTRP16: 0x6d1738a6
PCHSTRP17: 0xa404214c
HAP bit is set

Found Master Section
FLMSTR1:   0xf1b75d09 (Host CPU/BIOS)
  EC Region Write Access:            enabled
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     enabled
  EC Region Read Access:             enabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      enabled
FLMSTR2:   0x8bf2d9f3 (Intel ME)
  EC Region Write Access:            disabled
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     enabled
  EC Region Read Access:             disabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      enabled
FLMSTR3:   0x07d7b9b5 (GbE)
  EC Region Write Access:            disabled
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     enabled
  EC Region Read Access:             enabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      enabled
FLMSTR5:   0x6d1738a6 (EC)
  EC Region Write Access:            disabled
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           disabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     enabled
  EC Region Read Access:             enabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
Found Processor Strap Section
????:      0xf1b75d09
????:      0x8bf2d9f3
????:      0x07d7b9b5
????:      0xcd4db80c
????:      0x6d1738a6
????:      0xa404214c
????:      0x1d231487
????:      0x729b475d
```