corna/me_cleaner #36: skylake howto

bol-van, 2017-05-07:
I successfully applied me_cleaner to B75 and Z87  OEM bios.
Just to learn how it works on skylake I downloaded several bios images from newer botherboards and tried me_cleaner on them. Looks like it cant work with full bios images.

```
Full image detected
This image does not contains a ME/TXE firmware NR = 0)
```

However it looks working on extracted ME region blob.
```
ME/TXE image detected
Found FPT header at 0x10
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0x1000 to 0xa8000
ME/TXE firmware version 11.6.25.1229
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xd0)...
Modules removal in ME v11 or greater is not yet supported
Checking FTPR RSA signature... VALID
Done! Good luck!
```
Then I tried to use FIT v11 to integrate ME image into full image. It checks integrity and rejects modified blob. I tried to 'dd' modified me to offset 0x1000 of full image but FIT also refused to open modified image.
I'm wondering if its safe to flash this modified image ?
If I have ME region not write locked will it be good to "fpt -me -f modified_me.bin" ?

corna, 2017-05-08:
Can you try again on the full image with the latest version of me_cleaner? In commit ba885fa9fce412436594bbd98c5a644c9c3dbf40 I've removed the NR >= 2 requirement

bol-van, 2017-05-08:
It worked. Image was modified. At offset 1000 I see only "$FPT", "FTPR". Others were removed.
FIT v11 cant load image because from its point of view ME region is corrupted

skochinsky, 2017-05-08:
I think FIT requires the EFFS partition since it stores many system-specific settings and won't work without it.

platomav, 2017-05-08:
FIT v11.x cannot work without EFFS but older versions can.

No matter what, FIT should not be used after nuking the ME. The nuked region can be replaced either manually from the ME start & limit at Flash Descriptor or via UEFITool, provided that the size remains the same. To also save space from the removed modules, manual work is needed at the FD ME size first.

corna, 2017-07-25:
@bol-van I've just pushed a new commit (d8bf20d0f29d0add5606952d19414590dc18c113) in the testing_me11 branch which further reduces the amount of code in the Skylake (ME 11) images. Can you test it?

I've already widely tested it on my Skylake platform, it should be safe.