corna/me_cleaner #38: Boot delay after removing Intel ME on Thinkpad X220

peterimei, 2017-05-14:
me_cleaner applied successfully and flashed using Raspberry Pi Model B+.

Laptop now experiences what I can only describe as a boot delay from cold boot. During the delay the power LED, wifi indicator, and ThinkLight flash a few times while the screen remains black. After this period the bios screen appears and system seems to boot normally. Issue does not arise during reboot.

I noticed two people in #3 mentioned the same delay/slow boot but there seems to be no resolution so far. Looks like it's some kind of time out.

I decided to update the OEM bios from 1.25 to 1.43 but it made no difference. I'm at a loss.

- CPU architecture: Intel Core i5-2520M
- CPU model: Sandy Bridge
- Laptop: Lenovo Thinkpad X220
- OEM bios

intelmetool output:

```
Bad news, you have a `QM67 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI was hidden on PCI, now unlocked
MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

ME Status   : 0x1e003052
ME Status 2 : 0x1b320002

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : Image Failure
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Power cycle reset through Moff
ME: Progress Phase State    : M0 kernel load

ME: Extend SHA-256: 0311dee9ea06e2ed7a972729ddbefd07e4406ed70f136e3bcca940fa0b705e13

ME: has a broken implementation on your board with this BIOS
ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
Re-hiding MEI device...done
```

lspci output: [lspci.txt](https://github.com/corna/me_cleaner/files/999222/lspci.txt)

DarkCoridor, 2017-05-18:
After doing @corna [suggestion](https://github.com/corna/me_cleaner/issues/30#issuecomment-301197198), I was able to verify. 

- CPU:  Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz
- CPU Model: Sandy Bridge
- Machine: Lenovo Thinkpad X220
- BIOS: Coreboot

And the result of `intelmetool` is:
 
```
Bad news, you have a `QM67 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI was hidden on PCI, now unlocked
MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

ME Status   : 0x1e003052
ME Status 2 : 0x16320002

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : Image Failure
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : M0 kernel load

ME: Extend SHA-256: 69df9d77bfd0c242989cbefae38a11a6b07a33ba73553b792923d48c2e74abc1

ME: has a broken implementation on your board with this BIOS
ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
Re-hiding MEI device...done
```

peterimei, 2017-05-18:
@DarkCoridor how long is your delay? Do you get the same flashes? I was planning to try Coreboot but it sounds like it won't make a difference.

corna, 2017-05-19:
I use a X220 with coreboot and I don't have any boot delay.
Boot delays are generally caused by the BIOS (which waits for the ME to be ready) and using coreboot does make a difference.

Moreover I highly suggest you to use coreboot, as the coreboot port for the X220 is nearly perfect.

DarkCoridor, 2017-05-20:
The delay I notice is a few seconds longer than before I ran `me_cleaner` and reflashed. Also, I noticed about 10x the amount of boot text scrolling by on the screen before my LUKS prompt is displayed. @corna any idea about the text? 

Re: coreboot, when I first flashed it I need to increase the time keyboard drivers are loaded and such.

peterimei, 2017-05-20:
I just compiled and flashed Coreboot and the weird delay is indeed gone. However, it looks like it takes Coreboot longer to boot from SSD than the OEM BIOS (not taking into account the weird delay). Sometimes it takes really long and other times it's marginally slower than OEM. I'd much prefer to just have the OEM BIOS if only I can figure out what is causing the delay. I might play around with the ME modules to see if the removal of one of them is causing it.

corna, 2017-05-21:
With coreboot you can debug the boot time: in the menuconfig select the option "Collect the timestamp during the boot" (or something like that), flash the new build and run "cbmem -t" (the tool is available in the util/cbmem folder).

peterimei, 2017-05-23:
I tried leaving some ME modules in but it simply caused the machine to not boot. Eventually I got fed up waiting for the flashing process to complete each time so I gave up. I have since reflashed the OEM BIOS and will keep it like that for now. I'll try to ignore the delay and see if I can find the instructions that cause the timeout at some point.

javashin, 2017-07-11:
HELLO 

IM getting the same issue here in my lenovo x220 thinkpad same thing if you start the laptop with the power button the laptop start and shutdown quickly i notice this because i used a usb led and then start again and the bios start . any solution ? or i have to compile coreboot ? i used the bios 1.28 unlocked me and removed too ifdtool -u and me_cleaner . and somebody know why now my laptop dont suspend on any os macos sierra windows 10 linux when i close the lid ...... but no ME at all here its all clean .... no pci etc i even dont have problems with the linux kernel i dont have to use iomem=relaxed thanks ..

javashin, 2017-07-11:
hello i want to know where i get coreboot already compiled and working for this laptop like with uefi and mbr bios etc working to boot macos sierra linux and windows 10 and to boot some mbr usbs sticks some times i do use them a lot and a way secure to flash coreboot rom or any other modified bios *.BIN but without using a external programmer just in system i read that flashrom is not stable for this but i dont know never tried and i dont want to brick my laptop because the spi external programmer i ordered comes in one month from china :)

truehomie, 2019-09-25:
@corna 

Hello

I have exactly the same issue 25 seconds delay on initialization on power up (cold boot) on my Acer Swift 3 SF315-52-52YN, i have no IME at all and flash descriptor as well as bios region are write protected.
I bought this laptop, it's not Me cleaner fault

Can anyone advise solution please?