corna/me_cleaner #39: Support for ME 11.x

platomav, 2017-05-16:
I saw you mentioned at the readme that the ME 11.x FTPR structure is not yet know. Actually it has been since November 2015 or so. Of course it's not 100% understood yet like ME 2-10 but it's getting close. I created a wiki page with the structure of ME 11.x FTPR partition which is what me_cleaner targets.

https://github.com/platomav/MEAnalyzer/wiki/Intel-ME-11.x-FTPR-Structure

Also, the Huffman modules can be separated into compressed parts based on their "header".

corna, 2017-05-16:
10 minutes after writing the new README.md I've discovered [this](https://www.troopers.de/downloads/troopers17/TR17_ME11_Static.pdf) presentation, but I had already pushed the commit ;)
Thank you for sharing your knowledge, as usual you are really helpful.

I'd like to work on ME 11 but unfortunately I don't own a Skylake platform: as soon as I obtain one I'll try to understand if me_cleaner can be ported fully on it.

platomav, 2017-05-16:
Ah yes, although the above are not [based](http://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools-72.html#msg15047) on that recent presentation, it is a very cool read and verifies some of the 2015 findings. I read it half a month ago and the guy even mentioned looking into EFFS and 11.x Huffman de-compression so yes, things are looking good.

As for systems to test on, I don't have one either but if you decide to add (initial) support for it, I know someone who can do some tests. There are not a lot of sections for us to modify really, just $CPD and actual module filling since the Manifest is protected. Who knows, maybe it will be as easy as padding sections without any other change. ;)

corna, 2017-06-16:
I've analyzed the ME11 structure and it seems much simpler than ME < 11: all the partitions are consecutive (there shouldn't be the Huffman mess, chunks and similar, that was present before) and every partition seems to be paired with its metadata (which should contain the hash of the uncompressed data, but I didn't check), so it should be easy. The size reported in the CPD refers to the uncompressed data, but we can just use the modules offset as limits for the removal.

I've added (untested) initial support in the _testing_me11_ branch, where every partition is removed (except for BUP), while the metadata is kept. Can you ask him to test it?

platomav, 2017-06-16:
Yes exactly. It is much easier to work with and understand. The CPD Entry offsets (from CPD) are indeed consecutive but, contrary to non-x86 ME, the FPT entries are not (you've probably already noticed that). I hadn't noticed that the CPD module sizes relate to the uncompressed data. For the manifest (man) and metadata (met) they are definitely the same as seen in the blob, which makes sense since these are not compressed. Another idea that I've had is to also remove the (now) unused CPD entries since they are not protected by the signature. I've seen somewhere that the minimum CPD Entry size is 3 so I assume Manifest, 1 Module & 1 Module's Metadata. That works for us since bup is essential so we would have 3 (FTPR.man, bup and bup.met). Obviously the 8-bit CPD checksum would also need fixing in such scenario. For now though, the test needs to prove that the systems boots and stays operational for more than 30 minutes with everything but bup removed. I did a quick code check and also hex comparison at one neutered firmware, seems ok. As for the actual test, I'll ask and get back to you.

corna, 2017-06-16:
> I hadn't noticed that the CPD module sizes relate to the uncompressed data.

The sizes match when the modules are uncompressed, while the reported size is larger than the compressed one when they're Huffman compressed. I can't verify this without the Huffman dictionary, but it seems resonable.

> Another idea that I've had is to also remove the (now) unused CPD entries since they are not protected by the signature. I've seen somewhere that the minimum CPD Entry size is 3 so I assume [...]

But they're listed in the extensions (especially in extension 3, partition info), so I don't know if we can freely change them, even if they're not directly signed. We should investigate.

> Obviously the 8-bit CPD checksum would also need fixing in such scenario.

Yep. Like in the FPT, it's just the two's complement of the header, from $CPD to the end of the last entry (the byte before $MN2).

> As for the actual test, I'll ask and get back to you.

Wonderful, thank you.

corna, 2017-06-17:
@platomav are you sure about the "LZMA only" here?
https://github.com/platomav/MEAnalyzer/blob/master/MEA.py#L500

The compressed size for the bup module seems correct, even if it's Huffman-compressed

skochinsky, 2017-06-17:
here's what the internal FIT XML says:

`<LayoutEntry name="compressed_size"         value="0x0" type="dword"   description="Compressed image size. This is applicable for LZMA compressed modules only. For other modules should be the same as uncompressed_size field." />`

platomav, 2017-06-17:
Indeed that's what the XML says but it is mistaken in this case. Generally these XML have never been updated after first release back in 2014/early 2015 so they have some mistakes or omissions.

The CPD Entry size for Huffman modules corresponds to the decompressed data which is useless to us. However, as you probably noticed, the SizeComp of CPD_Ext_0A corresponds to the actual compressed size found at the firmware without the alignment padding. At ME 11.0 Intel was not aligning the Huffman module sub-sections (MEA -eker parameter) but at latter minor versions that changed. Since the SizeComp field of CPD_Ext_0A hasn't been updated to take into consideration the extra padding found after the latest sub-section of a given Huffman module, you need to a) use SizeComp to detect where the compressed Huffman module ends and b) detect the extra padding at the end which is sometimes FF but others 00.

corna, 2017-06-18:
> Thus, you need to a) use CPD_Ext_0A > SizeComp to detect where the compressed Huffman module ends and b) detect the extra padding at the end which is always(?) FF.

Or I can sort the module offsets and remove the data between the start of a module and the start of the following one, which should be simpler.

> As for the system test, I got a positive reply and it should be possible to do at or after Monday. 😉

Wonderful, thank you.

platomav, 2017-06-18:
That would work but it's no different than taking the next CPD entry's starting offset and cut based on that, as you do now. The only issue is the last module, but that can be solved either as you do currently (last module - end of FTPR) or by using the CPD Entry size, as long as it is uncompressed. Basically, I think CPD_Ext_0A > SizeComp is not really useful for me_cleaner's purpose but it is for tools such as MEA and/or Igor's me_unpack.

corna, 2017-06-18:
That's what I meant, I think the current solution is the best one for me_cleaner.

platomav, 2017-06-20:
I did some more research and so far concluded that:

- The Compression field of CPD_Entry is actually just one bit which is 0 for Uncompressed and 1 for Compressed (no LZMA/Huffman distinction). The "Offset" dword is basically 17 bits for actual Offset (yes, you read correctly --> 3 bytes + 1 bit from the last --> WTF Intel), 1 bit for Compression No/Yes and the rest are reserved. That's why we see only 00 (00 = Compression 0 & 17th Offset bit 0) or 02 (10 = Compression 1, 17th Offset bit 0) at CPD_Entry. Also, for some reason, LZMA modules are shown as "Uncompressed" at CPD_Entry and only Huffman show the expected 02 byte. That by itself causes me_cleaner to detect all non-Huffman modules are "Uncompressed", which is not true. Meaning, you cannot use CPD_Entry to determine if a module is Huffman or LZMA, just Compressed or non-Compressed in general. And that's without the LZMA weirdness mentioned above. For the record, all FTPR Modules (not .man/.met) are Compressed, either Huffman or LZMA. Nothing is Uncompressed at FTPR. At MEA v1.14.0 I now have one dword field called OffsetAttrib which is then "split" to get the Offset & (somewhat-wrong) Compression Yes/No info.
- The Compression field of CPD_Ext_03_Mod does not seem to be used, despite what Intel claims. It is always set to 0 "Uncompressed" which is not true. We can consider it "reserved" for now.
- The Reserved0 field of CPD_Ext_0A is used to determine Module Encryption at 11.6 or newer (0 No, 1 Yes). So far I've only seen it used at 11.6+ pavp module which is used for DRM purposes. I assume that move is related to the recent Neflix-4K/KBL-only deal we were hearing about some time ago. The Module is LZMA compressed and then encrypted. Not related to me_cleaner as it's found at NFTP an not FTPR.
- The SizeComp field of CPD_Ext_0A works as I said before. It stores the accurate Compressed (unlike CPD_Entry) size of each module, without any End-of-Module (EOM) padding, used for alignment. The padding is always FF though. Any sub-section (Huffman compressed module characteristic) padding is 00 but it is included in CPD_Ext_0A.SizeComp as part of the whole module so it doesn't break the previously mentioned FF-only padding rule at EOM.
- The Compression field of CPD_Ext_0A is the only way to truly know the Compression status and type of any given module. 0 is Uncompressed, 1 is Huffman and 2 is LZMA. The other Compression field candidates at CPD_Entry and CPD_Ext_03_Mod are respectively either not reliable or not used.
- By using all of the above info, it is now possible to safely extract for research/testing etc any ME x86 module. I added support for everything in MEA and replaced the old -eker with -emod parameter which shows some info for each FTPR module and extracts it. I've added that for my own research/testing but it may be useful to others as well so I thought I should mention it.

[image]

IllegalArgument, 2017-06-20:
I've been tinkering with the 11.x Huffman compression scheme and have managed to work out most of the symbols for the new dictionaries. I have a simple test script at https://github.com/IllegalArgument/Huffman11 that takes a Huffman compressed module file (including the header section that specifies the offsets of the streams of compressed data) and tries its best to decompress it. At the moment, it's still missing a few symbols used in the rbe, kernel, syslib, and bup modules. For other modules, I have been able to verify that the decompressed output using the incomplete dictionaries is still correct, since the hashes of the decompressed output from those modules match the hashes provided in the corresponding metadata files.

corna, 2017-06-21:
> The Compression field of CPD_Ext_0A is the only way to truly know the Compression status and type of any given module. 0 is Uncompressed, 1 is Huffman and 2 is LZMA.

I'll fix me_cleaner.

> By using all of the above info, it is now possible to safely extract for research/testing etc any ME x86 module. I added support for everything in MEA and replaced the old -eker with -emod parameter [...]

It is very useful also for me, thank you.

corna, 2017-06-21:
@IllegalArgument Good job, keep us updated!

platomav, 2017-06-21:
> I'll fix me_cleaner

Also, line 353 "comp_type = unpack("B", data[0xF])[0]" needs to be changed to use unpack_from instead like this: "comp_type = unpack_from("B", data, 0xF)[0]". At the person's system, who is trying to test the ME11 script, it always crashes at that command with error "TypeError: a bytes-like object is required, not 'int'". It doesn't occur at my own system even if we both run the same Python version of 3.6.1. After some attempts and script changes to pinpoint the issue by guessing, I concluded that the problem has to do with the usage of "unpack" in conjunction with a single byte, 0xF in this case. If for example you use [0xF:0xE] it works but in some (unknown - to me) cases, using bare unpack with a single byte causes a premature conversion to int which does not allow unpack to proceed because it expects a string. Using unpack_from solves that problem. Maybe it's because unpack is expected to be used after using pack whereas unpack_from works with any buffer, I don't know.

@IllegalArgument

Amazing work so far. I didn't even know Scala but once I figured out that it's a runable script, I downloaded the program and got it to work. One issue is that I had to "merge" HuffmanDecompress and HuffmanDictionaries because Scala was not able to detect the latter imported object any other way. I did some tests on Kernel and it works great. I'm wondering, is it possible to have an option to decompress the chunks separately and not create one file in the end? I'm asking because different bup modules may have different chunk sizes so doing comparisons after the fact is harder because of [different offset aligning](https://i.imgur.com/G72dwTD.png).

corna, 2017-06-21:
In Python 3 _unpack_ expects a _bytes_ object, while an element of _bytes_ is a _int_, hence the error. I think (and it actually works for me) that `data[0xf:0x10]` is the correct solution (as it is a _bytes_ object of 1 element), in which cases do you think it doesn't work?

In Python 2 both _data[0xf]_ and _data[0xf:0x10]_ are _str_ objects, no issue there.

platomav, 2017-06-21:
Yes, I believe that data[0xf:0x10] is the correct solution as well when using bare unpack. Unpack_from worked without a range because it starts "from" a certain offset and onward. So in this case it started at 0xF and picked one byte (0xF:0x10 so to speak). Meaning, for Python 3, bare unpack needs to have a "range", even for one byte due to the way it works. I don't understand why it works at some systems and not others but I've asked the person, who had the problem, to test data[0xf:0x10] instead and I'll get back to you.

Regarding the system tests, he did try yesterday one time. The system did start but after many reboots it entered at a BIOS recovery mode. Thing is, the fact that the system even booted after me_cleaner means that the script works and bup is probably the only requirement, as before. In my ME8 tests, the system wouldn't even post if something was wrong with bup. This one entered BIOS recovery so I assume that we're dealing with an OEM recovery and damage control measure here, not an issue with me_cleaner. I've asked him to test the master branch of me_cleaner which keeps FTPR intact to see if the BIOS recovery gets triggered and if that's the case, his system won't be a good candidate to test ME11 me_cleaner.

As for all the CPD_Ext_0A stuff I mentioned above, I know (from the recent MEA commits) that it takes a lot of code to retrieve the correct Compression info so maybe you can do something else at me_cleaner. Using MEA, I tested all 317 ME11 firmware I have and verified that all FTPR Modules are either Huffman or LZMA (no Uncompressed). Since Huffman are always/reliably tagged as 02 (Compressed) at CPD_Entry, you can just show "Huffman" for any 02 module and LZMA for anything 00. At TXE3, all FTPR modules are Uncompressed so (whenever support for that is added) you can always display Uncompressed, without any CPD_Entry check. The trade-off is that these "hacks" rely on observation and may not apply to future firmware.

corna, 2017-06-21:
> Yes, I believe that data[0xf:0x10] is the correct solution as well when using bare unpack. Unpack_from worked without a range because it starts "from" a certain offset and onward. So in this case it [...]

I prefer _unpack_ as it returns an error in case of mismatch between declared size (ex. `"<I"`) and given size (ex. `data[0xc:0x10]`). _unpack_from_ doesn't have this 2-way check and could lead me to introduce alignment bugs.

> Regarding the system tests, he did try yesterday one time. The system did start but after many reboots it entered at a BIOS recovery mode. Thing is, the fact that the system even booted after [...]

This doesn't exclude the 30-min window though, could you ask him to keep the system powered on enough to exclude it?

> This one entered BIOS recovery so I assume that we're dealing with an OEM recovery and damage control measure here, not an issue with me_cleaner.

I agree. However it seems that we were right, the ME11 port will probably be easy.

> I've asked him to test the master branch of me_cleaner which keeps FTPR intact to see if the BIOS recovery gets triggered and if that's the case, his system won't be a good candidate to test ME11 me_cleaner.

Great, thank you. I'm going to buy a ME11 platform just for _me_cleaner_ tests (a MSI B150M PRO-VD M.ATX, very similar to [this](https://github.com/corna/me_cleaner/issues/3#issuecomment-264148613) one).

> As for all the CPD_Ext_0A stuff I mentioned above, [...]

The compression info are not really useful for _me_cleaner_, they were easy to retrieve in ME < 11 so I included them in the output. The hacks proposed seem a good compromise, we have _MEAnalyzer_ for more detailed informations ;)

IllegalArgument, 2017-06-21:
@platomav 
Thanks, glad to hear you're getting good results with it too. If you want to run the script without having to merge the files together, you can compile them first with `scalac -d ./out HuffmanDecompress.scala HuffmanDictionaries.scala` and then run it with `scala -cp ./out HuffmanDecompress <args>` (Scala is a lot like Java in how it's compiled/executed, as would be expected for a language that runs on the JVM :P).

Regarding the option to decompress chunks separately, that would be easy to add, but I don't think it would solve that misalignment problem you're observing. As far as I can tell, each Huffman compressed stream always decompresses to a fixed size of 0x1000 bytes. This way, the first stream is the result from compressing bytes 0x0-0x1000 of the decompressed module, the second stream for bytes 0x1000-0x2000, and so on, effectively just slicing the module into 0x1000 byte chunks. Since the individual streams don't have any sort of "end of stream" marker, different chunk sizes would require the decompressed chunk size to be specified for each stream, but I haven't seen place in the files or their metadata to do this. That's also why the script requires the decompressed size of the module as an argument - it uses that size to determine how many chunks are needed to decompress the full module. My guess is that the misalignment in your screenshot is just due to the two bup versions you have actually having those common strings located at slightly different offsets, probably due to changes in the preceding contents of the module displacing the strings differently.

Another source of alignment issues can be when there are unknown symbols within a chunk. The script always inserts a single byte when it encounters an unknown symbol, but it's likely that the symbol might represent multiple bytes instead. For example, there's a single symbol that represents the hex sequence `FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF`. If the script encountered this symbol and didn't have it in its dictionary, it would write a single `7F` byte for that symbol, causing the rest of the data in that chunk to be shifted earlier by a 14 byte offset. The benefit of writing all of the chunks to a single file is that it prevents this type of alignment error from cascading into subsequent chunks. Since each chunk is 0x1000 bytes decompressed, we know that the compressed stream at index `i` should be decompressed to offset `i * 0x1000` in the module file. By seeking to this offset in the decompressed output before decompressing any of the stream data, the script ensures unknown symbols in a given chunk will only affect the alignment within that chunk and not displace the contents of later chunks. If the misalignment in your screenshot were due to missing symbols in the chunk for the bytes 0x2E000 through 0x2EFFF, and not underlying differences in the bup modules, you would see everything line up again at offset 0x2F000 in the decompressed output.

platomav, 2017-06-21:
> This doesn't exclude the 30-min window though, could you ask him to keep the system powered on enough to exclude it?

It is not possible unfortunately. The mobo stays at a back screen, restarts some times and then the BIOS recovery screen appears which is automatic and restarts once it is done. At this point I'm not even sure if we know that bup is enough to post and stay for 30 minutes. Maybe it does not post but the BIOS somehow catches that and triggers the recovery? Mmm, unlikely. Anyway, we cannot verify if everything works by running tests on that mobo unfortunately. It's great that you'll get a ME11 system for tests, that way you can be certain and do tests at your own pace.

> The compression info are not really useful for me_cleaner [...]

Exactly. me_cleaner only displays that info, it doesn't really use them in any way. So I agree that the hacks are a good compromise. Currently it reports LZMA modules as Uncompressed though, which is wrong, that's why I mentioned it.

@IllegalArgument

> As far as I can tell, each Huffman compressed stream always decompresses to a fixed size of 0x1000 bytes.

You certainly know about compressions etc so bare with me if I'm talking nonsense. When I separate a compressed module via my utility, MEA, I scan the header in order to determine the chunk count. Each dword of the header consists (in BE) of 3 bytes Offset and 1 byte which I don't really know what it signifies but its values are 0,64,128 or 192. By parsing the header in that way (until the last byte is not within those aforementioned values), I end up with a chunk count. I have verified that this method works for all firmware I have. Thing is, the chunk count is not equal Decompressed Size / 0x1000. For example, take [this](http://www.mediafire.com/file/0a1faa7uazf7zxh/Intel_ME_Firmware_v11.10.0.1287_%28CON_H%29.rar) firmware. If I use MEA -emod parameter and store kernel I'll end up with 21 chunks but the uncompressed size is 0x14000. So I have 21 chunks and not 14 as the uncompressed size would dictate. Correct me if I'm wrong of course.

> By seeking to this offset in the decompressed output before decompressing any of the stream data, the script ensures unknown symbols in a given chunk will only affect the alignment within that chunk and not displace the contents of later chunks.

That's a very good idea and I did notice it when checking out bup. I was thinking that the alignment issue could be because a hypothetical similar/almost-same fw1 and fw2 could have chunk 14 of slighly different size, maybe more padding at the first. So if the chunks are decompressed separated then I could just see that the two chunks 14 are the same but the 1st has extra padding at the end. However, that contradicts your hypothesis that number_chunks = uncomp_size / 0x1000 so I don't know.

platomav, 2017-06-22:
@IllegalArgument

Thank you for the extensive reply, I fully understand now.

> the low 25 bits make up the address (though I've never seen the highest address bit used) and the high 7 bits are flags.

Previously, I thought that the flags field was 8 bits and the remaining 24 relative offset. Now it makes sense with the 7/25 arrangement you mentioned as the Dictionary Type is indeed either 0x20 or 0x60.

> I haven't seen any 0x00 or 0x40 flag values in the 11.x firmware yet [...]

To that end, I used MEA and checked all the firmware I have. I can thus verify that there are no other types of Dictionary at FTPR partition. Presumably the same should apply everywhere else for that matter, even if I cannot verify that with my tool.

> I think that decompressedSize = chunkCount * 0x1000 still holds for the kernel module

You are right, it holds true everywhere. For some reason I missed yesterday the (obvious) fact that 0x1000 is a hex number and not dec, thus the math error I made. Indeed, the 20 chunks of that example result in 0x14000 when multiplied by 0x1000 for each uncompressed chunk. My bad.

> [...] in theory the scanning method could attempt to read past the end of the header section.

Exactly. I've now rewritten -emod with many changes. One of which is the proper parsing of the Huffman Module Header by using the Uncompressed Size and the 25/7 arrangement we discussed. An almost same thing can be seen at the CPD_Entry where the Offset dword actually consists of 25 bits Offset, 1 bit Compressed Yes/No and the rest 6 are Reserved. Once again, the highest bit of the Offset field is always 0. In that case though, it's even more ridiculous as the whole 3rd byte is always 0, let alone the LSB of the 4th. Intel really wants that extra bit! 😋

> those bytes are handled no differently than the rest of the compressed data.

Got it now, thanks.

skochinsky, 2017-06-23:
BTW, for ME11 you should also retain RBE (seems to be equivalent of ROMP) in addition to BUP.

corna, 2017-06-26:
@skochinsky Thank you, I'll confirm this once I have the Skylake PC (and some spare time).

corna, 2017-07-20:
I've received my test board (MSI H110 + i3-6300T) and I've started playing with it. So far:
 * with the master branch of me_cleaner (only FTPR left, modules untouched) the PC boots fine (with a delay of 15 seconds, probably a BIOS timeout waiting for ME, and a nice "The ME FW of system was found abnormal")
 * with the testing_me11 branch (all modules removed except for BUP and RBE) the PC doesn't boot at all
 * with the master branch (again, modules untouched) but a manual change on a single byte of the GPIO module (to invalidate the hash of a minor module) the PC boots fine. This should prove that the hashes of the modules are not checked at once, but only as needed, as in ME < 11.

Therefore it seems that either there is a bug in the current implementation of _testing_me11_, or ME11 needs some other module other than BUP and RBE.  I'll keep testing.

corna, 2017-07-21:
Works also with only the 4 Huffman modules left (bup, rbe, kernel and syslib). Yay!

corna, 2017-07-22:
ME 11 port completed! d8bf20d0f29d0add5606952d19414590dc18c113

Thank you all for your help!

closed 2017-07-22 by corna