corna/me_cleaner #42: [HP] Determine what features break OEM RSA signatures and make them optional or adjust to fix

ilikenwf, 2017-06-21:
My HP Spectre x360's bios uses AMI Aptio V, however it also uses a proprietary RSA based .sig file along with the factory .bin bios file.

In the past I had success flashing with me_cleaner (see the issue about the project status), but the current version breaks signature validation resulting in an error message instead of a flash.

Going back, I do find that revision fbc123e28ac852f280fb66be8c872ccb28fe6a24 and earlier versions work, even on the current F.46 bios version (ME 11.0) but the newer ones do something that causes the invalidation of the signature.

At the very least we know that the ME should be crippled/disabled with the older version, but some of the newer enhancements are surely safe, except for any that may change file size or otherwise invalidate the key somehow.

corna, 2017-06-21:
That's strange, as no new commit from fbc123e has introduced substantial changes.

Is fbc123e the last working commit or just one of the working ones?

corna, 2017-06-21:
The flash image on the HP website is compressed in the exe installer, so I can't check things directly. Can you do a binary diff between the image with fbc123e and the one with the latest commit?

ilikenwf, 2017-06-21:
The installer extracts them - once run it gives you the option to copy the
files somewhere, and spits out several .sig and .bin files.

ilikenwf, 2017-06-21:
Here they are extracted:
https://mega.nz/#F!IYEWiRjC!MBAN3JCNjYab9sVYoImeYA

VladimirSchowalter20, 2017-06-21:
I have similar issue with that hp laptop but on broadwell platform and ME 10.x but I can't verify modified bios even with https://github.com/corna/me_cleaner/commit/fbc123e28ac852f280fb66be8c872ccb28fe6a24 revison. I also tried https://github.com/corna/me_cleaner/commit/38fd19b9af71b42b0711d073fbe19cbb788f1433 without success, not sure how far back I can go while keeping me_cleaner usable.

My bios version is 0802D from the above dump.

ilikenwf, 2017-06-22:
fbc123e is just one of the working ones, I've not done a git bisect or anything.

zeepob, 2017-06-22:
@ilikenwf  I'm only saying that it's more complicated. Are you 100% sure me_cleaner actually strips your bios with that revision?

corna, 2017-06-23:
@ilikenwf I've used me_cleaner fbc123e and master on your image and the only difference on the resulting binaries is the checksum (byte 0x1c of the ME region), which was uncorrectly computed in fbc123e and has been fixed in 518445323ba15567a5268c5d14fec0e52b897333

Can you please post the output of `intelmetool` (see the guide [here](https://github.com/corna/me_cleaner/wiki/External-flashing#it-works))?

ilikenwf, 2017-06-26:
The current master still complains about the signature...

METool outputs "device not found" even when mei_me is loaded.

To clarify, of the many spectre x360 models, mine has a i7 6500U Skylake.

zeepob, 2017-06-26:
@ilikenwf  could you post output [./unhuffme](https://github.com/corna/me_cleaner/wiki/Check-the-integrity-of-the-ME-modules) <bios.bin> of your working bios image?

corna, 2017-06-26:
First I need the output of intelmetool. Which version are you using? You should use the intelmetool included in the latest revision of coreboot.

ilikenwf, 2017-06-26:
I grabbed it from their master branch but can try again, I suppose.

Unhuffme:

`./unhuffme ../bios/0804F.bin                                                                                [06/26 10:00 PM] number of flash regions: 1
        region: 0 in range: 00000fff-00000000

Flash partition table (1 entries):
        partition:  FTPR(type:0) at 00001000, size:a7000`

ilikenwf, 2017-06-28:
Same results. The mei module is loaded but intelmetool doesn't seem to get
anything out of the management interface. I guess whatever's going on it's
sort of bricking the ME and forcing it to be locked? I say that because a
flashrom version patched for Skylake isn't able to dump the rom due to read
errors, and complains about the ME being likely cause.

As an aside, I have the output from flashrom and also found the schems for
my laptops motherboard...tempts me to delve into actual porting of coreboot
someday if I get brave.

ilikenwf, 2017-06-28:
Same results. The mei module is loaded but intelmetool doesn't seem to get
anything out of the management interface. I guess whatever's going on it's
sort of bricking the ME and forcing it to be locked? I say that because a
flashrom version patched for Skylake isn't able to dump the rom due to read
errors, and complains about the ME being likely cause.

```
Warning: BIOS region SMM protection is enabled!
Warning: Setting Bios Control at 0xdc from 0xaa to 0x89 failed.
New value is 0xaa.
Warning: SPI Configuration Lockdown activated.
FREG0: Warning: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG2: Warning: Management Engine region (0x00001000-0x001fffff) is locked.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
PR1: Warning: 0x00640000-0x007fffff is read-only.
PR2: Warning: 0x00200000-0x0021ffff is read-only.
PR3: Warning: 0x0063f000-0x0063ffff is read-only.
Transaction error between offset 0x00001000 and 0x0000103f (= 0x00001000 + 63)!
Read operation failed!
```

As an aside, I have the output from flashrom and also found the schems for
my laptops motherboard...tempts me to delve into actual porting of coreboot
someday if I get brave.

https://pastebin.com/gsvK8ccj
https://github.com/corna/me_cleaner/files/1107669/HP_X360_Y0D_PIKE_PV_VPRO_ELP4G_r1a.pdf

corna, 2017-06-28:
> Same results.

Strange... Does it also fail with an unmodified Intel ME firmware?
Can you post the output of `lspci`?
@zamaudio any idea?

Can you give me the first 48 bytes of the original ME image and of both the modified images (with old _me_cleaner_, working, and with the master revision, not working)?

> [...] tempts me to delve into actual porting of coreboot
someday if I get brave.

I did the port for the [Sapphire Pure Platinum H61](https://review.coreboot.org/#/c/18564/) board and it was really interesting, I highly suggest you to give a try. Just a couple of notes:
 * You're on a laptop, so you have to reverse-engineer the EC
   * Which probably means, according to the schematic, no fan/temperature control, no capacitive home button and (probably) no advanced keyboard/touchpad support until you reverse it
 * You're on Skylake, so you have to use the FSP blob provided by Intel (no native raminit yet)

persmule, 2017-06-28:
@mytbk , for he is working on porting coreboot to hp laptops with sandy/ivy bridge cpu.

ilikenwf, 2017-06-28:
@corna that was actually the next thing I want to test, using stock firmware. I'll try to get around to that this evening.

I think I should be able to extract the ME image, I'll just have to figure out how to use MMTool to do so.

Re: coreboot - I'm still absorbing as much research material as I can, while also looking at the sources available. I'm pretty certain very similar hardware is used in the Chell/HP Chromebook 13 G1 as the Spectre, though, beyond the CPU difference. 

I do have an ASRock desktop board I could start porting on first if you think that'd be wiser? Either way, after I wrap my head around it I'll want to figure out the best way to setup an unbricking procedure.

And yes, the EC does control those things it seems. My guess is that the native raminit will come once more mainstream machines become compatible, to the point people want/demand fully open support?

@persmule is he working on HP laptops in general for Skylake, or for the raminit?

mytbk, 2017-06-28:
@ilikenwf I'm just working on coreboot porting on HP Sandy/Ivy Bridge Elitebooks (with autoport), and I'm still struggling on the EC interface. https://review.coreboot.org/q/topic:%2522coreboot_on_hp%2522+(status:open%20OR%20status:merged)

ilikenwf, 2017-06-28:
You're the same guy I saw on hackaday?

mytbk, 2017-06-28:
Ah, yes. But I don't use hackaday often, I was just finding people who can help me with this.

ilikenwf, 2017-06-28:
Well, I'm not sure helpful I am on the porting/hardware side yet but I'm pretty good at locating resources - schems and sometimes proprietary code for research.

corna, 2017-06-28:
> I think I should be able to extract the ME image, I'll just have to figure out how to use MMTool to do so.

You can do it with `ifdtool` (again, from the coreboot repo)

```
$ ifdtool -x dump.bin
$ dd if=flashregion_2_intel_me.bin of=header.bin bs=1 count=48
``` 

`flashregion_2_intel_me.bin` (or something like that) is the full ME image, `header.bin` is just the header

ilikenwf, 2017-06-28:
I like that better as I tire of Windows and it's evil ways. Many thanks!

zamaudio, 2017-06-30:
Your .sig thing seems to be unrelated to the management engine RSA signature.  I don't support flashing from a vendor tool that may have its own locks/checking (?)

ilikenwf, 2017-06-30:
Not everyone wants to disassemble their laptops.

corna, 2017-07-03:
I've just noticed that coreboot's _intelmetool_ doesn't support Surise Point (which you probably have). Can you post the output of `lspci -nn`?

ilikenwf, 2017-07-03:
I believe this is a likely case, when I bought the laptop it was listed as a Broadwell but I was disappointed to find it was the newer Skylake.

Like the tool for dumping bios there may be some gerrit patches that add support?

corna, 2017-07-03:
They're probably just a couple of lines, like [this commit for Wildcat Point](https://github.com/coreboot/coreboot/commit/8c247a2a796f158664d3f1bba130c48bf4e1497c). Once I get my Skylake PC I'll push a change on gerrit.

skochinsky, 2017-07-08:
> I just want to clarify, in case bios is signed then it's supposed to pass with OEM updater after me_cleaner is used?
This depends on how the vendor implements it. If they check only the BIOS region, then it "should work", if the check the whole rom, then obviously ME changes will invalidate the signature.

VladimirSchowalter20, 2017-08-06:
I did some research and it seems there is no way to flash modified bios in HP notebook using vendor tools. I guess @ilikenwf  bios wasn't actually stripped as me_cleaner didn't support ME11 when he tried it first time.

ilikenwf, 2017-08-07:
I believe full removal of 11 wasn't supported but disabling by breaking the
ME still was possible?

VladimirSchowalter20, 2017-08-07:
Did you compared ME output before and after flashing with me_cleaner? In my opinion every change in bios.bin file should break RSA signature. I tried it pretty excessively on Broadwell with many me_cleaner versions without success.

ilikenwf, 2017-08-07:
It's been quite some time but I think I did, yes.

skochinsky, 2017-08-07:
@VladimirSchowalter20 
>I did some research and it seems there is no way to flash modified bios in modern HP notebook using vendor tools.

It seems at least _some_ AFUDOS versions had an undocumented switch (`/gan`) which bypasses the signature check. See here for more info: http://www.win-raid.com/t286f16-Guide-Flashing-modified-AMI-Aptio-UEFI-using-AFU.html

It is of course possible that this has been fixed in recent BIOSes but you could still try it :)

ilikenwf, 2017-08-07:
> modified-AMI-Aptio-UEFI-using-AFU.html

I'll have to investigate.

ilikenwf, 2017-08-07:
AFU typically does not work on these laptops, though.

ilikenwf, 2017-09-18:
I recently got a couple Pomona clips and a RasPI, and after getting my feet wet with an old Thinkpad T400 (the images I built have some bugs so I used Libreboot's binary), I dumped, modified, and reflashed the Spectre's current ROM, which is the latest version, albeit stripped of the ME and with the lock bits unset, amongst a few other things.

I failed miserably, and reflashed the default. Then, I tried the oldest version I could find. Interestingly, this lost the serial number, half my RAM, but popped up a message about being in manufacturing mode, whcih was interesting.

Anyway, I then opened the current unmodified rom in a hex editor and found that aside from the pubkeys that are validated each boot, that the TPM default keys in the ROM are there, with a URL to download some of the crl files. That got me thinking, and then, I once again flashed my modified rom.

Upon doing so, and letting the normal update process that occurs after a flash go on, I got the usual error message - "Boot image did not authenticate." I then just hit F10 to get into settings after a reboot, and turned off the TPM and cleared the platform keys. After a reboot, this:

[image]

So, it seems to me that validation of the bios image itself at boot occurs only by the HP internal flashing mechanism, but that because modded bios breaks secureboot (correct me if I'm wrong there), that other than SPI flashing which is easy on this machine as the chip is right under the cover, is that you need to nuke the platform keys and either use your own or go without.

As an aside, the HP platform key CRL, for example: 
onsitecrl.verisign.com/HewlettPackardCompanyDeSPrintingDeviceCSIDTemp/LatestCRL

[image]

VladimirSchowalter20, 2017-09-18:
Nice work! Thank you for reporting this. I think it also confirms that there is no way to flashing modded bios through HP update tool.

ilikenwf, 2017-09-19:
Thanks, I agree, unfortunately. This also includes AFU.

Now I have to decide if I want to try to learn coreboot porting, or if I
just want to remove the network stack drivers from the rom and call it good
enough...

ilikenwf, 2017-09-22:
So being that I'm pretty bad at low level software work I did autoport and then cribbed a bit off of purism's Skylake work (read: copy/paste).

I did manage to build a rom after grabbing the various blobs I needed but the laptop seems to be giving me blink codes...which is either the CPU not being brought up, or the bios not starting in time, or something to that effect.

While I will continue using the ME cleaned factory UEFI for now, I wonder if anyone knows where I shoulld go from here?

ilikenwf, 2017-09-25:
Closing, as me_cleaner's work here is finished.

closed 2017-09-25 by ilikenwf

ilikenwf, 2017-09-29:
@VladimirSchowalter20 @mytbk  @persmule just for reference/talk, it does seem that the HP bios update tool includes some sort of modified version of AFIUWIN, but I don't know how we could separate it from the included tool. 

I notice in a hex editor that some of the strings include "engineering mode" and other things, along with switches. I think they're just using AMI's signing tool, so if we could find a leaked version of that it may be possible to bypass entirely, otherwise we'll need to disable the safeguards in place or figure out how these tools work so we can add the functionality to flashrom.

Also included, of course, are the efi binaries and signatures used by hp. I wonder if we could modify those somehow, unless the system board itself somehow checks the signatures too?

If they're doing "normal" efi signing and using it as the basis for all these signatures, could we just use the signing tool to sign all of our files and/or replace the default keychain somehow? Just a thought..

Latest bios for spectre:
http://ftp.hp.com/pub/softpaq/sp80501-81000/sp80709.exe

Likewise, the attached zip I extracted from another board's bios, but this tool should allow flashing from Windows if you use official bin/sig files. It can at least detect my board, amongst other things, so I would guess it would be another avenue to explore to reverse engineer or bypass sigs. Bios roms and sigs have to go inside a .cab file for it to be able to flash them.

[hpqflash.zip](https://github.com/corna/me_cleaner/files/1343145/hpqflash.zip)

ilikenwf, 2017-09-29:
I just found the AMI tool used to create the installer; I can extract the constituent files but the executables are DOS only and I've not yet tested them.

[AMIUCP.zip](https://github.com/corna/me_cleaner/files/1344842/AMIUCP.zip)

platomav, 2017-10-02:
@ilikenwf @corna 

To avoid issues, the attached file above should be removed as AMIUCP is given under NDA and AMI tends to send C&D/DMCA orders.

corna, 2017-10-02:
Removed, thanks for pointing it out.

ilikenwf, 2017-10-02:
Ah, in that case, I may know of a site or two where said file is hosted, in
the event someone needs it for providing anonymous work to coreboot,
me_cleaner, or to fix their machine. Just email me for details.

ilikenwf, 2017-12-08:
I read an article a while back about how most UEFI bios protections were a joke...turns out the same goes for HP if you have an SPI programmer.

While you must use an SPI clip to do this, I have found that if one removes a handful of UEFI modules that you can run a modified bios without the forced reflash of a signed unmodified one, so not only can we disable the ME, but we can now once again hack our own AMI bioses, at least on HP laptops.

Since this is just an issue, I'll not do a proper writeup, but will get around to it eventually.

Steps:
1. modify the bios options with certain tools we can't discuss here, also enabling the ME reflash bits
2. remove the modules below with UEFITool
3. me_cleaner
4. use a clip to reflash

In the meantime, the modules I removed (several were just due to personal preference):

```
ReFlash
CryptoPkgTest
HardwareSignatureEntry
BiosGuardDxe
BiosGuardNvs
CryptoDXE
BiosGuardServices
CryptoSMM
OpalSecurity
CryptoPPI
HpDevFwUpdate
HP_BBUpdate_Loader
HP_BBUpdate

All IP stack drivers and protocols
Computrace image (search non unicode strings)
```

ilikenwf, 2017-12-08:
Note that removal of the Crypto module(s) - probably PPI? Breaks secureboot. I can't say I've shed a tear over that one yet.