corna/me_cleaner #43: Coreboot/SeaBIOS/me_cleaner on Thinkpad X1 Carbon 1st gen

elpinguinoatgmxch, 2017-06-22:
Hi, I tried to install Coreboot with SeaBIOS and neutralized Intel ME on a Thinkpad X1 Carbon 1st gen and get the following issue: the keyboard briefly lights up and the fan starts spinning, but the laptop doesn't boot. I then restored the lower 8 MB of the original BIOS (which, if I'm correct, contains the Intel ME) and got the same result. I finally had to restore also the upper 4 MB of the original firmware image to be able to boot the machine up again. Maybe the problem is that I have Intel Boot Guard enabled or I did something wrong when compiling Coreboot. How can I find out if I have the Boot Guard activated? I tried intelmetool -s but didn't find any message about that?!

corna, 2017-06-23:
Intel Boot Guard is available only from Haswell onwards, so you don't have it (and Intel Boot Guard protects the BIOS, so coreboot doesn't work at all with it).

Can you please post the output of `ifdtool -d` on the dumps of the two chips? One should fail, it's normal.

elpinguinoatgmxch, 2017-06-24:
Thank you - I didn't realize that Intel Boot Guard exists only from Haswell on, but this is good news: if I understand correctly, this means that it should be possible to make run coreboot with a neuted Intel ME on this machine! I'll try again as soon as possible. For the moments here are the dumps (you said that one should fail, but to me it seems as if it works for both parts?!):

ifdtool -d x1_factory_left_4MB.rom 
File x1_factory_spi0_left_4MB_2.rom is 4194304 bytes
FLMAP0:    0x5ec0940f
  NR:      6
  FRBA:    0xc00
  NC:      1
  FCBA:    0xf0
FLMAP1:    0x5ec032c3
  ISL:     0x5e
  FPSBA:   0xc00
  NM:      2
  FMBA:    0xc30
FLMAP2:    0x026856c3
  PSL:     0x6856
  FMSBA:   0xc30
FLUMAP1:   0xfe81660e
  Intel ME VSCC Table Length (VTL):        102
  Intel ME VSCC Table Base Address (VTBA): 0x0000e0

ME VSCC table:
  JID0:  0x71737aa1
    SPI Componend Device ID 1:          0x73
    SPI Componend Device ID 0:          0x7a
    SPI Componend Vendor ID:            0xa1
  VSCC0: 0x38d31539
    Lower Erase Opcode:                 0x38
    Lower Write Enable on Write Status: 0x06
    Lower Write Status Required:        No
    Lower Write Granularity:            1 bytes
    Lower Block / Sector Erase Size:    64KB
    Upper Erase Opcode:                 0x15
    Upper Write Enable on Write Status: 0x06
    Upper Write Status Required:        Yes
    Upper Write Granularity:            1 bytes
    Upper Block / Sector Erase Size:    4KB
  JID1:  0x47176779
    SPI Componend Device ID 1:          0x17
    SPI Componend Device ID 0:          0x67
    SPI Componend Vendor ID:            0x79
  VSCC1: 0x62064e56
    Lower Erase Opcode:                 0x62
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            64 bytes
    Lower Block / Sector Erase Size:    8KB
    Upper Erase Opcode:                 0x4e
    Upper Write Enable on Write Status: 0x06
    Upper Write Status Required:        No
    Upper Write Granularity:            64 bytes
    Upper Block / Sector Erase Size:    8KB
  JID2:  0xf0acbddb
    SPI Componend Device ID 1:          0xac
    SPI Componend Device ID 0:          0xbd
    SPI Componend Vendor ID:            0xdb
  VSCC2: 0xa2a0a4a9
    Lower Erase Opcode:                 0xa2
    Lower Write Enable on Write Status: 0x50
    Lower Write Status Required:        No
    Lower Write Granularity:            1 bytes
    Lower Block / Sector Erase Size:    256 Byte
    Upper Erase Opcode:                 0xa4
    Upper Write Enable on Write Status: 0x50
    Upper Write Status Required:        Yes
    Upper Write Granularity:            1 bytes
    Upper Block / Sector Erase Size:    4KB
  JID3:  0xefb129e8
    SPI Componend Device ID 1:          0xb1
    SPI Componend Device ID 0:          0x29
[... 415 lines cut ...]
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x0000

FLMSTR3:   0x08080118 (GbE)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x0118

Found Processor Strap Section
????:      0x00000000
????:      0xffffffff
????:      0xffffffff
????:      0xffffffff

corna, 2017-06-24:
It didn't fail, but the output for the 4MB one is just garbage (as expected).

What is the value of `CONFIG_CBFS_SIZE` in your coreboot build?

elpinguinoatgmxch, 2017-06-25:
Hi, CBFS file size is 0x100000.

corna, 2017-06-25:
_me_cleaner_ didn't work on the Thinkpad x13e (#13), and also in that case there were 2 flash chips concatenated. Can you try to switch to a single flash layout?

Don't use _me_cleaner_ yet.

```
$ cat x1_factory_right_8MB.rom x1_factory_left_4MB.rom > x1_factory_combined.rom
$ ifdtool -f layout.txt x1_factory_combined.rom
```

Change the bios region in `layout.txt` from `00500000:00bfffff` to `00500000:007fffff`

```
$ ifdtool -n layout.txt x1_factory_combined.rom
$ ifdtool -d x1_factory_combined.rom.new > /tmp/before
```

Now open x2_factory_combined.rom.new in a hex editor ad change bytes 0x14:0x17 (which should be 0x03**01**0403) to 0x03**00**0403

```
$ ifdtool -d x1_factory_combined.rom.new > /tmp/after
```

`diff /tmp/before /tmp/after` should show only a difference on FLMAP and NC (Number of Components):

```
-  NC:      2
+  NC:      1
```

If so, run `ifdtool -x x1_factory_combined.rom.new` and use the extracted descriptor for the next build of coreboot. The resulting coreboot binary should be now 8 MB instead of the usual 12 MB.

Note that I never tested these steps and you may need to restore the original dump with an external programmer.

As usual, after the programming completely disconnect any power source and wait a couple of seconds before powering on the system.

elpinguinoatgmxch, 2017-06-26:
Thank you - did all that (see CL + outputs at the end). I then compiled coreboot adding the bios vga blob and choosing 12MB and 8MB of coreboot build size. My question now: (1) which one do I flash now and (2) to which chip?

I wanted to point out that on the coreboot wiki-page (https://www.coreboot.org/Board:lenovo/x1_carbon_gen1) combining the two parts into one "12MB virtual flash image" is described the other way arround (first 4MB, then 8MB instead of 8MB plus 4MB as you describe it). I suppose that this is an error on the coreboot wiki page?!

As soon as I know which image I should use and which chip flash I will try that (now problem if it doesn't work and I have to restore the original BIOS with an external programmer).

So here's what I did:
cat x1_factory_right_8MB.rom x1_factory_left_4MB.rom > x1_factory_8MB_4MB_combined.rom
ifdtool -f layout.txt x1_factory_8MB_4MB_combined.rom 

File x1_factory_8MB_4MB_combined.rom is 12582912 bytes
Wrote layout to layout.txt

vi layout.txt 
(change bios region from 0x00500000:00bfffff to 0x00500000:007fffff)

ifdtool -n layout.txt x1_factory_8MB_4MB_combined.rom 

File x1_factory_8MB_4MB_combined.rom is 12582912 bytes
DANGER: Region BIOS is shrinking.
    The region will be truncated to fit.
    This may result in an unusable image.
The image has changed in size.
The old image is 12582912 bytes.
The new image is 8388608 bytes.
Copy Descriptor 0 (Flash Descriptor) (4096 bytes)
   from 00000000+00000000:00000fff (      4096)
     to 00000000+00000000:00000fff (      4096)
Copy Descriptor 1 (BIOS) (3145728 bytes)
   from 00500000+00400000:00bfffff (   7340032)
     to 00500000+00000000:007fffff (   3145728)
Copy Descriptor 2 (Intel ME) (5230592 bytes)
   from 00003000+00000000:004fffff (   5230592)
     to 00003000+00000000:004fffff (   5230592)
Copy Descriptor 3 (GbE) (8192 bytes)
   from 00001000+00000000:00002fff (      8192)
     to 00001000+00000000:00002fff (      8192)
Writing new image to x1_factory_8MB_4MB_combined.rom.new

ifdtool -d x1_factory_8MB_4MB_combined.rom.new > /tmp/before

hex edit: change byte offset 0x15 to 00 (instead of 01)

ifdtool -d x1_factory_8MB_4MB_combined.rom.new > /tmp/after
diff /tmp/before /tmp/after

2c2
< FLMAP0:    0x03040103
---
> FLMAP0:    0x03040003
5c5
<   NC:      2
---
>   NC:      1

ifdtool -x x1_factory_8MB_4MB_combined.rom.new 

File x1_factory_8MB_4MB_combined.rom.new is 8388608 bytes
  Flash Region 0 (Flash Descriptor): 00000000 - 00000fff 
  Flash Region 1 (BIOS): 00500000 - 007fffff 
  Flash Region 2 (Intel ME): 00003000 - 004fffff 
  Flash Region 3 (GbE): 00001000 - 00002fff 
  Flash Region 4 (Platform Data): 00fff000 - 00000fff (unused)

ls -l

-rw-r--r-- 1 tux tux     4096 Jun 26 11:31 flashregion_0_flashdescriptor.bin
-rw-r--r-- 1 tux tux  3145728 Jun 26 11:31 flashregion_1_bios.bin
-rw-r--r-- 1 tux tux  5230592 Jun 26 11:31 flashregion_2_intel_me.bin
-rw-r--r-- 1 tux tux     8192 Jun 26 11:31 flashregion_3_gbe.bin
-rw-r--r-- 1 tux tux       87 Jun 26 11:20 layout.txt
-rw-rw-r-- 1 tux tux 12582912 Jun 26 11:18 x1_factory_8MB_4MB_combined.rom
-rw-r--r-- 1 tux tux  8388608 Jun 26 11:28 x1_factory_8MB_4MB_combined.rom.new
-rwxrwxr-x 1 tux tux  4194304 Jun 26 11:17 x1_factory_left_4MB.rom
-rwxrwxr-x 1 tux tux  8388608 Jun 26 11:17 x1_factory_right_8MB.rom

mkdir ../../../3rdparty/blobs/mainboard/lenovo/x1_e
cp flashregion_0_flashdescriptor.bin ../../../3rdparty/blobs/mainboard/lenovo/x1_e/descriptor.bin
cp flashregion_1_bios.bin ../../../3rdparty/blobs/mainboard/lenovo/x1_e/bios.bin
cp flashregion_2_intel_me.bin ../../../3rdparty/blobs/mainboard/lenovo/x1_e/me.bin
cp flashregion_3_gbe.bin ../../../3rdparty/blobs/mainboard/lenovo/x1_e/gbe.bin

corna, 2017-06-26:
I don't see any reference on the order of the chips in the coreboot page and, according to your report, the first chip is definitely the 8MB one (the region offsets and sizes are correct there, while the 4MB reports just garbage).

Maybe you're referring to the fact that they suggest to flash only the 4MB one? That's because the bootblock (the first piece of code that gets executed) is located at the end of the BIOS region (and you can see it in coreboot by running `build/cbfstool build/coreboot.rom print` from the coreboot directoty) so replace just the end of that region is sufficient to change the BIOS.

Now set CONFIG_COREBOOT_ROMSIZE_KB_8192 (Mainboard -> Rom chip size), use a small CONFIG_CBFS_SIZE (1MB is ok) and use the new descriptor.bin and me.bin for the next build. Build and flash the resulting `build/coreboot.rom` on the 8MB chip.

Before flashing you may want to:
 * run `ifdtool -d build/coreboot.rom` and check that the output is similar to the previous ones
 * run `build/cbfstool build/coreboot.rom print` and check that it prints the coreboot files
 * run `me_cleaner.py -c build/coreboot.rom` and check if the signature verification succeeds

elpinguinoatgmxch, 2017-06-26:
Thanks again for taking your time to look at this problem. 

First of all: I referred to the way the two parts are concatenated to a virtual flash image with the cat command:

(1) In the coreboot wiki (https://www.coreboot.org/Board:lenovo/x1_carbon_gen1) it's done 4MB + 8MB = 12MB:

`cat x1c_MX25L3273E_original_bios.bin x1c_MX25L6406E_original_bios.bin > x1c_original_bios.bin`

(2) In this (https://steemit.com/tutorial/@joeyd/run-don-t-walk-from-the-blob) tutorial for the x230 (which also has two flash chips), it's done 8MB + 4MB = 12MB:

`cat fpiemgde.rom top.rom > x230-bios.rom
`

I've sucessfully used the method (2) for an x230 (working with a 12MB for all steps and splitting them up again before flashing) and just wondered if method (1) on the coreboot wiki is correct (I don't think so?! and I've used method (2) for my x1 builds)

PHYSICALLY (= on the board and according to the offsets), there is no doubt that the 8MB chip is the „first“ or „lower“ one.

Then, second: I compiled coreboot with the 8MB option (as you said) and flashed it to the 8MB chip. The results are the same as at the beginning: the keyboard lights up briefly and the computer doesn't boot. I also compared the output of `ifdtool -d` to the older builds (they are exactly identical, a part from the values I've changed manually). `cbfstool print` shows all the coreboot files (the list is the same, just some minor changes here and there in the size) and `me_cleaner.py -c` gave me:

```
Full image detected
The ME/TXE region goes from 0x3000 to 0x4ff000
Found FPT header at 0x3010
Found 23 partition(s)
Found FTPR header: FTPR partition spans from 0x183000 to 0x24d000
ME/TXE firmware version 8.1.0.1265
Checking FTPR RSA signature... VALID
```

I can send you the corresponding dumps if it helps. I can also try other options if you have any other suggestions.

elpinguinoatgmxch, 2017-06-27:
In order to simplify things I applied `me_cleaner` to the original stock bios and it works:

```
sudo ./intelmetool -s

Bad news, you have a `QS77 Express Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI was hidden on PCI, now unlocked
MEI found: [8086:1e3a] 7 Series/C210 Series Chipset Family MEI Controller #1

ME Status   : 0x1e003052
ME Status 2 : 0x1632015a

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : Image Failure
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : M0 kernel load

ME: Extend SHA-256: 099b87484854301020c4c8cb95854b0c080ee7acf6d701104e20c7a3d7583410

ME: has a broken implementation on your board with this BIOS
ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
Re-hiding MEI device...done
```
The problem must be in my coreboot build.

corna, 2017-07-02:
It seems so.
Do you have a debugger for coreboot? EHCI debugger (like C.H.I.P., Raspberry Pi Zero, any Linux device with a device-mode USB port...), on-board serial port, OXPCIe952, spkmodem (so slow...),... If so we can obtain a early log from coreboot and try to debug the problem.

elpinguinoatgmxch, 2017-07-03:
Thx - the only thing I have for the moment is an x60 with a serial port in the docking station (but don't think that helps?!) I've read somewhere that there exist dongles that can be used for EHCI debugging? I've searched in the internet but couldn't find anything - maybe you can recommend something that works well?

corna, 2017-07-03:
It doesn't, you need a serial port on the x1 carbon (no add-on boards). Otherwise you can use an EHCI debugger, basically any Linux device with a device mode USB port (also called OTG, usb gadget...). You can find some info [here](https://www.coreboot.org/EHCI_Gadget_Debug). I've used a [C.H.I.P.](https://getchip.com/), but also Raspberry Pi Zero or any device listed in that page is OK, you just have to run `cat /dev/ttyGS0`, connect the USB of your debugger to a specific physical port on the laptop and power on the laptop.

You might need to recompile the kernel and enable a couple of options, as noted in the linked page.

elpinguinoatgmxch, 2017-07-03:
Ok, I'll try to get a Rasperry Pi Zero (normal or W) with an OTG-Adapter (if I understand correctly this is mini-USB to normal USB). How do I power-up the RPI? I mean: of course it get's power when I connect it to the X1C (via the OTG-adapter), but the problem is (I think) that it should already be runnig when I power on the X1C in order to grab the output from coreboot via `cat /dev/ttyGS0` ?! Can I connect the normal power cable plus the OTG-adapter?

corna, 2017-07-03:
Good point. The schematic for the RPi Zero isn't complete, but it reports that the power pin of the "data" micro USB connector is connected to VBUS, while the power pin of the "power" micro USB connector is connected to 5V, but the relationship between these two lines is unknown.

Some laptops provide USB power also when powered off, otherwise I think that the safest solution is to cut the power cable in the USB cable, and power the RPi Zero through the power USB connector.

elpinguinoatgmxch, 2017-07-14:
So I got the Rasperry Pi Zero W, set it up with this (https://gist.github.com/gbaman/50b6cca61dd1c3f88f41) guide and got it listening to /dev/ttyS0 (I think) using g_serial module and `cat /dev/ttyS0`. I connected the „target“-computer (= the computer to debug - I tried an x200 with Libreboot and an x230 with Coreboot) to the OTG-USB-port (= 2nd port) of the RPi via a modified cable (where only D+/D- = white and green cables are connected, i.e. power supply via red and black cable is disconnected). For the moment I get no output from /dev/ttyS0. 

My questions:
(1) Should I connect power supply (red/black cable in USB-connection)? As we've discussed earlier, the RPi get's power via the other (1st port) USB-min-connector (and I don't want to produce any short-circuits and destroy the RPi if possible ...)
(2) Is debugging mode (= output via serial console / usb) enabled in Coreboot (or Libreboot) by default or do I have to enable it somewhere?

A part from that: does anybody have any other ideas what could be wrong with my setup and what I should do to make it work (recompile kernel, load g_dbgp module)?

corna, 2017-07-14:
You need the _g_dbgp_ module, not _g_serial_, which unfortunately isn't included in the Raspberry Pi kernel, so you'll need to recompile the kernel (follow [this guide](https://www.coreboot.org/EHCI_Gadget_Debug), the serial port will be named _/dev/ttyGS0_).

(1) Leave them unconnected, it's the safest option.

(2) It's not enabled by default, you can enable it with
 * Generic drivers
   * USB 2.0 EHCI debug dongle support (`CONFIG_USBDEBUG`): **y**
     * Enable early (pre-RAM) usbdebug (`CONFIG_USBDEBUG_IN_ROMSTAGE`): **y**
     * Index for EHCI controller to use with usbdebug (`CONFIG_USBDEBUG_HCD_INDEX`): **board-specific**
     * Default USB port to use as Debug Port (`CONFIG_USBDEBUG_DEFAULT_PORT`): **0**
     * Type of dongle (Net20DC or compatible) (`USBDEBUG_DONGLE_STD`)
 * Console
   * USB dongle console output (`CONFIG_CONSOLE_USB`) : **y**

gnustomp, 2017-07-21:
I seem to be running into this issue on the ThinkPad X131e, which I'm trying to bring up a coreboot port on. Like #13, but with a SandyBridge CPU.

I'm unable to get a good read on the 4MB flash chip (EN25Q32B), so for now I've modified the flash descriptor to only use the 8MB flash chip (EN25Q64). But the ME takes up all of that, so I used me_cleaner to shrink that down.

The power LED will flash once when attempting to start the machine the first time when connected to power, then it will reset and boot coreboot. But coreboot halts due to an ME error.

`WARNING: ME has bad firmware` also shows up early in the coreboot log.

Some logs:
[me_cleaner.log](https://gist.github.com/gnustomp/fd88c2e012d6061c3b0c72afb49a68b8)
[last part of coreboot.log](https://gist.github.com/gnustomp/43b5114e564ba5d489132c3f36ddf5bc)
[original ifd layout.txt](https://gist.github.com/gnustomp/c624b80a4bc7c7bd66113c6438f48f68)
[modified ifd layout.txt](https://gist.github.com/gnustomp/9dc16a984f3f6ea1212dff59e875a3d1)

corna, 2017-08-03:
@gnustomp Before trying with me_cleaner I suggest you to make a good coreboot port. If ME takes too much space (but coreboot should fit in less than 1 MB and the ME firmware is at maximum 5.5 MB) you can use a non-AMT firmware downloaded from somewhere (~1.5 MB): as it's an official non-hacked image, it should always work. Moreover I suggest you to modify the layout, apply it, extract the descriptor from the new image and use it for the next build, as sometimes ifdtool messes up the flash partitions.

@elpinguinoatgmxch I've looked at the _g_dbgp_ module in the Linux kernel and I've noticed that it uses the maximum current limit set by `USB_GADGET_VBUS_DRAW` (which on Raspbian, is set to 2 mA), therefore it should be perfectly safe to connect the power lines of the USB.

elpinguinoatgmxch, 2017-08-14:
Thx a lot for your comments and sorry for my late answer (I have been quite busy the last weeks). To make a long story short: I've tried - but I'm stuck. 
(1) no (or at least too little) experience with crosscompiling (especially for the arm plattform, so I'm confused between arm-linux-gnueabi- and arm-linux-gnuebahihf-)
(2) no experience with applying the patches (which are for specific kernel versions)
So, for the moment - and for lack of time (and knowledge) - it seems that I have to leave this unfinished (but I'll eventually come back if enlightment strikes me ... :)

corna, 2017-09-11:
@elpinguinoatgmxch 

OT: see this [blog post](https://blogs.coreboot.org/blog/2017/05/08/announcing-coreboot-4-6/) (last section, **Code removal after the 4.8 release**). From the [Supported Motherboards page](https://www.coreboot.org/Supported_Motherboards) I see that your X1 Carbon 1st generation has no reported successful boot.

coreboot maintains a [list of user-supplied boot and build logs](https://review.coreboot.org/cgit/board-status.git/tree), to better understand the status of a board with the current codebase. The most recent boot is displayed in the colored box next to the board name in the [Supported Motherboards page](https://www.coreboot.org/Supported_Motherboards).

Can you build the latest version of coreboot and upload the logs to the repo, to keep your laptop supported by the coreboot team? No sensitive information is logged, as you can see by [one of my latest reports](https://review.coreboot.org/cgit/board-status.git/tree/sapphire/pureplatinumh61/4.6-1237-g3f3025d7f1/2017-08-26T16_30_37Z).

corna, 2017-09-11:
To do it, you just have to:
 * Register an account on https://review.coreboot.org (or login with your GitHub account)
 * In `Settings -> HTTP Password` generate a password (or upload the SSH public key of the `root` user)
 * Build and flash the latest version of coreboot
 * During the first boot after the flash (fresh boot, no resume from S3) go in the coreboot source folder and run `make -C util/cbmem && sudo make -C util/cbmem install && sudo util/board_status/board_status.sh`
 * Check that the output in `/tmp/coreboot_board_status.[...]` is OK
 * Upload it with `sudo util/board_status/board_status.sh -u`

WarOfDevil, 2019-01-18:
Did anyone succeeded  with this?
I just got a Thinkpad X1 gen1, and as soon as I receive the SOIC clip I will start to work on it. The end goal for me will be to use only one 8Mb flash with Coreboot and a shrink/clean ME. The info from corna posted above are quite helpful to be honest, I hope to make it work. And I will keep you updated! :)

elpinguinoatgmxch, 2019-01-20:
Hi WarOfDevil, just as a quick update: I've retried several times (also with newer coreboot commits). Unfortunately coreboot never worked for me. So, I'm using the stock BIOS with me_cleaned ME (which works fine). Please let me know if you get coreboot to work on your X1.

WarOfDevil, 2019-01-20:
Hi elpinguinoatgmxch, thank you so much for your update, that's really helpful to know. Because at this point if it doesn't work for me as well I will get in touch with Coreboot and report the issue. (Also because in two weeks I will be joining FOSDEM, there will be the Coreboot stand so I can actually talk with maintainers).

I have a few questions for you:
- What is your experience with your stock bios + clean me? Do you have any issues or everything is working fine comparing with using the laptop with just stock bios?
- Would you mind explaining exactly what you did to clean the ME from the stock bios? (Because of the 2 flash I don't exactly understand, since the me_cleaner utility should be run on an image with descriptor file).
- Could you please send me your rom files? (stock and cleaned) In such a way I can start investigating stuff. 

Thanks a lot, and I will keep this post updated as soon as I can. :)

corna, 2019-02-17:
Hi guys.
I've just bough this laptop (great PC) and coreboot works well with a "cleaned" me.
I've tried it just for a couple of hours but everything seems to work well, so far no differences with the stock firmware.

Some infos:
 * I'm currently using the whole "12 MB memory" but, like @WarOfDevil, I plan to use just the 8MB chip (to ease the reflash in case of issues)
 * GPU works both with the original VBIOS (extracted from the Lenovo's firmware) and with _libgfxinit_ (with the usual issues, like wrong resolution for GRUB and no video in linux before KMS, which requires the use of an early KVM in initramfs)
 * I've tested only Linux, no plans to install Windows
 * Currently it doesn't have support for Always-ON USB, but I'll add it soon to coreboot
 * I've used the `-S` flag of _me_cleaner_, I haven't tried without it

For more info see the report [here](https://review.coreboot.org/cgit/board-status.git/commit/lenovo/x1_carbon_gen1/4.9-727-g358cbb3a30/2019-02-15T23_22_41Z).

Since this issue is fixed I'm closing this, feel free to keep on with the discussion, I'll be glad to provide info regarding coreboot + me_cleaner on this platform.

closed 2019-02-17 by corna