corna/me_cleaner #44: intelmetool saying that it didn't work

stone212, 2017-06-29:
This me clean was done with coreboot (make menuconfig) so maybe is a coreboot issue.  Anyway, I installed coreboot with me_cleaner, and compiling seemed to indicate that me was removed.

Then I tested  as explained on the wiki: https://github.com/corna/me_cleaner/wiki/External-flashing

$ cd coreboot/util/intelmetool
$ make
# ./intelmetool -s

Report: 

```
Bad news.  You have a 'QM67 Express Chipset Family Controller' so you have ME hardware on board and you can't control or disable it

Error mapping physical memory 0x..... [0x4000] ERRNO=1 Operation not permitted
Could not map MEI PCI device memory
```

So does that mean the me_cleaner didn't work?  Or is this a false negative and should I be posting a success story?

corna, 2017-06-29:
It doesn't mean anything, the tool failed to retrieve any information. You just have to add `iomem=relaxed` to the kernel parameters, like [here](https://github.com/corna/me_cleaner/issues/30#issuecomment-301193328).

Which kernel version are you using?

stone212, 2017-06-30:
Ohhhh....  Yes, that worked.  Thank you.  My output from intelmetool now matches what you say it should be on your page here:
https://github.com/corna/me_cleaner/wiki/External-flashing

closed 2017-06-30 by stone212