corna/me_cleaner #45: Experiences with Macbook Pros

p1g30n, 2017-07-10:
Is there any data or reports on Macbook Pros? 
I'm personally interested in experiences with i7-3720QM's but also generally if anyone has tried this on Macbooks in general.

corna, 2017-07-12:
According to #3 and the issues, no one has tried me_cleaner on a Macbook so far. You can be the first one though ;)

hinxx, 2017-08-30:
FYI, I tried your tool on EFI update image that I extracted manually on Linux, from https://support.apple.com/kb/DL1848 dmg and it says:

```
$ python me_cleaner.py -c Mac2015002EFIUpdate.pkg/Tools/EFIPayloads/MBP112_0138_B16_LOCKED.scap
Unknown image
```

Also:

```
$ ./ifdtool -d Mac2015002EFIUpdate.pkg/Tools/EFIPayloads/MBP112_0138_B16_LOCKED.scap 
File Mac2015002EFIUpdate.pkg/Tools/EFIPayloads/MBP112_0138_B16_LOCKED.scap is 8520304 bytes
No Flash Descriptor found in this image
```

persmule, 2017-08-30:
@hinxx 
It should not be an firmware image, but a efi capsule, which contains no Flash Descriptor.

archfan, 2017-08-30:
Is there a way to extract scap files?

skochinsky, 2017-08-30:
@archfan you should read the full SPI image from flash; the capsule contains the ME firmware but not in a form usable for flashing; besides the capsule is signed and replacing ME image inside it won't work; you likely need an external flasher (unless you installed linux in which case flashrom *might* work).

archfan, 2017-08-30:
Yes, that makes perfect sense. I just wanted to take a look at the files in the capsule.

skochinsky, 2017-08-31:
You can use UEFITool to look at the stuff inside. The ME region image seems to be in the file with GUID `FC1BCDB0-7D31-49AA-936A-A4600D9DD083` (search for `$FPT`)

hinxx, 2017-08-31:
Thanks @skochinsky! Got it:

```
$ python me_cleaner.py -c me.bin
ME/TXE image detected
Found FPT header at 0x10
Found 19 partition(s)
Found FTPR header: FTPR partition spans from 0x47000 to 0xcf000
ME/TXE firmware version 9.0.5.1367
Checking the FTPR RSA signature... VALID
```

hinxx, 2017-08-31:
Also got it out using **flashrom-0.9.9**:

```
 $ sudo ./flashrom -p internal -V -c "MX25L6406E/MX25L6408E" -r ../original_dump.bin
```

**ifdtool** reports:

```
...

Found Master Section
FLMSTR1:   0x0aff0000 (Host CPU/BIOS)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  enabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x0000

FLMSTR2:   0x0c0d0000 (Intel ME)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x0000

FLMSTR3:   0xffff0118 (GbE)
  Platform Data Region Write Access: enabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: enabled
  Flash Descriptor Write Access:     enabled
  Platform Data Region Read Access:  enabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       enabled
  Host CPU/BIOS Region Read Access:  enabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x0118
...
```

corna, 2017-08-31:
The Intel ME region is read-only, you need an external programmer. Unfortunately the MX25L6406E/MX25L6408E are also available in non-SOIC8 packages, let's hope Apple didn't use some weird small footprint package.

hinxx, 2017-09-01:
Looking at https://www.terapeak.com/worth/820-3787-a-apple-macbook-pro-retina-15in-late-2013-a1398-16gb-i7-logic-board/291664089801/ (2nd pic) my late 2013 A1398 model should be packing Micron 25Q064A SO8W powered by 1.8V.

corna, 2017-09-01:
Big enough and it is compatible with flashrom (N25Q064..3E), the only issue is the 1.8 V, which is not very common. You'll need either a 1.8 V programmer or a level shifter.

hinxx, 2017-09-01:
Good to know it would be possible.. I will need to think this through if it is worth the risk, though. Thanks for the help all!

corna, 2017-09-01:
IMO, it is worth the risk, but I'm the author of _me_cleaner_ so my opinion is a bit biased ;)

Consider that, once you have the external programmer and a valid dump (that you can do the first time you connect the programmer), you're safe (unless you break the hardware obviously).

caingraywood, 2017-09-22:
Hi guys. i am logicboard(Apple's motherboard) technician, i have experience with apple EFI BIOS flashing using external programmer(RT809F). For now, i still havent tried to remove the ME from the BIOS, i just Cleared the ME with RGN ME. I'll let you guys know if i get any results.

corna, 2017-09-25:
@caingraywood Keep us updated, I'm really interested in this.

ThomasBrierley, 2017-10-11:
Just chiming in... I'm planning on attempting this on my rather old MacbookPro5,1 which has an SST25VF032B flash chip, here on the right:
https://d3nevzfk7ii3be.cloudfront.net/igi/plv2D1eTcMZFmIQY.huge
http://ww1.microchip.com/downloads/en/DeviceDoc/20005071B.pdf

p1g30n, 2017-10-12:
Has anyone had success with just setting the HAP bit on a macbook?

ThomasBrierley, 2017-10-12:
@p1g30n You could probably do that, and you _might_ remove potential attack vectors (or you might not)... point being that setting the HAP bit alone means that you must still _trust_ the ME to do what it says, the base problem with ME is that it relies solely on trust to be secure.

p1g30n, 2017-10-12:
@ThomasBrierley Good point. I assume there's no way to determine wether setting the bit actually consistently disables ME (and is unable to be reset)?

ThomasBrierley, 2017-10-12:
I guess, you could periodically probe it with ifdtool? other people here will be able to answer this better than me. but (again from the more extreme point of view) you are still trusting ME that way (remember that as long as it's potentially running you can't even trust your OS), that's why using an external SPI programmer is the sure way to disable it because it's independent of ME.

corna, 2017-10-22:
It seems to work guys! #76