corna/me_cleaner #46: T530: currently does not work <--> CPU fan won't start

3hhh, 2017-07-20:
Just tested two coreboot + SeaBios images on a T530 that were identical 
except one time with ME and one time without (using the compile option, 
rev 54db255529ce8afc689ae425c24b7fb1d45654e8).

Unfortunately it seems that ME does some CPU fan initialization (ACPI?) 
that coreboot doesn't, i.e. on the image without ME the CPU fan didn't 
start anymore = not usable. I tested with Fedora 26 and ubuntu 14.02 LTS 
up to temperatures of 98 degrees celsius, but it wouldn't start. I 
didn't test manual fan control.

What is interesting:

After re-trying the cleaned ME build right after the one with ME, the 
CPU fan did start at first. Then however I did a hard reset (remove 
power except for CMOS battery, press and hold the power button multiple 
times for some seconds, attach power and boot) and the CPU fan wouldn't 
work anymore (it actually spins a couple of times before SeaBios pops 
up, but after that it's dead).

So maybe ME writes something to NVram or whatever that makes it work and 
the NVram is cleaned upon hard reset? I don't know that topic well enough...

https://mail.coreboot.org/pipermail/coreboot/2017-July/084726.html

3hhh, 2017-07-20:
Not sure whether this is a coreboot issue or ME cleaner related.

3hhh, 2017-07-22:
Just found out that ME-cleaner works with the OEM BIOS (CPU fan will start).

-->

It's a coreboot issue.

corna, 2017-08-03:
I see from the coreboot ML that you've already fixed the problem. It's definitely a coreboot issue, glad you've solved it.

closed 2017-08-03 by corna