corna/me_cleaner #48: tips for choosing a PC/laptop

dpocock, 2017-08-02:
Thanks for providing the current status and boot guard information in the wiki.

Can you provide any more tips to help people choosing or about to buy a PC/laptop, what strategy should they follow for choosing what to buy if they hope to neutralize ME and how effective such efforts might be on the CPUs currently on the market?

dryoma, 2017-08-04:
I'll second that, and a couple of more direct (yet stupid) questions for those tips:

1. Do usual, non B and Q Intel desktop systems need the cleaning? I assume, the hardware for ME is always there, but the firmware is no, is that right?
1. Is there a chance that any kind of system update (bios, etc.) will restore the ME firmware? If yes, what are the best ways to avoid that?

corna, 2017-08-04:
@dpocock 
I'll think about it, in case I'll add a wiki page. However, if someone is going to buy a PC and he's concerned about Intel ME, he should buy something that doesn't have Intel ME at all (like the ASUS Chromebook C201, which is also supported by [Libreboot](https://libreboot.org/docs/hardware/c201.html)).

@dryoma 
> 1. Do usual, non B and Q Intel desktop systems need the cleaning? I assume, the hardware for ME is always there, but the firmware is not, is that right?

**Every** Intel platform from Nehalem (2008) onwards requires an ME firmware, no matter the line. Even the Intel SoC and the server boards requires an Intel ME-like firmware (TXE and SPS). The Intel Quark line (the SoC used in the Arduino Galileo/Edison) doesn't have Intel ME, but that's definetly not a SoC for PCs.

> 2. Is there a chance that any kind of system update (bios, etc.) will restore the ME firmware? If yes, what are the best ways to avoid that?

Yes, the BIOS updates can also update the Intel ME firmware. If you want you can flash only the bios manually with `ifdtool --layout layout.txt && flashrom --layout layout.txt --image bios [...]`, but the simplest option is to follow the official update process, check if it has restored the Intel ME firmware and, in that case, re-apply _me_cleaner_.

> And an awesome project, by the way. Less than a year ago it was all dark and gloomy, and now there is hope 😄

Thanks :)

K1773R, 2017-08-04:
@corna: regarding servers, coreboot tells me theres no Intel ME on Westmere based servers. Did you try cleaning any servers?

skochinsky, 2017-08-04:
@K1773R by "coreboot's ME check", do you mean "intelmetool"? AFAIK it supports only a fixed set of platforms so you may need to amend its PCI ID check for it to even try talking to ME.
Servers run SPS (Server Platform Services) firmware which is quite different from the desktops. AFAIK it provides some of the functionality necessary for server's functioning (sensors etc.) so I would not advise modifying it (it still can be done though, as the overall firmware structure is very similar).

corna, 2017-08-04:
@skochinsky Are you sure about SPS? Aren't the sensors controlled by the SuperIO controller nowdays? [This](https://github.com/corna/me_cleaner/issues/3#issuecomment-272395918) user used me_cleaner on his server board and didn't report any particular bug (but maybe it's just a lucky case).

skochinsky, 2017-08-06:
@corna I don't think server boards use SuperIO (I could be wrong though). Check e.g. this document mentioning "ME commands" related to sensors: 

https://www.intel.com/content/dam/www/public/us/en/documents/technical-specifications/intel-power-node-manager-v3-spec.pdf

The linked issue seems to be a no-op due to the missing ME11 support at the time.

NM (Node Manager) is one of the modules present in the SPS 2-3 firmwares. I did not look closely at SPS4(~ME11 equivalent) yet.

dryoma, 2017-08-07:
> Every Intel platform from Nehalem (2008) onwards requires an ME firmware ... 
> Yes, the BIOS updates can also update the Intel ME firmware ...

Thanks @corna! Good to know that.

closed 2017-08-30 by corna

dryoma, 2017-09-03:
> if someone is going to buy a PC and he's concerned about Intel ME, he should buy something that doesn't have Intel ME at all (like the ASUS Chromebook C201,

The thing here is there are cases when high performance machine is needed, so Chromebook, old or purify-ed Thinkpads are not an option. So as, for example, Sandy/Ivy bridge based laptops, that can simple be not accessible on the market, even second hand. Also I think it would be nice to touch AMD options (i.e., FX are safe, Carrizo and Ryzen have ME counterpart which can't be neutralized) could be nice on Tips page if it will be created, as some people might not know that. Just thinking.