corna/me_cleaner #49: me_cleaner or libreboot?

dpocock, 2017-08-02:
another thing that would be nice to have in the wiki is a brief comparison of a system cleaned by me_cleaner vs a libreboot system such as X200: in which cases would somebody be satisfied with me_cleaner and in which cases would somebody need to go all the way with libreboot?

vitacell, 2017-08-22:
Hi, I have 3 librebooted Macbook2,1, x200t, x60, t60, two t400, one Gigabyte ES2L, x220t, imac5,2. They will work same way. So, why Libreboot won't add more already corebooted computer to their list? because computers like x220 still will runs non-free code, yeah maybe ME got neutralized, but it still here inside your computer. This will not happen with computers with core-duo or core2duo, because their ME can be removed completely.

corna, 2017-08-22:
It depends on what you want. As @vitacell pointed out, there is still some non-free code left even after the use of me_cleaner. As explained in the book [Ruan, Xiaoyu - Platform Embedded Security Technology Revealed - Apress](https://www.apress.com/us/book/9781430265719), when a fundamental ME module has an invalid signature (= it has been removed by me_cleaner), the execution of Intel ME is halted. We can all agree that the kernel module is a fundamental one and that the execution of Intel ME is halted (and _intelmetool_ confirms it). Moreover the code left is just the one needed for the initialization, so now the topic is the freedom, not the security. You have no control on the code that's running in your ME: you can't redistribute it, study or modify it.

The question then is: how much do you care for free software? If you refuse to run non-free software on your machine your only option is libreboot*, otherwise any coreboot PC with me_cleaner should be sufficient.

\* Note that the libreboot project allows a non-free EC firmware, for example the X200 has a non-free EC firmware, while the Chromebook ASUS C201 has a free one. So, if you really want only free software on your laptop, not every Libreboot laptop is OK for you.

closed 2017-08-30 by corna