corna/me_cleaner #50: Unable to shrink ME

lenzj, 2017-08-06:
Following [internal flashing instructions with coreboot](https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-coreboot), and the process without shrinking works fine, however I am unable to execute the instructions around shrinking the ME.  I'm using me_cleaner [f4e3d14](https://github.com/corna/me_cleaner/commit/f4e3d1401efa79a3d40190c8a86293b601a0e87e), and ifdtool is from coreboot [5cb2263](https://review.coreboot.org/cgit/coreboot.git/commit/?id=5cb2263004c2068090ecdf955d3da01fd46daee7).  The steps and output I followed are below:
```
$ me_cleaner.py -O stock-BIOS-compressed.rom -r ../original/stock-BIOS.rom
Full image detected
The ME/TXE region goes from 0x3000 to 0x4ff000
Found FPT header at 0x3010
Found 19 partition(s)
Found FTPR header: FTPR partition spans from 0xcf000 to 0x145000
ME/TXE firmware version 7.1.40.1161
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xed)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x1131bd - 0x11324f): removed
 BUP              (Huffman, fragmented data    ): NOT removed, essential
 KERNEL           (Huffman, fragmented data    ): removed
 POLICY           (Huffman, fragmented data    ): removed
 HOSTCOMM         (LZMA   , 0x11324f - 0x118808): removed
 RSA              (LZMA   , 0x118808 - 0x11d2b9): removed
 CLS              (LZMA   , 0x11d2b9 - 0x121ccb): removed
 TDT              (LZMA   , 0x121ccb - 0x127e79): removed
 FTCS             (Huffman, fragmented data    ): removed
Relocating FTPR from 0xcf000 - 0x145000 to 0x3400 - 0x79400...
 Adjusting FPT entry...
 Adjusting LUT start offset...
 Adjusting Huffman start offset...
 Adjusting chunks offsets...
 Moving data...
The ME minimum size should be 86016 bytes (0x15000 bytes)
The ME region can be reduced up to:
 00003000:00017fff me
Checking FTPR RSA signature... VALID
Done! Good luck!
```
Execute ifdtool to extract layout from original stock BIOS:
```
$ ifdtool -f layout.txt ../original/stock-BIOS.rom
File ../original/stock-BIOS.rom is 8388608 bytes
Wrote layout to layout.txt
```
Modify layout.txt to reflect shrinking ME from:
```
00000000:00000fff fd
00500000:007fffff bios
00003000:004fffff me
00001000:00002fff gbe
```
to:
```
00000000:00000fff fd
00018000:007fffff bios
00003000:00017fff me
00001000:00002fff gbe
```
Executed ifdtool with the new layout.txt:
```
$ ifdtool -n layout.txt stock-BIOS-compressed.rom
File stock-BIOS-compressed.rom is 8388608 bytes
DANGER: Region Intel ME is shrinking.
    The region will be truncated to fit.
    This may result in an unusable image.
Copy Descriptor 0 (Flash Descriptor) (4096 bytes)
   from 00000000+00000000:00000fff (      4096)
     to 00000000+00000000:00000fff (      4096)
Copy Descriptor 1 (BIOS) (3145728 bytes)
   from 00500000+00000000:007fffff (   3145728)
     to 00018000+004e8000:007fffff (   8290304)
Copy Descriptor 2 (Intel ME) (86016 bytes)
   from 00003000+004e8000:004fffff (   5230592)
     to 00003000+00000000:00017fff (     86016)
Copy Descriptor 3 (GbE) (8192 bytes)
   from 00001000+00000000:00002fff (      8192)
     to 00001000+00000000:00002fff (      8192)
Writing new image to stock-BIOS-compressed.rom.new
```
Checked the integrity of the resulting BIOS file with me_cleaner and found that it was corrupted:
```
$ me_cleaner.py -c stock-BIOS-compressed.rom.new
Full image detected
The ME/TXE region is corrupted or missing
```
Running the same check on the original file looks fine:
```
$ me_cleaner.py -c stock-BIOS-compressed.rom
Full image detected
The ME/TXE region goes from 0x3000 to 0x4ff000
Found FPT header at 0x3010
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0x3400 to 0x79400
ME/TXE firmware version 7.1.40.1161
Checking FTPR RSA signature... VALID
```
I don't know if it helps, but below is a diff of the ifdtool dump from the two images:
```diff
1c1
< File stock-BIOS-compressed.rom is 8388608 bytes
---
> File stock-BIOS-compressed.rom.new is 8388608 bytes
150,153c150,153
< FLREG1:    0x07ff0500
<   Flash Region 1 (BIOS): 00500000 - 007fffff
< FLREG2:    0x04ff0003
<   Flash Region 2 (Intel ME): 00003000 - 004fffff
---
> FLREG1:    0x07ff0018
>   Flash Region 1 (BIOS): 00018000 - 007fffff
> FLREG2:    0x00170003
>   Flash Region 2 (Intel ME): 00003000 - 00017fff
156c156
< FLREG4:    0x00001fff
---
> FLREG4:    0x00000fff
```
Any ideas or suggestions?  Thanks!

corna, 2017-08-07:
`ifdtool -n` is buggy and sometimes it corrupts the ME region. I've just pushed a new commit on the _dev_ branch which add two options, `-D` and `-M`, to extract the modified descriptor and truncated ME image. For example this command

     ./me_cleaner.py -O cleaned_image.bin -d -t -r -M truncated_me_region.bin -D modified_descriptor.bin original_image.bin`

generates a `modified_descriptor.bin` (with shrinked me region) and `truncated_me_image.bin`, which can be used to build a new coreboot image.

It seems to work, but I haven't tried it on hardware.

lenzj, 2017-08-07:
Thank you for the quick response.  I've used your dev commit [6e097a0](https://github.com/corna/me_cleaner/commit/6e097a02d4ab11ecba2864474e191ee6575b5db7) as instructed and it works great!  I'm running coreboot on a Lenovo x220 with a __significantly__ larger cbfs and have had no issues.  Thank you!

corna, 2017-08-07:
No problem.

closed 2017-08-07 by corna