corna/me_cleaner #51: Witness of Government Exploits (WHY YOU NEED THE INTEL MANAGEMENT ENGINE)

H2KChicken, 2017-08-21:
Corna, please don't close this.

My name is Carlos Royal and I've witnessed several zero day exploits used against my computer. 

Mandatory backstory:
 
A while ago, I decided to challenge myself by attempting to obtain the Offensive Security Certified Professional certification in an effort to break into the penetration testing field. Over the course of 120 days, I managed to successfully breach and escalate on 16 systems within the OSCP lab. 

**Firefox Zero Day:**

During my progress, I noticed unusual activity on my computer. I make heavy use of the Linux terminal on an everyday basis and I noticed that the shell that I was using wasn’t the first shell that was open.
 
Upon further investigation, I noticed two bash processes running on my PC. Upon closing the one that I wasn’t using, my Firefox browser closed at the exact same instant. This leads me to believe that I was targeted by the FoxAcid system due to my activity from the OSCP labs and that the zero day exploit didn't use the proper escape sequence.
 
**Tor Zero Day:**
 
I utilized an Open-WRT router as the base of my build. From behind it, I built an Arch Linux “transparent TOR router” that was designed to fail-close (where if my PC could not connect to the internet through TOR, it wouldn’t be able to connect to the internet whatsoever). From behind this router, I rebuilt my new PC using the Arch Linux distro. A few weeks later, after my build was complete, in use, and thoroughly tested, I observed on the “check.torproject.org” page (which was a page that I would check compulsively) that I “wasn’t using TOR.”
 
This would lead me to believe that the government is in possession of a risky zero day exploit that exists to target TOR only users. Instead of targeting the TOR network directly, it would seem that this exploit works at the modem level and intercepts and possibly redirects the user to a malicious TOR node that’s not on the TOR network.

**DBUS Daemon Exploit:**

The "bash" and "sh" Linux binary aren't the only binaries that the government can target. They are also capable of targeting other system binaries, such as dbus-daemon, to create a secondary session for the purpose of viewing, and perhaps interacting, with the target's PC. This exploit is loud and leaves two lock screens that a user has to unlock (since there are two sessions in use instead of one). This exploit, when killed, also closed my Firefox browser, which indicates that the browser is one of the primary avenues of attack for the government when targeting a users PC.

**How Corna's Intel ME removal script saved my skin:**

Because of the nature of the incidents that I've witnessed, I've designed a program that utilizes the killall command that will kill all specified processes specified that are older than 5 seconds. 
 
killall "sh" -q -v -y 5s
 
This command, when run in a loop every two seconds, kills all shells ("bash" and "sh" specifically) that are younger than 5 seconds. So long as a terminal process that **THE USER CONTROLS** is already running, the user gains the ability to **use their own terminals** while **denying access to terminals that are opened as a result of any exploits that are used against their computer.** The terminal is **THE HEART** of pentesting, and in denying this resource to an attacker, they deny an attacker the ability to gain control over their PC. The idea is to open a few terminal processes before running this command in a loop, and then connecting to the internet as normal.

This can be tested **ON YOUR OWN WITH A KNOWN VULNERABLE VM.**
I actually ENCOURAGE anyone and EVERYONE to try this for themselves (I want to be proven wrong).

What's important to note here is that **THIS IS USELESS WITHOUT THE INTEL MANAGEMENT ENGINE REMOVED FROM YOUR COMPUTER.** The Intel management engine is an Operating system that shares physical resources with the target machine. Without Corna's removal efforts, I would be up the creek with no paddle. To obtain a better stance on PC security, this must be used IN COMBINATION WITH CORNA'S REMOVAL SCRIPT.

I leave my post here for the security experts to judge for themselves.

Thank you so much, Corna, for all of your efforts. They were not in vain and you are making a huge difference in this community. Thank you to everyone else that participated and helped with the study of this as well.

archfan, 2017-08-21:
Can you share the exploit?

corna, 2017-08-26:
Thanks for sharing your experience and for your support, this project required a great amont of time (and bricked hardware) and it's always a pleasure to receive such feedback.

Unfortunately this is not a project issue, so I'm closing it; I'm not locking it obviously, feel free to share this page's link and continue the discussion here if you want.

closed 2017-08-26 by corna

H2KChicken, 2017-09-07:
As I've been given permission to continue this thread, I will post my story in this post and all posts below it, and all witnessed exploits in the first post.

Who I've communicated the leaks to and how it turned out for me (this is me now throwing third parties that I've shared this information with under the bus):

**Offensive Security, The story of me being banned from the labs for sending their "team" exploits and unbanned shortly afterwards (currently in possession of written communication) when I began to consider releasing everything like I've done now:**

Understandably, the government being caught red handed wasn't something that they'd take lightly. I've been subject to several forms of covert abuse as a result of the things that I've witnessed.

Thinking that it was a good idea to do so, I attempted to send Offensive Security (an organization that supposedly "champions" security and exploitation) details of the things that I had witnessed, but received no response. Months later after being prepared to return to OSCP, I found that I was banned and I wasn't given a reason why. Shortly after considering the release of all of this information (after inquiring about why they banned me), I received an email (still in my possession) from "Offensive Security Management" that they unbanned me and didn't want me to send them more details about zero day exploits.

What I find curious about this is: Why would a security company such as OSCP (again, I'm in possession of the communication that occurred between us and other third parties) ban me for sending them exploits instead of simply responding that they weren't interested? Sending exploits and details of what I've seen shouldn't be considered a bannable offense (and, again, I'm in possession of the communication that has occurred between us).

I theorize that my emails were either intercepted or that there are shady dealings occurring with OSCP and the Government (and, again, I'm in possession of the communications that have occurred between us). One would think that a SECURITY-CENTRIC company such as theirs would take interest in the things that I've posted here instead of simply banning me on the spot with no warning to cease and desist. One would also find it slightly suspect that I was immediately unbanned when I began to consider posting everything that I've discovered in a public space (such as this one).

As I have zero cares left (I've gotten everything that I need from OSCP and care not if they ban me again or not), my student ID is OS-16134.

My next post will most likely detail the written communication that I've had with the EFF (they have yet to help me) and with my local Congressman (who I've received no response from, despite there being an option to receive a response from them in their contact form).

H2KChicken, 2017-09-20:
**More Important Portion of this Post:**

OIG Filed Complaint:

To make it a matter of public record (to prevent the OSCP incident from occurring again), I've filed a complaint with OIG (and linked them to this post). I've also saved screenshots of the complaint just to ensure that nothing suspicious happens since the complaint has been filed.

What will occur as a result of this, I'm unsure. I feel that it's to my benefit to make it a matter of public record that I've filed a complaint with the proper parties (which have been ignored).

**Less Important (but still pretty important) Portion of this Post (events that occurred AFTER THE OSCP BAN/UNBAN INCIDENT):**

EFF Contact Attempt:

I've attempted to contact the EFF for legal assistance and advice regarding this matter (currently in possession of the communications that have occurred between myself and "Amul Kalia" at info@eff.org). The last response that I received was that their "legal team was busy preparing for the Vegas Hacker Conference." I have yet to receive any further communication from the EFF, despite asking for an update.

Congress Contact attempt:

I've attempted to contact Congressman/representative, Anthony G. Brown, through his online contact form. I've listed the details of everything that I've posted here. His contact form has a tick option that specifically requests for him (or his staff, most likely) to respond to what was sent. I have yet to receive a response and it's been two months.

H2KChicken, 2017-12-26:
Reserved for kill script.

H2KChicken, 2018-03-11:
Definitely gaslighting.

https://www.archives.gov/federal-register/codification/executive-order/12356.html#1.6

Sec. 1.6Limitations on Classification.

(a) In no case shall information be classified in order to conceal violations of law, inefficiency, or administrative error; to prevent embarrassment to a person, organization, or agency; to restrain competition; or to prevent or delay the release of information that does not require protection in the interest of national security.

H2KChicken, 2018-03-18:
**My motivation:**

I'll cut right to the chase, it's public knowledge that "LOVEINT," known as love intelligence, a concept that was publically leaked by Edward Snowden, is a large problem within the NSA.

Oh how right he was. LOVEINT is the practice of "intelligence service employees" using their spying capabilities to spy on their love interests or spouses (or ex partner, in my case).

https://en.wikipedia.org/wiki/LOVEINT

This post is an act of protection from my ex girlfriend Melody Leighton/Stewart (her name was Melody Stewart the last we interacted seven years ago), who is currently employed by or associated with the National Security Agency.

After spotting "the NSA" in my computer (by noticing two shells open on my computer and closing the one that wasn't in use, which closed my firefox browser... and was on an operating system where Firefox does NOT require a shell
to run), strange things began to happen. Melody Leighton/Stewart began to stalk me in person, appearing in locations that she shouldn't have access to and should have no knowledge of me being. NOTE THAT I HAVE HAD ZERO PHYSICAL
INTERACTION WITH MELODY LEIGHTON/STEWART FOR YEARS, up to the point where I caught the NSA mid hack in my computer.

https://www.linkedin.com/in/melody-leighton/

https://i.imgur.com/CHAqP4q.jpg

The first major instance was during a Cisco router install for Lafyette Federal Credit Union, on the date of February 19, 2016 (between 11 am to 1 pm), where she walked in and positioned herself in a chair in the area's
cafeteria (where I would see her).

The second major instance was during a Cisco Wireless Controller cutover for Stinger Ghaffarian Technologies, on May 13, 2016, (between 6 am tp 8 am). This building requires keycards for access and performs contracts for the DOD.
Free roam of the building is not permitted. During the cutover, she made a purposeful note to walk past me to the end of the hallway, followed by walking back and out of the location where I was performing the cutover (again, to
make me notice her). Note that she should have no authorization to this building whatsoever, and that it's protected by video surveillance.

With both instances, Melody Leighton/Stewart being aware of my location should be impossible without outside assistance (such as hacking). My schedule is not public knowledge and, until recently, was performed internally on my
employing company's sharepoint server. This was only accessed with the laptop that my company has provided me.

Upon telling my company that Melody Leighton/Stewart was appearing in places that I performed work at, my company failed to assist with the matter (and Stinger Ghaffarian should have/have had video surveillance footage of the
incident to collaborate my story).

It is my belief that Melody Leighton/Stewart actively engaged in "Love Intelligence" gathering and began physically stalking me ONCE SHE GOT CAUGHT RED HANDED. I've attempted to get my employers to assist with this matter
(particularly the video surveillance portion, which should still exist), but my employers seem to have an interest in not assisting.

This is why I've posted my experience in the public community. I am currently being stalked by a vindictive ex girlfriend (who most likely has cluster B personality disorder and is actively targeting me because I rejected her 
and wounded her false self... and this is NOT a joke. I'd bet my soul on it.) who actively engaged in gathering "LOVEINT," and got caught red handed abusing her NSA capabilities (that she probably shouldn't even have access to in 
the first place).

There are other instances that I have suspicions of her involvement in, such as attacking the V.A. benefits that my former school was providing (which forced me to stop school early, find a job, and encouraged me to jump head 
first into OSCP. Without OSCP, I would have never gained the knowledge to catch the NSA red handed).

TLDR: I am being stalked by my ex girlfriend, who works for the NSA, exhibits cluster B traits, and was allowed to go this far due to the NSA's culture ACTIVELY ENCOURAGING/PROMOTING the gathering of "LOVEINT."

**Why did I not post this sooner?**

Gaslighting is an insidious form of abuse that is carried out to erode the targets confidence, memory, and to instill self doubt. It's the ultimate form of covert control that is designed to render the target silent. During my
experience, I had no leg to stand on. I believed that no one would believe my story and this was taken full advantage of through the gaslighting of my PC (being "in control" of my PC and constantly letting me know that she was),
gaslighting via constant BGE power outages, abuse of the management engine on my work laptop, physically breaking into my home and damaging my property, etc. The idea of posting this and having my sanity questioned was one that
haunted me for years, but I'm done living in fear and the work that I've put into this should be sufficient enough to have my story at least considered. But the short version is: after I caught the NSA in my computer (Melody
Leighton/Stewart), a gaslighting campaign began to make me question and doubt my memories.

**What do I hope to accomplish?**

This is me setting a permanent boundary by making my experience a public record, a boundary which has hopefully been amplified by the skill that I've gained and the knowledge that I've shared with the community.

**I'm done being a victim of narcissistic abuse. This is my act of exposing her and the abuse that she has put me through (because she got caught red handed).**