corna/me_cleaner #52: problem with flash descriptor detection and image recognition

dr-Chernobyl, 2017-08-24:
today I decided to play a little bit with old x58 platform that I have and remove intel ME from my ex58-ud3r board, the problem was when I pulled the firmware of the flash chip ifdtool got fatal error, it couldnt find flash descriptor, and when I tried me_cleaner it would just print unknown image

so my question is what does that mean, is it possible there is no ME firmware installed?

PS the firmware that is written in the chip is 2MB in size, I did snoop a bit with hex editor and found a ton of padding inside the image

corna, 2017-08-24:
I've downloaded the firmware from their website and it appears that there's no ME firmware inside it. Moreover the flash chip is very small (2 MB), 1.5 MB of ME + BIOS can't fit. Just to be sure, can you dump the other chip on the MB and check if it's identical to the first one?

You have an ICH10 LGA1366 board, which (I think, I don't have the official Intel PDF) [can work without ME in non-descriptor mode](https://review.coreboot.org/#/c/20915/1/util/me_cleaner/README.md@15). This means that, instead of using a partitioned flash layout with an Intel Flash Descriptor, the BIOS uses the whole flash. This mode was supported before Nehalem (ICH10 is an exception) and disables Intel ME (as no firmware is provided).

dr-Chernobyl, 2017-08-24:
well Im glad to hear that, I was suspecting that chip just wasnt big enough for both the ME and the BIOS especially when I saw a ton of padding in the flash descriptor area

regarding the backup BIOS chip it is identical to the first one but the firmware on it is very old from 2009., Im using firmware from 2012., you did give me a good idea now, I should flash backup chip with new firmware because old one doesnt seem to support my new 6-core xeon

thank you for your help, may I make a suggestion, could you add a note to the me_cleaner status that on ICH10 boards ME may be removed entirely or if you get an error with ifdtool and your script that old non-descriptor mode is in usage
maybe it would help people because I was a little bit puzzled

corna, 2017-08-24:
I'll add it, I just didn't have the chance to test that platform.

Can you post the output of `intelmetool -s`?

dr-Chernobyl, 2017-08-24:
no problem, I will post it when I get some OS on that computer, right now it is empty and I am slightly busy these days

dr-Chernobyl, 2017-08-24:
so I managed to get intelmetool up n' running quickly enough, here is the output:

```
[lks@localhost intelmetool]$ sudo ./intelmetool -s
Bad news, you have a `82801JIR (ICH10R) LPC Interface Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI not hidden on PCI, checking if visible
MEI device not found
```

Im not sure what exactly this means though, do I only have ME hardware on board but no software to control it?

platomav, 2017-08-25:
X58 does not have ME

dr-Chernobyl, 2017-08-25:
you mean in general? but it seems that hardware supports ME, it is strange they didnt put it on that platform

corna, 2017-08-25:
It's not particularly strange that they didn't put it on that platform (it was optional, maybe they just wanted to differentiate their products).

AFAIK the X58 does have Intel ME but it can be optionally disabled, like in the older architectures. In the datasheets of X58 and ICH10 it's often recalled the Management Engine and the possibility to have it disabled; for example, from the ICH10 datasheet:

> The Intel® Management Engine’s capability allows for a FW workaround which ensures [...]
> Note: As long as ME FW (PC or later release) is included in the ME region on the SPI device the FW workaround will always be enabled, even if ME is disabled in the flash descriptor.
>A workaround is not available for platforms that do not support ME FW.

I think `intelmetool` is wrong, as the ME can be easily disabled on your platform.

No need for _me_cleaner_ ;)

closed 2017-08-25 by corna

ValoWaking, 2020-06-03:
I has checked many BIOS on 1366 socket by asus and msi.

All asus BIOS have size ~2mb ("Unknown image")
Some MSI BIOS have size ~4mb and they have same trouble with me_cleaner error "Unknown image"