corna/me_cleaner #53: ptsecurity's research

persmule, 2017-08-28:
May it be useful to you.
[Disabling Intel ME 11 via undocumented mode](http://blog.ptsecurity.com/2017/08/disabling-intel-me.html#more)

archfan, 2017-08-28:
Can we update me cleaner to enable the HAP bit by default?

corna, 2017-08-28:
I've just received an email pointing me to that post, I've just finished reading it. Thanks for the link, **very** interesting, it also confirms my empirical observations about ME11.

Modifying the "Reserved" field in FIT changes the bit 16 in the PCHSTRP0, as confirmed by the corresponding XML. I'll do some tests on my PC the next days and, if everything works correctly, I'll add it to me_cleaner.

Yes, if a full image is passed I'll add it by default (as this operation is far safer than the rest of _me_cleaner_, there is no point of making it optional).

corna, 2017-08-28:
[Here it is](https://github.com/corna/me_cleaner/tree/dev), not tested yet (but it's just a flipped bit in the PCHSTRP0, it'll probably work).

If you want to test it note that the HAP bit it in the **descriptor** region, not in the **me** region, so you have to provide (and flash back) a full dump, not just the ME image.

skochinsky, 2017-08-28:
setting PCHSTRP10[7] to 1 *may* activate "Alt ME Disable" mode  in ME <11.0.

yoonsikp, 2017-08-28:
Would it be possible to add a flag to the tool so that the only change is flipping the bit? If I'm running modified firmware I would want as few changes as possible.

archfan, 2017-08-28:
Alright. Who wants to go first and test this commit? :P

corna, 2017-08-28:
@skochinsky On the "Intel 7 series - SPI programming guide" it is indeed a reserved bit, to be set at 0. How did you find this information?

@yoonsikp Sure, as soon as I check if everything works as expected I'll add also that option

@archfan I'll test it in a few days if you're not in a hurry ;)

matts3h, 2017-08-28:
I'll try it out :)

>If you want to test it note that the HAP bit it in the descriptor region, not in the me region, so you have to provide (and flash back) a full dump, not just the ME image.

So would I build coreboot as normal, but then run me_cleaner at the end on `coreboot.rom` instead of on the `flashregion_2_intel_me.bin` file dumped by ifdtool?

skochinsky, 2017-08-28:
@corna 
> @skochinsky On the "Intel 7 series - SPI programming guide" it is indeed a reserved bit, to be set at 0. How did you find this information?

It was mentioned in the XML of one version of the Flash Image Tool as "AltMeDisable". Even though it's actually hidden in the UI,  the code  supports it and the corresponding help text says: 

> When set true, ME is disabled in BringUp.

Besides, the BUP code indeed checks this bit (in some cases) and sets `OperationMode` field in FWSTS1 register to `ME_OPERATION_MODE_ALT_DISABLED` (2).

corna, 2017-08-28:
@matts3h This new discover is only for ME >= 11 (Skylake and following), I've not added yet to _me_cleaner_ the suggestion from Skochinsky for ME < 11. Moreover I think _intelmetool_ doesn't support this "AltDisabled" operation mode yet, I'm afraid we'll have to use MEInfo (the Windows tool from Intel) for now.

@skochinsky thanks!

matts3h, 2017-08-28:
Whoops, sorry. I don't know how I missed that.

grudnevkv, 2017-08-28:
Tried new commit on i5 6500 (Q170 chipset)
`./me_cleaner.py today.rom`

Gave me error 
> Traceback (most recent call last):
>   File "./me_cleaner.py", line 657, in <module>
>     fdf.seek(fpsba)
> NameError: name 'fdf' is not defined

Tried with option -d 
`./me_cleaner.py -d today.rom`

> Full image detected
> The ME/TXE region goes from 0x3000 to 0xa00000
> Found FPT header at 0x3010
> Found 1 partition(s)
> Found FTPR header: FTPR partition spans from 0x3400 to 0x132400
> Found FTPR manifest at 0x3848
> ME/TXE firmware version 11.6.27.3264
> ......
> ......
> Setting the HAP bit to disable Intel ME...
> Removing ME/TXE R/W access to the other flash regions...
> Checking the FTPR RSA signature... VALID
> Done! Good luck!

flashed back to chip and it doesn't start. HDMI output is not turned on, CPU fan is working. Need to check if it CPU fan works more, than 30 minutes.

Hikari-chin, 2017-08-29:
Could someone with an SPI programmer or other way to recover test if setting this bit causes Verified Boot (in FVE or FVME mode) to shut down/"brick" the computer, like modifying the ME normally would?

persmule, 2017-08-29:
I have managed to set HAP on ME8 by the modification below:

> @@ -652,7 +652,7 @@ if __name__ == "__main__":
>                  print("Truncating file at {:#x}...".format(end_addr))
>                  f.truncate(end_addr)
>  
> -    if me_start > 0 and me11:   # Intel ME > 11 only, ignored on TXE/SPS
> +    if me_start > 0:# and me11:   # Intel ME > 11 only, ignored on TXE/SPS
>          print("Setting the HAP bit to disable Intel ME...")
>          fdf.seek(fpsba)
>          pchstrp0 = unpack("<I", fdf.read(4))[0]

skochinsky, 2017-08-29:
HAP bit only works in ME11, it won't have any effect in ME8.

skochinsky, 2017-08-29:
@Hikari-chin 
> Could someone with an SPI flash programmer or other way to recover test if setting this bit causes Verified Boot (in FVE or FVME mode) to shut down/"brick" the computer (on a PC with it enabled - like the XPS 15 9550 or 9560), like modifying the ME normally would?Could someone with an SPI flash programmer or other way to recover test if setting this bit causes Verified Boot (in FVE or FVME mode) to shut down/"brick" the computer (on a PC with it enabled - like the XPS 15 9550 or 9560), like modifying the ME normally would?

I don't have one, but if BIOS is using ME's self-reported hash as part of the measurement as suspected, the hash won't be set if ME is disabled, likely causing measurement to fail.

persmule, 2017-08-29:
@skochinsky

skochinsky, 2017-08-29:
@persmule something like this might work:

      print("Setting AltMeDisable bit in PCHSTRP10...") 
      fdf.seek(fpsba+0x28) 
      pchstrp10 = unpack("<I", fdf.read(4))[0] 
      pchstrp10 |= (1 << 7) 
      fdf.write_to(fpsba+0x28, pack("<I", pchstrp10))

corna, 2017-08-29:
Here it is 9fa15dc9a9af2a4e1d9721e55e4641c215d9aaf5
I wrote it yesterday but I forgot to commit and push it

Not tested yet

h0t, 2017-08-29:
Hi, I saw that in you commits https://github.com/corna/me_cleaner/commit/173c7433770fe808e18cfac216aae050a29e68fb ignored on TXE/SPS, but we checked it also for  SPS  and TXE.

persmule, 2017-08-29:
@corna @skochinsky 
Already applied 3d060eb , but still seems to have no different:
[x230-log-1.txt](https://github.com/corna/me_cleaner/files/1259730/x230-log-1.txt)

skochinsky, 2017-08-29:
3d060ebb9b831b65162a1fce2d21e1254ce1bd58 is buggy, it writes back to wrong offset (0 instead of 0x28).

corna, 2017-08-29:
Fixed

platomav, 2017-08-29:
Works at ME8

[image]

corna, 2017-08-29:
@platomav Great! Do you also have the output of MEInfo with the "plain" _me_cleaner_ (_master_ branch)?

I've tested the _dev_ branch on my H110: the machine boots fine and the BIOS still takes 3-4 seconds more for the boot and then complains about a corrupted ME FW. Unfortunately both _intelmetool_ and _MEInfo_ fail to retrieve the ME status, so my tests won't be very useful.

If anyone can test it on a ME 11 platform (Skylake and later), please post here the output of [MEInfo](http://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html).

corna, 2017-08-29:
@h0t Hi! From the [blog post](http://blog.ptsecurity.com/2017/08/disabling-intel-me.html#more) I got that the HAP bit is ignored on TXE and SPS, am I wrong? Moreover _me_cleaner_ can't currently distinguish between ME, TXE or SPS firmware, so the HAP bit is set also on those images.

platomav, 2017-08-29:
> Great! Do you also have the output of MEInfo with the "plain" me_cleaner (master branch)?

I didn't check today but it should be the same as seen at my [previous report](https://github.com/corna/me_cleaner/issues/7#issuecomment-270801399).

> Unfortunately both intelmetool and MEInfo fail to retrieve the ME status, so my tests won't be very useful.

Make sure you use -verbose parameter at MEInfo. It should at least show the FWSTS. If not, try MEManuf -verbose.

h0t, 2017-08-29:
> Hi! From the blog post I got that the HAP bit is ignored on TXE and SPS, am I wrong? Moreover me_cleaner can't currently distinguish between ME, TXE or SPS firmware, so the HAP bit is set also on those images.

Hi, thank you for your work!  We saw this mode on SPS and  TXE firmware. 

> Unfortunately both intelmetool and MEInfo fail to retrieve the ME status, so my tests won't be very useful.

try "meinfo -FWSTS"

corna, 2017-08-29:
@platomav Great, thanks, I'll try it

@h0t Thank you!

It seems you were right @skochinsky, on my Ivy Bridge platform [this](https://gist.github.com/corna/7af389b96eb9081c67901b73b9055d71) is the output of _intelmetool_ after setting PCHSTRP10[7] to 1 (while [this](https://gist.github.com/corna/8b3f693a234f48bb3f456d589073ba0f) was the output with that bit set to 0).

platomav, 2017-08-29:
@corna Ok, here is a full ME8 report from today:

**ME8 Healthy** (no me_cleaner):

[image]

**ME8 FTPR-only** (me_cleaner master 2e8761e):

[image]

**ME8 FD-only** (me_cleaner dev e29a50e):

[image]

**ME8 FTPR + FD** (me_cleaner dev e29a50e):

[image]

As expected, Alt ME Disable takes precedence over anything else that me_cleaner can do to the ME at the SPI firmware level.

archfan, 2017-08-29:
Amazing progress here. I'd like to thank everyone for their contributions!

Wouldn't it make sense now to secure the firmware against malicious rewrites? Anyone looking into that?

grudnevkv, 2017-08-29:
@archfan 
You can make flash WP (WriteProtected via SPI-flasher)

h0t, 2017-08-29:
Skylake (gigabyte brix gb-bsi5ha-6300): 
[image]

@corna you can delete module's metadata (we checked it). Thanks!

grudnevkv, 2017-08-29:
HAP worked on Jetway NF592-Q170 (i5 - 6500 - Skylake)
In BIOS: ME status says "ME Failed", but I booted some Linux LiveUSB and it worked over 1 hour.

With me_cleaner (master branch) ME mode was: "ME temporary disabled"
Don't have Windows, so no MEInfo

corna, 2017-08-30:
@platomav Thanks for the report, it confirms our expectations.

@archfan Yes, as @grudnevkv said you can write protect the flash with the access settings in the flash descriptor (you can do it with _ifdtool -l_, which locks both the ME and the BIOS regions). If in the future you want to update one of those region you'll need an external programmer.

@h0t Thank you for the report. I suspected the modules metadata can be removed (and they're also easy to be removed, just an extra `f.fill_range` [here](https://github.com/corna/me_cleaner/blob/2e8761ea8be286c394cad06a576df76243d015f5/me_cleaner.py#L399)), but I don't think it has any advantage as they don't contain code and the space they use is not much. I prefer leaving them where they are.

@grudnevkv It's enough, thank you.

dryoma, 2017-08-30:
Hey there,

Just to clarify: with this new feature one will be able to turn the IME off as a whole entity, including its the code in its blackbox rom won't fire; without the need to strip the firmware, as was being done previousely? Meaning, it will be as if this shit doesn't exist, and a backdor switch will not be able to reanimate it?

> you can write protect the flash with the protection settings in the flash descriptor (you can do it with ifdtool -l, which locks both the ME and the BIOS regions). If in the future you want to update one of those region you'll need an external programmer.

Sorry for the offtopic, a noob question: so using an spi flasher a user can effectively prevent their bios from being tampered with? Does this make the bios totally immutable, and immune to all kinds of stuff like [such attacks](https://www.cylance.com/en_us/blog/gigabyte-brix-systems-vulnerabilities.html), restoring backups via dual boot? Also, is it reversible (I assume, with the flasher too)?

grudnevkv, 2017-08-30:
> Just to clarify: with this new feature one will be able to turn the IME off as a whole entity, including its the code in its blackbox rom won't fire; without the need to strip the firmware, as was [...]

Nope ME is still there with essential core modules (SYSLIB, KERNEL e.t.c), but with this mode (Alt Disable Mode) it just do something with BootGuard and then idles

> Sorry for the offtopic, a noob question: so using an spi flasher a user can effectively prevent their bios from being tampered with? Does this make the bios totally immutable, and immune to all [...]

Yes, unless someone has physical access to your motherboard and SPI flash chip on it

skochinsky, 2017-08-30:
For ME<11, the AltMeDisable bit is checked in the BUP module, so  you still need to have the BUP module present, otherwise boot ROM will enter the 30-minute reboot cycle, same as with missing firmware. And boot ROM and BUP still run the very basic initialization, so you can't get rid of them, but in theory it should let us remove *everything* but BUP (e.g. even KERNEL and maybe ROMP too).

skochinsky, 2017-08-30:
@grudnevkv 
> Don't have Windows, so no MEInfo

You can try the DOS or UEFI version.

corna, 2017-08-30:
> Nope ME is still there with essential core modules (SYSLIB, KERNEL e.t.c), but with this mode (Alt Disable Mode) it just do something with BootGuard and then idles as far as I understood article.

Correct. The problem is that recent (ME > 6, ~2008) x86 platforms **require** Intel ME, period. For ME 6-10 it is required as it is the only entity capable to turn off the 30 minutes watchdog, for ME > 11 the integration is tighter and ME also initializes part of the system (I think). Unless the crypto has some flaws and we're able to write our code and reimplement the initialization done by the basic modules, we will never be able to remove completely the Intel ME firmware.

The HAP/AltMeDisable bit is useful because it tells Intel ME to stop gracefully (instead of our previous method, which was to corrupt the non fundamental modules to force ME to hang), which **might** be better for Boot Guard, but for the current users of _me_cleaner_ nothing should change.

> Yes, it can prevent, unless someone has physical access to your motherboard and SPI flash chip on it. Cause to disable WriteProtect you need physical access to SPI chip.

Mainly yes, however there are also some other methods, see [this](https://github.com/flashrom/flashrom/blob/staging/Documentation/mysteries_intel.txt). For example, on my Sapphire Pure Platinum H61 with coreboot the security override pin is enabled, which causes a total unrestricted access to the flash. It is currently unknown to me what drives that pin.

@skochinsky KERNEL is already removed by _me_cleaner_, I'll check if we can remove also ROMP.

corna, 2017-08-30:
> You can try the DOS or UEFI version.

Is there a UEFI version?

skochinsky, 2017-08-30:
yes, there is `MeInfo.efi`.

persmule, 2017-08-30:
coreboot log on my x230 with Alt ME Disable applied:

[x230-log-1.txt](https://github.com/corna/me_cleaner/files/1263095/x230-log-1.txt)

grudnevkv, 2017-08-30:
@skochinsky
Where to download DOS MEInfo version ?

@corna
>Mainly yes, however there are also some other methods, see this. For example, on my Sapphire Pure Platinum H61 with coreboot the security override pin is enabled, which causes a total unrestricted access to the flash. It is currently unknown to me what drives that pin.

Well, yes, I read somewhere WP pin can be connected to chipset, so if you are paranoid about it you need accurately cut trace to WP pin.

@corna
In what me_cleaner version KERNEL module is removed ?
I get this output:
>rbe (Huffman , 0x007cc0 - 0x00a380): NOT removed, essential
kernel (Huffman , 0x00a380 - 0x019fc0): NOT removed, essential
syslib (Huffman , 0x019fc0 - 0x02cfc0): NOT removed, essential
bup (Huffman , 0x02cfc0 - 0x055c40): NOT removed, essential

corna, 2017-08-30:
>  Where to download DOS MEInfo version ?

MEInfo is not redistributable, don't ask where it can be downloaded

> Well, yes, I read somewhere WP pin can be connected to chipset, so if you are paranoid about it you need accurately cut trace to WP pin.

My bad, I misunderstood you. There are two types of flash write protections:
 1) Hardware WP pin (active low), which is usually connected to VDD (or, like in the Chromebooks, to GND with a [jumper to disable the write protection](https://libreboot.org/docs/hardware/c201.html#flash-chip-write-protection-the-screw)). This is set by the manufacturer and, unless you want to do hardware modifications of your board, you can't change this. Moreover this protection covers only a small part of the flash chip (usually the bootblock, to have the read-only root of trust of the firmware).
 2) Descriptor security settings, which are a set of bitmasks read by the flash controller that allow/disallow the read/write access on the various regions (BIOS, descriptor, ME, GbE, EC, ...) for ME or the system. These are our "security access bits", as once we have set the descriptor and the ME region as read only for everyone, no one can change them without an external programmer (except for the cases I linked before).

We only care about 2), the WP pin is useless for us.

>  In what me_cleaner version KERNEL module is removed ?

It is removed only for ME < 11, for ME >= 11 those four modules are required and unremovable (as confirmed by the RE of ptsecurity).

nroach44, 2017-08-30:
X220:

`Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0x3400 to 0x79400
ME/TXE firmware version 7.1.20.1119
Removing extra partitions...
Removing extra partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xed)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x0475c5 - 0x047657): removed
 BUP              (Huffman, fragmented data    ): NOT removed, essential
 KERNEL           (Huffman, fragmented data    ): removed
 POLICY           (Huffman, fragmented data    ): removed
 HOSTCOMM         (LZMA   , 0x047657 - 0x04cc0c): removed
 RSA              (LZMA   , 0x04cc0c - 0x0516bd): removed
 CLS              (LZMA   , 0x0516bd - 0x0560cf): removed
 TDT              (LZMA   , 0x0560cf - 0x05c27a): removed
 FTCS             (Huffman, fragmented data    ): removed
The ME minimum size should be 86016 bytes (0x15000 bytes)
The ME region can be reduced up to:
 00003000:00017fff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
`

`Intel ME early init
Intel ME firmware is ready
CBFS: 'Master Header Locator' located CBFS at [700100:7fffc0)
CBFS: Locating 'mrc.cache'
CBFS: Found @ offset 1fec0 size 10000
find_current_mrc_cache_local: No valid MRC cache found.
  Row    addr bits  : 15
  Column addr bits  : 10
  Number of ranks   : 2
  DIMM Capacity     : 4096 MB
  CAS latencies     : 5 6 7 8 9
  tCKmin            :   1.500 ns
  tAAmin            :  13.125 ns
  tWRmin            :  15.000 ns
  tRCDmin           :  13.125 ns
  tRRDmin           :   6.000 ns
  tRPmin            :  13.125 ns
  tRASmin           :  36.000 ns
  tRCmin            :  49.125 ns
  tRFCmin           : 160.000 ns
  tWTRmin           :   7.500 ns
  tRTPmin           :   7.500 ns
  tFAWmin           :  30.000 ns
  Row    addr bits  : 15
  Column addr bits  : 10
  Number of ranks   : 2
  DIMM Capacity     : 4096 MB
  CAS latencies     : 5 6 7 8 9
  tCKmin            :   1.500 ns
  tAAmin            :  13.125 ns
  tWRmin            :  15.000 ns
  tRCDmin           :  13.125 ns
  tRRDmin           :   6.000 ns
  tRPmin            :  13.125 ns
  tRASmin           :  36.000 ns
  tRCmin            :  49.125 ns
  tRFCmin           : 160.000 ns
  tWTRmin           :   7.500 ns
  tRTPmin           :   7.500 ns
  tFAWmin           :  30.000 ns
ME: Wrong mode : 2
ME: FWS2: 0x120a0000
ME:  Bist in progress: 0x0
ME:  ICC Status      : 0x0
ME:  Invoke MEBx     : 0x0
ME:  CPU replaced    : 0x0
ME:  MBP ready       : 0x0
ME:  MFS failure     : 0x0
ME:  Warm reset req  : 0x0
ME:  CPU repl valid  : 0x0
ME:  (Reserved)      : 0x0
ME:  FW update req   : 0x0
ME:  (Reserved)      : 0x0
ME:  Current state   : 0xa
ME:  Current PM event: 0x2
ME:  Progress code   : 0x1
Waited long enough, or CPU was not replaced, continue...
PASSED! Tell ME that DRAM is ready
ME: FWS2: 0x120a0000
ME:  Bist in progress: 0x0
ME:  ICC Status      : 0x0
ME:  Invoke MEBx     : 0x0
ME:  CPU replaced    : 0x0
ME:  MBP ready       : 0x0
ME:  MFS failure     : 0x0
ME:  Warm reset req  : 0x0
ME:  CPU repl valid  : 0x0
ME:  (Reserved)      : 0x0
ME:  FW update req   : 0x0
ME:  (Reserved)      : 0x0
ME:  Current state   : 0xa
ME:  Current PM event: 0x2
ME:  Progress code   : 0x1
ME: Requested BIOS Action: No DID Ack received`

Not up for 30 minutes yet but AFAICT it looks good.

skochinsky, 2017-08-30:
> ME: Wrong mode : 2

mode 2 is "Alt Disable". I guess coreboot needs to be updated to handle this situation.

archfan, 2017-08-30:
Just tested  ced3b46 on my workstation. No problems so far. In fact setting the HAP bit has decreased my boot time. Before I had to power cycle a couple of times before I could see anything on my screen - or sometimes it would take ~ 5 minutes. Those bugs are gone now.

[image]

grudnevkv, 2017-08-30:
@corna MeInfo DOS output
[image]

corna, 2017-08-30:
Great, it seems that everything is working as expected, I'll merge soon the _dev_ branch into _master_. Thank you all for the tests.

@skochinsky I'll fix it, however it's just a warning message, [nothing more](https://github.com/coreboot/coreboot/blob/master/src/southbridge/intel/bd82x6x/early_me.c#L133-L136).

@archfan Great! This could mean that some BIOS handle this "Alt Disable Mode" better than the previous disable.

citypw, 2017-08-31:
Tested "me_cleaner -s *.rom" on P10S-M WS( Skylake): The booting process became very slow after enabled HAP bit. I tried about 10 times to boot but only succeed at once. Maybe some private UEFI messed something with? It works mostly. Thanks!

corna, 2017-08-31:
@citypw So it worked well and fast without the HAP bit and now it is slower?

nroach44, 2017-08-31:
Interestingly I've also noticed it takes slightly longer (like 3s more) for coreboot/seabios to turn on the backlight, but I'm not certain that was because of this test or the ME relocation/truncation I did. It's not huge so if it did indeed effect it, it may just be coreboot timing out waiting for the ME to respond.

skochinsky, 2017-08-31:
I think coreboot needs to skip the rest of ME init if opmode is 2, this may improve boot speed.

raburton, 2017-08-31:
FYI, tested new `-s` option with i7-4790 on MSI B85-G41, bios v2.9. MSInfo produces a very odd output, although I assume ME has been disabled and the PC has been running for 40 minutes so far without a watchdog reset.

<details>
  <summary>Full MEInfo output</summary>
C:\temp>MEInfoWin64.exe -verbose

Intel(R) MEInfo Version: 11.0.15.1003
Copyright(C) 2005 - 2016, Intel Corporation. All rights reserved.

  Windows OS Version : 6.2.9200 ""
    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes
    Table Type   1 ( 0x 01 ) found, size of  27 (0x 1B ) bytes
    Table Type   2 ( 0x 02 ) found, size of  15 (0x 0F ) bytes
    Table Type   3 ( 0x 03 ) found, size of  25 (0x 19 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type  11 ( 0x 0B ) found, size of   5 (0x 05 ) bytes
    Table Type  12 ( 0x 0C ) found, size of   5 (0x 05 ) bytes
    Table Type  24 ( 0x 18 ) found, size of   5 (0x 05 ) bytes
    Table Type  32 ( 0x 20 ) found, size of  20 (0x 14 ) bytes
    Table Type  34 ( 0x 22 ) found, size of  11 (0x 0B ) bytes
    Table Type  26 ( 0x 1A ) found, size of  22 (0x 16 ) bytes
    Table Type  36 ( 0x 24 ) found, size of  16 (0x 10 ) bytes
    Table Type  35 ( 0x 23 ) found, size of  11 (0x 0B ) bytes
    Table Type  28 ( 0x 1C ) found, size of  22 (0x 16 ) bytes
    Table Type  36 ( 0x 24 ) found, size of  16 (0x 10 ) bytes
    Table Type  35 ( 0x 23 ) found, size of  11 (0x 0B ) bytes
    Table Type  27 ( 0x 1B ) found, size of  15 (0x 0F ) bytes
    Table Type  36 ( 0x 24 ) found, size of  16 (0x 10 ) bytes
    Table Type  35 ( 0x 23 ) found, size of  11 (0x 0B ) bytes
    Table Type  27 ( 0x 1B ) found, size of  15 (0x 0F ) bytes
    Table Type  36 ( 0x 24 ) found, size of  16 (0x 10 ) bytes
    Table Type  35 ( 0x 23 ) found, size of  11 (0x 0B ) bytes
    Table Type  29 ( 0x 1D ) found, size of  22 (0x 16 ) bytes
    Table Type  36 ( 0x 24 ) found, size of  16 (0x 10 ) bytes
    Table Type  35 ( 0x 23 ) found, size of  11 (0x 0B ) bytes
    Table Type  26 ( 0x 1A ) found, size of  22 (0x 16 ) bytes
    Table Type  28 ( 0x 1C ) found, size of  22 (0x 16 ) bytes
    Table Type  27 ( 0x 1B ) found, size of  15 (0x 0F ) bytes
    Table Type  29 ( 0x 1D ) found, size of  22 (0x 16 ) bytes
    Table Type  39 ( 0x 27 ) found, size of  22 (0x 16 ) bytes
    Table Type  41 ( 0x 29 ) found, size of  11 (0x 0B ) bytes
    Table Type  41 ( 0x 29 ) found, size of  11 (0x 0B ) bytes
[... 79 lines cut ...]
    Table Type  41 ( 0x 29 ) found, size of  11 (0x 0B ) bytes
    Table Type   4 ( 0x 04 ) found, size of  42 (0x 2A ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type  16 ( 0x 10 ) found, size of  23 (0x 17 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  19 ( 0x 13 ) found, size of  31 (0x 1F ) bytes
    Table Type 136 ( 0x 88 ) found, size of   6 (0x 06 ) bytes
    Table Type 143 ( 0x 8F ) found, size of  16 (0x 10 ) bytes
    Table Type  13 ( 0x 0D ) found, size of  22 (0x 16 ) bytes
    Table Type 127 ( 0x 7F ) found, size of   4 (0x 04 ) bytes
SM BIOS table entry point structure not found

Error 79: Fail to read FW Status Register value

C:\temp>
</details>

grudnevkv, 2017-08-31:
@raburton 
Strange output is a result of running Intel(R) MEInfo Version: 11.0.15.1003 
for Intel i7-4790 
It's 4-th generation, not Skylake or higher, thus you need MEinfo utility for lower versions of ME

raburton, 2017-08-31:
@grudnevkv Thanks for the tip. I did wonder if matched versions was important, but MEInfo v11 worked fine before the modification, so I assumed the tools were backwards compatible! My ME version is 9.0.30.1482. Using MEInfo 9.0.22.1467 I get the normal output again (which shows `Alt Disable Mode`) plus a load of errors trying to communicate with the ME and all that junk I got from the v11 tool. Anyway, I think this is the first tested ME v9 and it looks like the disable has worked.

<details>
<summary>Full output for reference</summary>
C:\temp>MEInfoWin64.exe -verbose

Intel(R) MEInfo Version: 9.0.22.1467
Copyright(C) 2005 - 2013, Intel Corporation. All rights reserved.

FW Status Register1: 0x1E020191
FW Status Register2: 0x104D0106
FW Status Register3: 0x00000300
FW Status Register4: 0x00000000
FW Status Register5: 0x00000000
FW Status Register6: 0x00000000

  CurrentState:                         Init
  ManufacturingMode:                    Enabled
  FlashPartition:                       Valid
  OperationalState:                     Bring Up
  InitComplete:                         Initializing
  BUPLoadState:                         Success
  ErrorCode:                            No Error
  ModeOfOperation:                      Alt Disable Mode
  Phase:                                BringUp
  ICC:                                  Valid OEM data, ICC programmed
  ME File System Corrupted:             No
  PhaseStatus:                          UNKNOWN

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

  Windows OS Version : 6.2.9200 ""
  OS BIOS Support    : UEFI

    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes

  Windows OS Version : 6.2.9200 ""
  OS BIOS Support    : UEFI

    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes
    Table Type   1 ( 0x 01 ) found, size of  27 (0x 1B ) bytes
    Table Type   2 ( 0x 02 ) found, size of  15 (0x 0F ) bytes
    Table Type   3 ( 0x 03 ) found, size of  25 (0x 19 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   8 ( 0x 08 ) found, size of   9 (0x 09 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
[... 34 lines cut ...]
    Table Type   4 ( 0x 04 ) found, size of  42 (0x 2A ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type  16 ( 0x 10 ) found, size of  23 (0x 17 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  40 (0x 28 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  19 ( 0x 13 ) found, size of  31 (0x 1F ) bytes
    Table Type 136 ( 0x 88 ) found, size of   6 (0x 06 ) bytes
    Table Type 143 ( 0x 8F ) found, size of  16 (0x 10 ) bytes
    Table Type  13 ( 0x 0D ) found, size of  22 (0x 16 ) bytes
    Table Type 127 ( 0x 7F ) found, size of   4 (0x 04 ) bytes

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

Error 9459: Internal error (Could not determine FW features information)

C:\temp>
</details>

citypw, 2017-09-01:
@corna yes, it's running much faster w/o HAP enabled. "-s" doesn't remove the code modules disabled the ME only: https://twitter.com/hardenedlinux/status/903158983015358464

nroach44, 2017-09-01:
Running -s on an Asus P8Z68M-Pro with the stock ASUS BIOS delays boot slightly, and seemingly broke AHCI for a few boots, but that might have been my connections being funky.

```
Bad news, you have a `Z68 Express Chipset Family LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI was hidden on PCI, now unlocked
MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

ME Status   : 0x1e020191
ME Status 2 : 0x160a0100

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Initializing
ME: Current Operation State : Bring up
ME: Current Operation Mode  : Debug
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : Check to see if straps say ME DISABLED

ME: Extend SHA-256: fcf867b12599c9a59e661ee6c20fa9c22b5b9e0a513bb21791cec49b33570ce5

ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
ME: failed to become ready
ME: GET FWCAPS message failed
Re-hiding MEI device...done
```

corna, 2017-09-01:
I'm afraid I'll have to make the HAP/AltMeDisable a non-default option.

[18 more replies]

closed 2017-09-26 by corna