corna/me_cleaner #56: Did not work on Gigabyte GA-P55A-UD3R (1.0)

lbschenkel, 2017-08-30:
Motheboard: GA-P55A-UD3R <https://www.gigabyte.com/Motherboard/GA-P55A-UD3R-rev-10#sp>
BIOS: F14B <http://download.gigabyte.eu/FileList/BIOS/motherboard_bios_ga-p55a-ud3r_f14b.exe>

Used version 2e8761ea8be286c394cad06a576df76243d015f5.
Follow guide from here: https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-OEM-firmware

`ifdtool` and `me_cleaner` passed all checks, both pre- and post-cleaning. Tried flashing cleaned image via built-in motherboard "QFlash" utility, it refused claiming BIOS image was corrupt. Tried via `flashrom` in Linux, it "worked" (as in: flash succeeded). After power cycling, computer booted but froze during POST (screen is displayed with CPU and memory information but cannot enter BIOS, keyboard is unresponsive, it never reaches the next stage when you see messages from Intel AHCI).

Motherboard has dual BIOS, triggered backup BIOS in order to recover.

This is an old machine and removing ME in it is not critical for me, I was mostly trying it out to check if it worked. I can provide additional logs, BIOS images, etc. if requested.

rnhmjoj, 2017-08-30:
QFlash must be using a checksum: It fails even if a single bit is flipped.
Can you tell me which options for flashrom you used? I can't get it to detect the BIOS.

lbschenkel, 2017-08-30:
I compiled `flashrom` from source, commit is aa91d5c16858cb400cc61e8a759838f645e3f314:
`flashrom -p internal -c MX25L1605 ...`
I had to specify `-c` because the probing detects three candidates, I picked the first one.

rnhmjoj, 2017-08-30:
Well, it looks like my system is not supported. I will have to try with an external programmer.

corna, 2017-08-30:
Quite strange, as it seems that the BIOS supports the modifications introduced by _me_cleaner_ (= the raminit succeeds), however it refuses to continue with the boot. Can you try with the _dev_ branch of _me_cleaner_?

lbschenkel, 2017-08-31:
@corna By using the dev branch you mean specifically the new `-s` option? Since that requires extracting ME from the BIOS, how do I flash it afterwards (or merge it back into the BIOS update file)?

lbschenkel, 2017-08-31:
OK, I misread "requires a full dump" as "requires separate ME/TXE image". I tried now with `-s`, and QFlash did not complain, however I am not sure if it had any effect:

Output of `intelmetool -s`:
```
Bad news, you have a `P55 Chipset LPC Interface Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI was hidden on PCI, now unlocked
MEI found: [8086:3b64] 5 Series/3400 Series Chipset HECI Controller

ME Status   : 0xee110205
ME Status 2 : 0xb001f

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : Preboot
ME: Current Operation Mode  : (null)
ME: Error Code              : No Error
ME: Progress Phase          : ROM Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : (null)

ME: Extend Register not valid

ME: timeout waiting for data: expected 8, available 6
ME: GET FW VERSION message failed
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed
Re-hiding MEI device...done
```

Unfortunately I did not save the output pre-flashing, but I did check it multiple times and from memory it looks visually identical to the above (so no change).

I did try flashing with `flashrom` as well but the results were the same.

I can flash the original ROM again, but I'm reasonably confident that the output of `intelmetool` in the pre-clean state is the same.

seropp, 2017-09-03:
Confirming output from a non-cleaned Gigabyte GA-P55A-UD3R:
```
Bad news, you have a `P55 Chipset LPC Interface Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI was hidden on PCI, now unlocked
MEI found: [8086:3b64] 5 Series/3400 Series Chipset HECI Controller

ME Status   : 0xee110205
ME Status 2 : 0xb001f

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : Preboot
ME: Current Operation Mode  : (null)
ME: Error Code              : No Error
ME: Progress Phase          : ROM Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Global reset after an error

ME: Extend Register not valid

ME: timeout waiting for data: expected 8, available 6
ME: GET FW VERSION message failed
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed
Re-hiding MEI device...done
intelmetool -s  6.19s user 0.43s system 34% cpu 19.030 total
```

seropp, 2017-09-03:
I decided to flash it after all, after reboot I got about 5 seconds of black screen, after that it reached the gigabyte logo screen where it freezes with all the indicator LEDs on the mobo lit up and CPU fan at max. Backup bios didn't automatically kick in, had to short pin 4+7 on M_BIOS chip with a paper clip for a second after powering on to force it to work (this is a delicate operation if you don't have proper tools), but at least it booted now.

lbschenkel, 2017-09-03:
I managed to trigger the backup BIOS by powering it on holding the reset button at the same time, after a few seconds it turns off and on and starts restoring the BIOS from the backup chip.