corna/me_cleaner #58: Backport IFD-related features to coreboot's ifdtool

persmule, 2017-08-31:
For ifdtool has been integrated into coreboot's build process, I believe it is better to have ifdtool handle all IFD-related operations, so I am going to backport those IFD-related features to ifdtool.

@corna I remember that you have said that ifdtool's manipulation may somtimes breaks IFD. Could you point out which data get broken, and how me_cleaner prevents this, please?

corna, 2017-08-31:
Also _me_cleaner_ is integrated in the build process, I don't know if _ifdtool_ should handle ME-specific flags.

With a modifed layout where the _me_ region has been shrinked (the end of the region has moved to a lower address, even just by 4 kB), `ifdtool -n newlayout image.rom` corrupts the _me_ region. The descriptor is fine, however it seems that the starting chunk of the _me_ region has been cut off. _me_cleaner_ doesn't have this problem as it doesn't move regions at all: `-D` generates a descriptor with a modified layout, while `-M` extracts the _me_ region (and, if the `-t` flag has been passed, it just extracts the region up to _end_addr_).

persmule, 2017-08-31:
That is because ifdtool uses "right-align" scheme when moving region (seen from commit 4eabe1e, when feature new-layout was initially implemented), though such scheme is only valid for the BIOS region.

corna, 2017-08-31:
Makes sense. Thank you.

persmule, 2017-09-01:
Could you help me review [this change](https://review.coreboot.org/#/c/21315/)? It may help to resolve the issue mentioned above.

corna, 2017-09-01:
Of course. However I'm a bit busy these weeks, I'll try to review it ASAP.