corna/me_cleaner #59: Did not work on Gigabyte GA-P55A-UD6 (1.0)

Neo1171, 2017-08-31:
Because the problem is very similar to another thread (https://github.com/corna/me_cleaner/issues/56),
i have copy&pasted some lines and added more information.

CPU architecture: Lynnfield

CPU model: Intel Xeon X3480
Link: http://ark.intel.com/products/48501/Intel-Xeon-Processor-X3480-8M-Cache-3_06-GHz

Motherboard: Gigabyte GA-P55A-UD6
Link: https://www.gigabyte.com/Motherboard/GA-P55A-UD6-rev-10#sp

OEM Bios: F12c "P55AUD6.12C"
Link: download.gigabyte.eu/FileList/BIOS/motherboard_bios_ga-p55a-ud6_f12c.exe

Ignition Runtime Image FW version: 6.0.30.1199
Ignition Factory Default Image FW Version: 6.0.30.1199

Used "me_cleaner.py" Version: It should be the version **before** 72fff7a

Here are the checksums of the used file "me_cleaner.py"
CRC-32: 1c195e2d
MD5: b431b802f9bed5b80d240335c651d6a1

Follow guide from here: https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-OEM-firmware

ifdtool and me_cleaner passed all checks, both pre- and post-cleaning. Tried flashing cleaned image via built-in motherboard "QFlash" utility, it refused claiming BIOS image was corrupt.
Tried another tool called (@Bios) for Windows. It "worked" (as in: flash succeeded). Everything worked good, when i rebooted the PC or used the reset switch several times.

But after a shutdown / power cycling, computer booted but froze during POST (screen is displayed with CPU information but cannot enter BIOS, keyboard is unresponsive, it never reaches the next stage when you see messages from Intel AHCI.

Onboard debug LED simplifies motherboard troubleshooting. The last displayed digitnumber is "50".

Motherboard has dual BIOS, triggered backup BIOS in order to recover.

Thank you very much for you effort!
IntelME is the reason why i did not buy a newer PC.

skochinsky, 2017-08-31:
There's very little sense in editing an ignition firmware, it's already pretty much the minimum necessary for the hardware init. 

I guess `me_cleaner` removes the `IGRT` partirtion containing the `IGRT_BUP` module which does some necessary initialization without which the PC cannot boot.

Neo1171, 2017-08-31:
I think so too. 

But it is still questionable.. Maybe there could be some hidden IDs which could be accessed from outside to lower your privacy. Maybe there could be more.

After i saw that it is possible to remove ME 6.0 Ignition COMPLETELY from ASRock P55 Extreme, i wanted to play around with my system too and check if everything is working.

http://www.win-raid.com/t2443f39-Remove-ME-Ignition-firmware-completely-amp-Panther-Point-PCH-observations.html

skochinsky, 2017-08-31:
removing ME firmware completely is a different thing and me_cleaner won't help you there. Maybe @platomav can guide you but I personally doubt it will work on your board.

platomav, 2017-08-31:
ME Ignition firmware at the SPI can be removed entirely and the platform will run with ME disabled by design. So you don't need me_cleaner, you can just padd the entire ME region and remove the Offset and Limit from the Flash Descriptor. Is your Flash Descriptor unlocked so that you have read/write access to the FD and ME regions? If yes, use either the in-BIOS flasher (works at the entire SPI) or Flash Programming Tool to flash. From the latest GB BIOS, replace 01002000 at 0x48-0x4C with FF0F0000 to disable the ME region at the FD and padd its contents at 0x1000 - 0x21000. Once the system boots, verify that the ME is disabled & removed by running "fptw -d STDOUT -me". If you encounter any problems after that point, you can assume that it is the BIOS which does not like the absence of ME.

Neo1171, 2017-09-01:
@platomav Thank you! I will try it the next week.

Neo1171, 2017-09-23:
Hi again. I was able to modify the BiosImage with HexEditor and flash it with fptw.exe like you have told me.
But with no success - the computer still froze during POST.

After playing around the only thing that worked, was to delete the data about IGRT, FAD1, FTPR and FAD0 in the codearea 0x1000 - 0x10CF. So after deleting, there were only data about $FPT and NFTP.

But it is still loading the Factory Default Image of IntelME.

Maybe i should try to delete data about $FPT or NFTP too, but i think it would be waste of time.
It seems that this mainboard wont boot without intelME.

Thank you for your help.
.
.
.
.
FW Status Register: 0xE2110605
FW Status Register1: 0x000B0007

  Ignition FW Status(FAD_IDX):      Using Data from Factory Default Image
  Ignition FW Status(BOOT_IMAGE):   FW booted from Factory Default Image
  Ignition FW Status(EVENT_LOG):    Event Log is empty
  Ignition FW Status(ICC_STS):      Clock programming successful
  Ignition FW Status(CF):           Processor feature configuration successful
  Ignition FW Status(SWC):          SWC application successful
  Ignition FW Progress:             Intel(R) ME is sleeping

Factory Default Partition Data is validated and healthy.
Runtime Partition Data is validated and healthy.
Factory Default Partition Code is validated and healthy.
Runtime Partition Code is corrupted.
PMXUtil: Error during PMX Call: PMxDrv!MAPPHYS - Bad Input Parameters

platomav, 2017-09-23:
As I've shown in on my P55 ASRock board, Ignition can be fully removed and per Intel POR, the platform should work. Since it doesn't work in your case, the only logical conclusion is that the BIOS has some kind of check for the ME (existence of $FPT section maybe) and it does not allow booting without it. So it a system/OEM specific quirk.