corna/me_cleaner #60: Documentation on me_cleaner vs TXT and other Intel's firmware

dryoma, 2017-09-03:
On [this wiki page](https://github.com/corna/me_cleaner/wiki/How-does-it-work%3F) as well as [in some issues](https://github.com/corna/me_cleaner/issues/53#issuecomment-325692873) and in me_cleaner reports there are statements that the tool can also work with other Intel firmware. In this case maybe a wiki page would make sense with info on that subject? I.e. what other firmware me_cleaner can lighten, how it is done (by the main process or separately), a short info on what this firmware is, what it does and what a user risks/loses if they use me_cleaner. What do you think?

dp-alvarez, 2017-09-05:
I may be wrong, but from what I understand me_cleaner doesn't really know (or care) if you run 'regular' ME, TXE, SPS or whatever.
All of those share the same boot process and core modules, the difference between then is which extra modules are loaded after boot.
me_cleaner removes all modules that are not required so it treats all "variants" of the ME firmware in the same way. Now it also sets the HAP bit, which works the same way in all ME variants (apparently).
Not quite sure which functionality would be lost by removing TXE or SPS. I know that regular ME handles things like PAVP (intel's built-in media drm), but not sure what else. You will probably lose TPM and secure boot functionality if you disable TXE, don't think there is much else to it.
Apparently SPS has more functionality. It may be responsible for fan control, sensor readings, etc if your motherboad does not have an BMC and it handles things like intel node manager and some other power-related stuff.
Few people have used me_cleaner on a server/workstation board so there is some information lacking on what to expect if you have SPS. I'm going to try to flash my supermicro x11ssm-f when I have the time and will report my results.

corna, 2017-09-06:
@dp-alvarez all correct.

SPS hasn't been widely tested, so it is currently not reported in the status page (and the integration with the system is tighter than ME/TXE, so the chances to remove it without consequences are smaller). TXE has been tested and its support status is reported in the [me_cleaner status page](https://github.com/corna/me_cleaner/wiki/me_cleaner-status).

Anyways, the firmware structure and the fundamental modules are the same for every kind of firmware (except for some versions of SPS, which have a different set of required modules, but SPS is not fully supported yet) so, as @dp-alvarez pointed out, _me_cleaner_ doesn't care which firmware it is.

The usual recommendations and risks applies as well on the other firmware.

TL;DR _me_cleaner_ supports every ME-like firmware, with some exceptions.

closed 2017-09-06 by corna

dryoma, 2017-09-07:
Thanks guys, now I get it.