corna/me_cleaner #61: Does not seem to work on Asrock h87 Performance

Raduq91, 2017-09-07:
Bios link ftp://europe.asrock.com/BIOS/1150/H87%20Performance(2.40)ROM.zip

CPU: 4690k

After using me_cleaner:

ubuntu@ubuntu:~/coreboot/util/intelmetool$ sudo ./intelmetool -s
MEI not hidden on PCI, checking if visible
MEI found: [8086:8c3a] 8 Series/C220 Series Chipset Family MEI Controller #1

ME Status   : 0x1e000245
ME Status 2 : 0x69002306

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : Host Communication
ME: Power Management Event  : Non-power cycle reset
ME: Progress Phase State    : Host communication established

ME: Extend SHA-256: 37b69b9ed1f6691f59e41637d438b0a259ac90285e43c9946fec06551b0525dc

ME: timeout waiting for data: expected 8, available 0
ME: GET FW VERSION message failed
ME Capability: Full Network manageability                 : OFF
ME Capability: Regular Network manageability              : OFF
ME Capability: Manageability                              : ON
ME Capability: Small business technology                  : ON
ME Capability: Level III manageability                    : OFF
ME Capability: IntelR Anti-Theft (AT)                     : OFF
ME Capability: IntelR Capability Licensing Service (CLS)  : ON
ME Capability: IntelR Power Sharing Technology (MPC)      : OFF
ME Capability: ICC Over Clocking                          : ON
ME Capability: Protected Audio Video Path (PAVP)          : ON
ME Capability: IPV6                                       : OFF
ME Capability: KVM Remote Control (KVM)                   : OFF
ME Capability: Outbreak Containment Heuristic (OCH)       : OFF
ME Capability: Virtual LAN (VLAN)                         : ON
ME Capability: TLS                                        : ON
ME Capability: Wireless LAN (WLAN)                        : OFF
ubuntu@ubuntu:~/coreboot/util/intelmetool$ 

and sometimes:

ubuntu@ubuntu:~/coreboot/util/intelmetool$ sudo ./intelmetool -s
MEI not hidden on PCI, checking if visible
MEI found: [8086:8c3a] 8 Series/C220 Series Chipset Family MEI Controller #1

ME Status   : 0x1e000245
ME Status 2 : 0x69002306

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : Host Communication
ME: Power Management Event  : Non-power cycle reset
[... 141 lines cut ...]
Protected Range Register Limit #4 0x0
BIOS boot State:                        Post Boot
OEM Id:                                 00000000-0000-0000-0000-000000000000
Capability Licensing Service:           Enabled

Get ME FWU OEM Tag command...done
OEM Tag:                                0x00000000

Get System Inegetrator ID command...This slot is unused.
Slot 1 Board Manufacturer:              Unused

Get System Inegetrator ID command...This slot is unused.
Slot 2 System Assembler:                Unused

Get System Inegetrator ID command...This slot is unused.
Slot 3 Reserved:                        Unused

Get M3 Autotest command...done
M3 Autotest:                            Disabled

Get ME FWU Platform Attribute (WLAN ucode) command...done
Localized Language:                     English

Get ME FWU Info command...done
Independent Firmware Recovery:          Enabled

skochinsky, 2017-09-07:
Read the flash back and compare; I suspect you flashed an unmodified image.

platomav, 2017-09-07:
Did you perform a ME reset after the modification? It is essential for 5MB SKU. Either run Flash Programming Tool with parameter "fptw -greset" or do it manually by removing all power (AC + any battery, RTC is ok) for 1 minute.

Raduq91, 2017-09-07:
@skochinsky, prob I flashed the modified image, I did this more than prob 10 times today only(whole process- bios mod with me_cleaner and bios flash), plus I have a bios image with unlocked menu and I can see the menu missing by flashing a different bios.

@platomav, did the fptw64 -greset two times, each time with fresh bios cleaned with me_cleaner

Intel ME System Tools v9.0 r1\MEInfo\Windows64>MEInfoWin64 -verbose

Intel(R) MEInfo Version: 9.0.22.1467
Copyright(C) 2005 - 2013, Intel Corporation. All rights reserved.

FW Status Register1: 0x1E000245
FW Status Register2: 0x66002306
FW Status Register3: 0x00000300
FW Status Register4: 0x00000000
FW Status Register5: 0x00000000
FW Status Register6: 0x00000000

  CurrentState:                         Normal
  ManufacturingMode:                    Disabled
  FlashPartition:                       Valid
  OperationalState:                     M0 with UMA
  InitComplete:                         Complete
  BUPLoadState:                         Success
  ErrorCode:                            No Error
  ModeOfOperation:                      Normal
  Phase:                                HOSTCOMM Module
  ICC:                                  Valid OEM data, ICC programmed
  ME File System Corrupted:             No

Get ME FWU version command...done

  Windows OS Version : 6.2.9200 ""
  OS BIOS Support    : UEFI

    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes

  Windows OS Version : 6.2.9200 ""
  OS BIOS Support    : UEFI

    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes
    Table Type   1 ( 0x 01 ) found, size of  27 (0x 1B ) bytes
    Table Type   2 ( 0x 02 ) found, size of  15 (0x 0F ) bytes
    Table Type   3 ( 0x 03 ) found, size of  22 (0x 16 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type  11 ( 0x 0B ) found, size of   5 (0x 05 ) bytes
    Table Type  24 ( 0x 18 ) found, size of   5 (0x 05 ) bytes
    Table Type  32 ( 0x 20 ) found, size of  20 (0x 14 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   4 ( 0x 04 ) found, size of  42 (0x 2A ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type  16 ( 0x 10 ) found, size of  23 (0x 17 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  19 ( 0x 13 ) found, size of  31 (0x 1F ) bytes
    Table Type 129 ( 0x 81 ) found, size of   8 (0x 08 ) bytes
    Table Type 130 ( 0x 82 ) found, size of  20 (0x 14 ) bytes
    Table Type 131 ( 0x 83 ) found, size of  64 (0x 40 ) bytes
        MEBx Version found is 9.0.0.0011

Get ME FWU info command...done
[... 50 lines cut ...]
SPI Flash ID #1:                        EF4017
SPI Flash ID VSCC #1:                   20252025
SPI Flash BIOS VSCC:                    20252025
Protected Range Register Base #0 0x0
Protected Range Register Limit #0 0x0
Protected Range Register Base #1 0x0
Protected Range Register Limit #1 0x0
Protected Range Register Base #2 0x0
Protected Range Register Limit #2 0x0
Protected Range Register Base #3 0x0
Protected Range Register Limit #3 0x0
Protected Range Register Base #4 0x0
Protected Range Register Limit #4 0x0
BIOS boot State:                        Post Boot
OEM Id:                                 00000000-0000-0000-0000-000000000000
Capability Licensing Service:           Enabled

Get ME FWU OEM Tag command...done
OEM Tag:                                0x00000000

Get ME FWU Platform Attribute (WLAN ucode) command...done
Localized Language:                     English

Get ME FWU Info command...done
Independent Firmware Recovery:          Enabled

corna, 2017-09-07:
@Raduq91 How are you flashing it? External programmer or internal flashing tool?

I think you're running an unmodified firmware. If you're using the OEM flashing tool it may be possible that it detects the ME image as corrupted, and flashes only the BIOS region.

Moreover the image you've linked is not in a plain format: did you use _me_cleaner_ directly on the image taken from the download link you posted?

Raduq91, 2017-09-07:
@corna I am flashing it from the bios menu, no external.
I used UEFITool to extract the image so it works with me_cleaner.

skochinsky, 2017-09-07:
> I am flashing it from the bios menu, no external.

This probably only updates the BIOS region, and leaves ME alone.

Raduq91, 2017-09-07:
Can Intel ME System Tools v9.0 flash the full image, or try flashrom from ubuntu?

corna, 2017-09-07:
The descriptor inside that file tells us that the ME region is read/write protected, so the only ways to flash a new ME image are through the UEFI firmware or with an external programmer, flashrom won't help you unfortunately.

Raduq91, 2017-09-07:
How I go about using the "UEFI firmware" you are referring?

corna, 2017-09-07:
Through your current bios. However it seems that it refuses to flash the modified ME image (signature? checksum?) but I don't know UEFI very well, so I don't know if it is possible to force it to flash the image. The safest option (remember that _me_cleaner_ can brick your PC) is to use an external programmer (which is not something expensive or hard to use, any Linux board with a SPI interface, like a Raspberry Pi, is OK).

Raduq91, 2017-09-07:
Where is AltMeDisable bit set by me_cleaner, in ME image?
MeInfo shows "Local FWUpdate:                         Enabled",  related to "Intel ME System Tools v9.0 r1\FWUpdate"?

skochinsky, 2017-09-07:
> Where is AltMeDisable bit set by me_cleaner, in ME image?

It is in the descriptor region, so you need writeable descriptor to set it.

"Local FWUpdate" allows you to use `FWUpdLcl`  tool to update the ME firmware without reflashing the SPI chip directly. This requires working (not neutered) ME and only accepts full firmwares.

ilikenwf, 2017-09-25:
If it is an AMI bios, go download AMIBCP v4 or v5 depending on your aptio version, and then use it to toggle the option to "true" to enable ME reflash on bios updates, and if you so desire, toggle the lock bits that block flashrom use. Optionally you can modify image options with the Intel FIT for your ME version too (http://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html)

Then, flash the modified bios twice. If this doesn't work, flash a prior/older version, then flash the modified bios twice.

Let me know if you have trouble finding AMIBCP or the other AMI tools, I can point you in the right direction.

Source: I have an ASRock z97 pro4 with a clean ME.

Raduq91, 2017-09-25:
I have my bios's hidden menus unlocked. Setting "Disable ME" to "Enabled" results in power cycle loop, clear cmos does not fix it, had to use the secondary bios to restore, same thing happens if I set to "Enable" "Me  FW Image Re-Flash".
This board supports "Intel Small Business Advantage 2", recently discovered after clearing cmos I can enter MEBx(verry few settings there) using I think Ctrl+P.
What are the lock bits that block flashrom use?

Raduq91, 2017-12-04:
Hello, I bought a CH341A(black version) USB, removed the main motherboard bios chip from the socket(bent a few pins on the chip, they are really malleable), switched to the backup bios using the jumper, booted into windows 10(x64), opened the tool, did a read test and backed up the bios, did a erase and read test, flashed a moded bios from months ago( I dont remember if its Alt Disable Mode only or cleaned or both, maybe both), put the bios chip back on the motherboard switched the bios jumper to normal and:

Intel ME System Tools v9.0 r1\MEInfo\Windows64>MEInfoWin64 -verbose

Intel(R) MEInfo Version: 9.0.22.1467
Copyright(C) 2005 - 2013, Intel Corporation. All rights reserved.

FW Status Register1: 0x1E020191
FW Status Register2: 0x164D2142
FW Status Register3: 0x00000300
FW Status Register4: 0x00000000
FW Status Register5: 0x00000000
FW Status Register6: 0x00000000

  CurrentState:                         Init
  ManufacturingMode:                    Enabled
  FlashPartition:                       Valid
  OperationalState:                     Bring Up
  InitComplete:                         Initializing
  BUPLoadState:                         Success
  ErrorCode:                            No Error
  ModeOfOperation:                      Alt Disable Mode
  Phase:                                BringUp
  ICC:                                  No valid OEM data, ICC programmed with default values
  ME File System Corrupted:             Yes
  PhaseStatus:                          UNKNOWN

HECI device is found to be disabled.

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

  Windows OS Version : 6.2.9200 ""
  OS BIOS Support    : UEFI

    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes

  Windows OS Version : 6.2.9200 ""
  OS BIOS Support    : UEFI

    Table Type   0 ( 0x 00 ) found, size of  24 (0x 18 ) bytes
    Table Type   1 ( 0x 01 ) found, size of  27 (0x 1B ) bytes
    Table Type   2 ( 0x 02 ) found, size of  15 (0x 0F ) bytes
    Table Type   3 ( 0x 03 ) found, size of  22 (0x 16 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type   9 ( 0x 09 ) found, size of  17 (0x 11 ) bytes
    Table Type  11 ( 0x 0B ) found, size of   5 (0x 05 ) bytes
    Table Type  24 ( 0x 18 ) found, size of   5 (0x 05 ) bytes
    Table Type  32 ( 0x 20 ) found, size of  20 (0x 14 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   4 ( 0x 04 ) found, size of  42 (0x 2A ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type   7 ( 0x 07 ) found, size of  19 (0x 13 ) bytes
    Table Type  16 ( 0x 10 ) found, size of  23 (0x 17 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  17 ( 0x 11 ) found, size of  34 (0x 22 ) bytes
    Table Type  20 ( 0x 14 ) found, size of  35 (0x 23 ) bytes
    Table Type  19 ( 0x 13 ) found, size of  31 (0x 1F ) bytes
    Table Type 127 ( 0x 7F ) found, size of   4 (0x 04 ) bytes

Error 9256: Communication error between application and Intel(R) ME module (FW Update client)

Error 9459: Internal error (Could not determine FW features information)

Intel ME System Tools v9.0 r1\MEInfo\Windows64>

Previews there was a Intel management engine interface driver in Device Manager, now I dont see it.

kocoman1, 2023-05-07:
maybe the backup bios is flashing it back or its using the backup bios's ME ?