corna/me_cleaner #62: Asus H81M-D (cap/rom) Success - Instructions

Noneneeded2, 2017-09-08:
Hello Corna, thank you very much for your hard work. I'd like to contribute a bit to the community with the instructions for Asus mobos (cap files are a pain). Instructions are for Windows and I'm not taking responsibility in case you screw things up, so here goes: 

1. Get your bios from the official Asus site

2. Get UefiTool 0.21.5

3. Open up your .CAP file in Uefitool, right click and extract body. Now you got a .ROM file

4. Get the dev version of ME_Cleaner (I got it on 7th September 2017) 

5. Use this command for HAP switch: "me_cleaner.py -s -O modified.rom youroriginal.rom" without quotes. Remember, lowercase -s, not uppercase -S. If all is OK, me_cleaner will wish you good luck.

6. Now the tricky part is ASUS EZ-Flash from UEFI doesn't accept .ROM files. You need to force it either through DOS or from Windows. In my case Windows works just fine,

7. Get the Aptio AFU Win Flasher IV or V depending on your mobo. For H81M-D it's IV but there is a security lock so we need the previous AFU 3.05.04 - yes, exactly this version because it supports to security overriding /GAN feature. Can't link here because it's not redistributable.

8. Now first try the IV or the V versions (depending on your mobo, AFU will tell you to use IV or V) with this exact command: "Afuwinx64 modified.rom /P /B /K" without quotations, if you're doing this via dos, well you need the DOS version and command becomes "afudos modified.rom /P /B /K"

8a. If you're getting this: 

"Reading flash ............... done
- FFS checksums ......... ok
Erasing Boot Block .......... done
Updating Boot Block ......... done
Verifying Boot Block ........ done
Erasing Main Block .......... done
Updating Main Block ......... done
Verifying Main Block ........ done
Erasing NCB Block ........... done
Updating NCB Block .......... done
Verifying NCB Block ......... done"

Then SUCCESS, you've flash your .ROM file. However, if you're getting "security verification failed" error, this is where the 3.05.04 version comes into play.

8b. Use 3.05.04 version with this command: "afuwinx64 modified.rom /GAN" without quotes. /GAN overrides the security. It's going to take a little while to flash. You should be getting these outcome:

"Reading flash ............... done
Erasing All Block ............ done
Updating All Block ........... done
Verifying All Block .......... done" 

You're done! Congrats. 

Now your start times will be bad though. In my case it's like seven seconds additional waiting at the screen before Windows kicks in loading.

And again, thanks to Corna for his efforts.

corna, 2017-09-26:
Thank you for sharing your discovers, I've linked it to [this Wiki page](https://github.com/corna/me_cleaner/wiki/How-to-apply-me_cleaner).

closed 2017-09-26 by corna

ninjagoat, 2017-12-01:
Thanks a lot for the guide @Noneneeded2 

Tried this on my Asus B85M-G but I'm not really sure if it succeeded.

When I ran me_cleaner, it said that the AltMeDisable bit has been set. So all good I guess!

After that I got the same security verification error as you, while flashing the rom file.

So I did it again with 3.05.04 as you suggested. Everything went well and the BIOS got reset completely. I then reconfigured it and booted into the OS.

Now I ran intelmetool and here are some observations -

@corna you have mentioned that we should see ME DISABLED for Progress Phase State but it still shows "Host communication established" like before

Power Management Event has changed from "Clean Moff -> Mx wake" to "Psuedo-global reset"

And the **first** time I run intelmetool it now shows all the "capablities" as OFF (earlier some were on and others were off)
After that, it just shows the following error every time I run it
ME: timeout waiting for data: expected 5, available 0
ME: GET FWCAPS message failed

So I'm guessing that it worked? And ME DISABLED will only appear in ME >=11?

Thoughts @corna ?

GreenBeard, 2018-08-02:
If anyone is looking for AFU 3.05.04 the official website's link does not appear to currently be working. The direct link (which still is working) is: https://www.wimsbios.com/files/flashers/ami/afuwin/afuwin-3.05.04.zip . Change the ending to the version you want if you want a different one. (I found the correct URL using archive.org).

Jedihoth, 2019-02-26:
> 
> 
> Thanks a lot for the guide @Noneneeded2
> 
> Tried this on my Asus B85M-G but I'm not really sure if it succeeded.
> 
> When I ran me_cleaner, it said that the AltMeDisable bit has been set. So all good I guess!
> 
> After that I got the same security verification error as you, while flashing the rom file.
> 
> So I did it again with 3.05.04 as you suggested. Everything went well and the BIOS got reset completely. I then reconfigured it and booted into the OS.
> 
> Now I ran intelmetool and here are some observations -
> 
> @corna you have mentioned that we should see ME DISABLED for Progress Phase State but it still shows "Host communication established" like before
> 
> Power Management Event has changed from "Clean Moff -> Mx wake" to "Psuedo-global reset"
> 
> And the **first** time I ran intelmetool it showed all the "capablities" as OFF (earlier some were on and others were off)
> After that, it just shows the following error every time I run it now
> ME: timeout waiting for data: expected 5, available 0
> ME: GET FWCAPS message failed
> 
> So I'm guessing that it worked? And ME DISABLED will only appear in ME >=11?
> 
> Thoughts @corna ?

can anyone make a visual step-by-step tutorial on youtube ???

clip123, 2020-12-03:
Thanks for great work on ME Cleaner. I tried this guide for Asus H81M-D bios and me_cleaner.py worked - no issues but I don't have that motherboard and it was for testing purpose.

Then I grabbed bios for Asus ZenBook series online (listed below) but I get error message for both bios stating "Unknown file" after extracting and running me_cleaner.py script. **Any suggestions?**

Zenbook bois:
UX32LNAS204-20190625
UX32VDAS214-20130307

Manoa1911, 2022-10-19:
thank :) I think it worked on my Z97-A :)