corna/me_cleaner #63: Internal flashing fails due to make failing.

FrostKnight, 2017-09-09:
https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-OEM-firmware

it says there are no targets for make stop

on the third step of that guide. Am I missing something?

corna, 2017-09-09:
Do you mean the `make` here?

```
 $ git clone --depth=1 http://review.coreboot.org/p/coreboot
 $ cd coreboot/util/ifdtool
 $ make
```

FrostKnight, 2017-09-09:
Alas, yes that is what I meant. Any solution?

FrostKnight, 2017-09-09:
> Are you sure you're running |make| inside the |coreboot/util/ifdtool|
> folder?

okay, new problem(s), one: ifdtool -d original-firmware.bin says, no
tool found.

two:

how do you dump the original firmware onto a disk or something? aka
backup the firmware... that's my main question.

corna, 2017-09-21:
> ifdtool -d original-firmware.bin says, no tool found.

ifdtool should be used on the dumped firmware (`original-firmware.bin`), do you have it?

> how do you dump the original firmware onto a disk or something? aka backup the firmware... that's my main question.

See [here](https://github.com/corna/me_cleaner/wiki/External-flashing) (recommended) or [here](https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-OEM-firmware). Which part is causing you troubles?

Flashing a modified image is **not** an easy or safe operation, be careful.

xcookie910, 2017-09-22:
`sudo apt-get install build-essential`

corna, 2017-10-13:
I assume you've solved the problem, feel free to reopen if you need more help.

closed 2017-10-13 by corna