corna/me_cleaner #64: funny sidequest ... race condition?

mereportertmp12432, 2017-09-09:
hi, i have a pansonic cf19mk6 with qm77 chipset and this is the story of how I was able to corrupt - and hopefully disable - my ME using fwupdlcl and the integrated programmer, so it may inspire future "research".

results:
other oem strings @ panasonic pcinfo http://picpaste.com/diff-pcinfo.png
PRE-BOOT and other ME-Name @ meinfo http://picpaste.com/diff-meinfo.png
Recovery state and two wiped registers @ http://picpaste.com/diff-intelmetool.png
fwuplcl -fwver shows version, but -save and -f just hang
memanuf reports some error
ctrl-p reports "FW Status Recovery Error" and then just boots

my highly scientific approach:
# panasonic sa00075 ME.bin
C:\UpdateMeFirmware\Data801>FWUpdLcl.exe -oemid D6B09D64-DA23-49A9-8888-F663BE603389 -allowsv -f "ME.bin"
Intel (R) Firmware Update Utility Version: 8.1.40.1456
Copyright (C) 2007 - 2013, Intel Corporation. All rights reserved.
Communication Mode: MEI
Checking firmware parameters...
Warning: Do not exit the process or power off the machine before the firmware update process ends.
Sending the update image to FW for verification: [ COMPLETE ]

FW Update: [ 35% (Stage: 13 of 19) (-)]
# HIBERNATE, last seen "50%" and Stage 14/19

^C Update: [ 0% (Stage: 0 of 19) (|)])]
# RESUME, now see 0% ... and program hangs, so press ctrl-c

C:\UpdateMeFirmware\Data801>FWUpdLcl.exe -oemid D6B09D64-DA23-49A9-8888-F663BE603389 -allowsv -f "MEREG-muchdisable.bin"
# me_cleaned version of a MERegion.bin that has been extracted via FIT from ... any of the many ME.bins that I have tried, including selfmade ones
Intel (R) Firmware Update Utility Version: 8.1.40.1456
Copyright (C) 2007 - 2013, Intel Corporation. All rights reserved.
Communication Mode: MEI
Checking firmware parameters...
Warning: Do not exit the process or power off the machine before the firmware update process ends.
Sending the update image to FW for verification: [ COMPLETE ]

FW Update: [ 35% (Stage: 13 of 19) (-)]
# it directly jumps to Stage 13 35% ... cool?

FW Update: [ 100% (Stage: 19 of 19) (-)]
FW Update is complete and a reboot will run the new FW.

no issues so far, doesnt powercycle after 30 min or anything. seems good to me, especially the "pre-boot"but what the heck do i know

mereportertmp12432, 2017-09-09:
the files that i used
https://filebin.ca/3ZoqtxiQEx5m/ME.bin
https://filebin.ca/3ZorKoSiEbI2/MEREG-muchdisable.bin

mereportertmp12432, 2017-09-09:
i should also mention that 
a) a direct fwupdlcl -f MEREG-muchdisable.bin would error at step 4 (2nd verification)
b) fptw doesnt work due to host read/write disabled

mereportertmp12432, 2017-09-10:
any suggestions for one of this hipster vulnerability names?
i'll throw "UpDateME" and "Sleeper Hold" in the room for starters.

archfan, 2017-09-10:
> any suggestions for one of those hipster vulnerability names?
Hodl me tight.