corna/me_cleaner #66: How does me_cleaner affect SMM?

xcookie910, 2017-09-21:
_Sorry if this question might look offtopic I could not find a straight answer anywhere._

How does this me_cleaner tool affect the SMM that is also present in many Intel processors since the 90's. While cleaning the ME is good, it removes ring -3 vulnerabilities, the SMM operates in ring -2.

- Does this tool remove some attack vectors presented in SMM (I guess it does for UEFI since there were rootkits for AMT+SMM, but I am wondering more about legacy BIOS) ?
- Will you @corna work on a SMM cleaner/patcher in the future?
- Can the SMM be cleaned in any other way, perhaps refer to somebody who already created a similar tool to clean it?

Thanks.

corna, 2017-09-21:
* No
 * No

The ME firmware, while it is located on the same chip of the BIOS, it is logically separated in its own flash partition. The SMM is part of the BIOS and is quite hard to be replaced without replacing also the BIOS and the ACPI tables. Moreover, while the Intel ME firmware can be shrinked just to the minimum possible (thanks to the fact that it is fully modular, it maintains the same firmware structure across different versions and it has non-fundamental parts), it is difficult to do it for the SMM, as it plays a major role in the correct PC functioning.

* Yes (sort of). The SMM is board-specific and it's impossible to create a generic tool (like _me_cleaner_) to remove the problematic parts of the SMM code from any board. However it can be fully reimplemented, and the coreboot project does it. The coreboot developers consider the SMM a dangerous feature and try to write as little code as possible in SMM, resulting in a very small implementation of it (on my X220 on coreboot, [these](https://gist.github.com/corna/fdf21360318c049569a301b34560cd8c) are the source files built in SMM, not much code).

If you want to remove all the attack vectors, the best option is coreboot (if your board is supported, or if you can port coreboot to it):
 * FOSS
 * Support for _me_cleaner_ on any platform (except for Nehalem, WIP)
 * Minimal SMM

closed 2017-09-21 by corna

xcookie910, 2017-09-22:
Yes but Coreboot & Libreboot barely support a few hardware architectures, a more general patch is needed for others.

Alright perhaps the SMM can't be removed entirely since that would take a lot of work but it can be patched, which would be farely simple, and it can even be integrated in this python project or another one similar. Just running a python script on the firmware binary and it would patch all known vulnerabilities so far.

Actually I did some research on this. It turns out that the firmware actually has protection mechanisms, against SMM vulnerabilities, which are either disabled by default or wrongly configured.

So it would be trivial to patch them, basically just flipping a few bits like with the HAP stuff here. Things like: Write protect the BIOS, Lock SPI memory range, Lock SMRAM and things like this can all be done by flipping a few bits.

It would be much easier and might be multi-platform compatible.

Then there are other things like a few non-essential modules that can be completely removed, but this might be harder and more specific.

There is a tool called "Chipsec" that can be of a good guide in this research:
* https://github.com/chipsec/chipsec

I would love to see somebody working on this. SMM vulnerability is the next big issue after ME.

citypw, 2017-09-24:
@xcookie910 Unfortunately, the most OEM vendor's FW has much more SMIs handlers than coreboot that will cause more attack surfaces. SMM is an old problem:
https://github.com/hardenedlinux/firmware-anatomy/blob/master/hack_ME/firmware_security.md

CHIPSEC is a very good starting point to audit your machine. On the other hand, the attacker must gain the root priv to mess with SMM. Try PaX/Grsecurity( linux kernel) to prevent the attacker getting root priv or it doesn't make much sense to let the attacker run the exp on the target with priviledged permission.

skochinsky, 2017-09-24:
In theory you could try removing SMM modules from the UEFI volumes, similar to how me_cleaner removes ME modules  from ME firmware, but you're likely to get unbootable box this way (and/or power modes/ACPI/sleep issues). However, this is a separate problem.