corna/me_cleaner #67: Is this a problem for Macs?

mustafasezgin, 2017-09-21:
Hi

Ive been looking into this nasty 'feature' that intel provides and wanted to know if I can run this on OSX machines or if the ME is indeed available macs with intel CPUs.

Ive tried searching online and through the issues/docs here but cannot find any reference or info on this. Any info would be greatly appreciated.

corna, 2017-09-21:
Hi,

Intel ME is present on every Intel processor, also on Macs then. You can use _me_cleaner_ on any platform that has Python, but you have to find a way to flash back the modified firmware (and I think that on Macs the only way is through an external programmer). Note that no one has tested _me_cleaner_ on Macs yet (see #45).

mustafasezgin, 2017-09-21:
Thanks @corna

closed 2017-09-21 by mustafasezgin