corna/me_cleaner #68: Does not work on Gigabyte GA-Z170MX Gaming 5

adolfintel, 2017-09-22:
The BIOS image can be flashed with an external tool or a modded efiflash.exe (attached for your convenience), but the machine keeps rebooting after POST. It is still possible to access the BIOS setup to reflash the original image without external aid.
[Efiflash.zip](https://github.com/corna/me_cleaner/files/1323474/Efiflash.zip)

platomav, 2017-09-22:
Gigabyte has a Main and Backup BIOS (M_BIOS, B_BIOS) chip. On older boards there was a switch but on newer ones the recovery is done automatically upon a failed boot attempt. These reboots should be the B_BIOS trying to replace the contents of the M_BIOS but maybe the automatic process fails somehow.

adolfintel, 2017-09-22:
I donct think this is the case because when the main BIOS is not working or corrupt it does that reboot before POST, here the machine reboots after POST; I can enter the BIOS setup and it works correctly, it reboots when it starts the OS instead. I suspect that there may be a check for the ME firmware in the BIOS, is there any way to know without reverse engineering the BIOS?

platomav, 2017-09-22:
It is very likely that there is a ME health check at the BIOS but it doesn't make sense to trigger when the OS has started to load, as you say, since that's after BIOS/POST. To my understanding, modern GB Dual BIOS implementation boots from M_BIOS and if something is wrong, it sets a flag/bit etc and reboots. At the next reboot, it automatically runs from B_BIOS and flashes its contents to the M_BIOS. Try to shutdown the system and remove power (cord, turn off PSU switch) and leave it for some seconds. Maybe that will help the recovery process in case it is "stuck" or similar. Otherwise, maybe your B_BIOS is also corrupt?

adolfintel, 2017-09-22:
I tried that, and I also tried flashing the backup bios; it does not try to load it, which is what lead me to think that it checks if the ME has started properly and reboots the machine if it didn't. But this check seems to be done after POST (the ME needs a few seconds to boot I guess)

adolfintel, 2017-09-22:
In other words, I don't think it's a sanity check, it just wants to make sure that everthing is working as expected before booting the OS

platomav, 2017-09-22:
> I also tried flashing the backup bios
What do you mean by that? Does GB allow reflashing of the B_BIOS in modern systems? At my old Z77 it does allow that via the BIOS switch but I doubt the modern automatic process facilitates such action.

> the ME needs a few seconds to boot I guess
No it actually initializes even before the main CPU but does communicate with the BIOS while that does POST.

adolfintel, 2017-09-22:
When you flash using the Q-Flash tool from the BIOS setup it flashes both chips, or so it says. Efiflash does not.

platomav, 2017-09-22:
Did you flash both chips with Q-Flash? That would explain what's going on. The system is stuck in a boot loop because the B_BIOS also fails to pass the ME check and thus cannot repair M_BIOS.

adolfintel, 2017-09-22:
No no, I flash the original bios with Q-Flash, then the modified one using that Efiflash that I've attached. I modifed it to ignore the checksums (just a couple of JMPs to change) and it flashes only the main BIOS.

platomav, 2017-09-22:
Honestly there are way too many variables to pinpoint what went wrong (efiflash - even more so modded efiflash, how GB Dual BIOS works exactly, how GB BIOS deals with problematic ME firmware etc). Provided that you have read/write access to the ME region (unlocked Flash Descriptor), I suggest you use other flashers such as flashrom and/or Flash Programming Tool. That rules out some efiflash mess-up. Everything else is GB BIOS related (Dual BIOS recovery, corrupt ME reaction) so technically not a me_cleaner issue. For future reference it would help if you mentioned what exact options you used at me_cleaner (none, -s etc) but for now, I suggest you use a programmer to repair the BIOS.

adolfintel, 2017-09-22:
Before modifying efiflash I tried flashing using an external raspberry pi, but it gave the same result, that's why I tried to use efiflash.
flashrom does not work on my system unfortunately or I would have used it.
As for me_cleaner options, I tried both deleting and just disabling with the bit, same result.
In my opinion, there is some check in the BIOS. I'm not blaming you for it of course, your tool can't know.

platomav, 2017-09-22:
I'm not the developer of me_cleaner, [Nicola Corna](https://github.com/corna) is. By excluding efiflash you end up with the conclusion that it is a matter of Gigabyte's way of handling corrupt ME firmware. I would assume that the normal behavior is for the B_BIOS to reflash the entire M_BIOS even in case of corrupt ME but this doesn't seem to happen.

syboxez, 2017-11-07:
Just wanted to say that I am having the exact same results with my Gigabyte GA-Z170X-UD3. After POST when flashing using the modified Efiflash (only HAP bit set and nothing else), it shutdown and rebooted every time I tried to load any operating system (either UEFI or BIOS). Was able to reflash the original using the built in utility.

When using me_cleaner without the HAP bit however, it did not boot at all, and it simply flashed from the backup BIOS.

seropp, 2017-12-31:
For the sake of excluding the modded efiflash as a variable, you could try the FPT executable from the Intel (CS)ME bundle

https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html

I tested it on a GA-B250M-D3H (the r6 version), it also requires the fparts.txt file to be on the same stick, and there's an EFI executable in there somewhere that works the same as the exe on freedos. Unfortunately I had results identical to the modded efiflash tool. It has a dump feature that could be useful, but as far as getting it to boot, no success.

adolfintel, 2017-12-31:
@seropp same result for me too.
I had already tried flashing the chips externally with a raspberry pi, with the same result.