corna/me_cleaner #69: Windows 8

kolor33, 2017-09-24:
Hello I'm stupid man from Poland .I can't know how to start this file me_cleaner.py on Windows 8.May somebody show me instructions how to start py on Windows.
Step by step 

Thanks

matt123b, 2017-09-25:
You aren't stupid, just a little misinformed. ME Cleaner isn't meant to be flashed internally, which means from an OS running on the computer (at least not as it is now from my understanding). I think there's another thread up about this very issue under the issues section of this project if you want to take a look. You'll need to flash your BIOS ROM externally, meaning you need to strip your machine down to the motherboard, hook a clip to the pins of the actual chip, and then use the ME Cleaner program to strip down the ME firmware and flash it back. You have to do this because the ROM is write protected on a lot of computers from within the OS. I know that alternative BIOS like Coreboot or Libreboot open other functionality and allow you to modify from the OS level but this usually isn't the case with stock BIOS, and I'm guessing you're using that if you're running Windows. Even if you could flash internally I would recommend not doing it because it will just be a huge pain if you mess stuff up, whereas with external flashing you already have your tools set up and if you made a backup of the stock firmware you can restore it easily should something go wrong. Any way you put it this is a somewhat risky process and you should not attempt it on any of your important work/school computers or stuff like that. 

If you're an average user who is looking for a machine that respects your freedom then I recommend buying a laptop with Libreboot preinstalled. I don't know who sells them these days but I'm sure the FSF website has more info.

matt123b, 2017-09-25:
Here's a tool called flashrom that can be used for internal flashing. Do a ctrl+f on this list of hardware to see if your device is supported, if not you have to flash externally. 

https://www.flashrom.org/Supported_hardware

corna, 2017-09-25:
@matt123b Correct.

> You aren't stupid, just a little misinformed.

Exactly; it's a complex and undocumented field, it's very common to be misinformed, don't worry.

Applying _me_cleaner_ is divided in three parts:
 * obtain the original firmware
 * modify it
 * flash back the modified image

The complete process is described in the [wiki](https://github.com/corna/me_cleaner/wiki).
You can safely test _me_cleaner_ by applying it to a firmware dump without flashing it: just download the firmware from your motherboard's website (if available), download and install Python (don't forget to set the `Add Python 3.x to PATH` option during the installation), open PowerShell, run `python me_cleaner.py <downloaded image>` and check the output. Read carefully the wiki and the past Issues before flashing it.

> ME Cleaner isn't meant to be flashed internally

You can flash me_cleaner internally if your motherboard:
 * is supported by flashrom (quite common)
 * has the ME region RW (uncommon)

You can also use the OEM flashing tool, which can override the RW protection of the ME region; however it often (especially with UEFI) has a signature on the ME image and refuses to flash it (or even silently fails) if it is invalid.

Moreover recovering from a brick without an external programmer is harder, and depends on the OEM recovery tool (integrated recovery flasher, dual BIOS chip, ...).

TL;DR: you can flash internally, but you shouldn't (unless you have coreboot).

JonnyTech, 2017-09-25:
Can you recommend an external flashing device? Preferably something that is compatible with Linux.

corna, 2017-09-25:
Any device [supported by flashrom](https://www.flashrom.org/Supported_programmers), but I suggest you a Raspberry PI (or any Linux board with a SPI interface), very fast and no troubles.

JonnyTech, 2017-09-25:
Ah, I have several pi's. How would I connect a chip to it? (Sorry for hijacking this thread)

corna, 2017-09-25:
See [here](https://github.com/corna/me_cleaner/wiki/me_cleaner-on-a-MSI-H110M-ECO) and [here](https://hardenedlinux.github.io/firmware/2016/11/17/neutralize_ME_firmware_on_sandybridge_and_ivybridge.html).

JonnyTech, 2017-09-25:
Excellent - thank you!